linrunner/TLP #382: Kernel Boot Parameters not working when under TLP

reikdas, 2019-01-21:
OS - Manjaro i3 community edition
Laptop - MSI GS65 Stealth Thin 8RF https://wiki.archlinux.org/index.php/MSI_GS65

### Symptom data

1. Does the problem occur on battery or AC or both?
Both

2. Attach the *full* output of `tlp-stat` via https://gist.github.com/ for *all* cases of 1.
I am unable to attach the output of the command because I have disabled and removed TLP - It took a lot of debugging to enable the systemd power management in place of TLP so I am unwilling to risk going through that again. 

### Actual behavior
This is the relevant line from /etc/default/grub,
GRUB_CMDLINE_LINUX_DEFAULT=‘quiet acpi_osi=! acpi_osi=“Windows 2009” modprobe.blacklist=nouveau’

I pressed e on my Manjaro option on boot and these commands were present in the kernel boot parameters.

But,
I added the modprobe.blacklist=nouveau because that is listed as a fix for poweroff crashing on my laptop. (https://wiki.archlinux.org/index.php/MSI_GS65#Lockup_Issue_(lspci_and_poweroff_hang) 2)
I added the acpi option to enable wifi unblocking after suspend. (https://wiki.archlinux.org/index.php/MSI_GS65#Wifi_is_hardblocked_(airplane_mode)_after_waking_up_from_suspend 2)
But I am still facing both the issues.

Disabling TLP fixed both the issues (which means that the Kernel Boot Parameters started working properly again).

linrunner, 2019-01-21:
Hi, please use the [Troubleshooting Guide](https://linrunner.de/en/tlp/docs/tlp-troubleshooting.html) to identify the offending setting

Btw: your tlp-stat output is missing.

linrunner, 2019-02-11:
Any news on this?

reikdas, 2019-02-11:
It took a lot of debugging to enable the systemd power management in place of TLP so I am not very keen on setting up TLP on my system again.. 
Maybe close the issue and I'll get back to it later?

linrunner, 2019-02-11:
This can't be solved without your contribution, so feel free to reopen when you have more data.

Btw: to my knowledge a "systemd power management" doesn't exist, just wondering ...

closed 2019-02-11 by linrunner