linrunner/TLP #386: tlp default configuration prevent platform from reaching S0ix

RussianNeuroMancer, 2019-02-06:
### Symptom data

1. Problem occur on battery
2. https://gist.github.com/RussianNeuroMancer/6470afd69e2400f8f2062303eef6d755

### Expected behavior

tlp default configuration doesn't prevent platform from reaching S0ix on supported devices such as Dell XPS 13 9370 and HP Elite x2 1013 G3 (latter support S0ix with additional patches on top of Linux 5.0). Other more widely available supported devices is Intel BayTrail-based tablets and laptops, S0ix should work on most of such devices since Linux 4.20 (for example S0ix works on HP Stream 7 Tablet, including S0i3 state).

### Actual behavior

tlp default configuration prevent platform from reaching S0ix, while removing tlp allow to get S0ix working. Example of such behaviour is available here:
With tlp: https://bugzilla.kernel.org/show_bug.cgi?id=201579#c35
Without tlp: https://bugzilla.kernel.org/show_bug.cgi?id=201579#c39 (running `echo 'auto' > '/sys/bus/usb/devices/2-3/power/control'` or `powertop --auto-tune` let MPHY CORE LANE 2 to get power gated).

With tlp:

/sys/kernel/debug/pmc_core/mphy_core_lanes_power_gating_status content before running `echo 'auto' > '/sys/bus/usb/devices/2-3/power/control'`:

```
MPHY CORE LANE 0                	State: Power gated
MPHY CORE LANE 1                	State: Power gated
MPHY CORE LANE 2                	State: Not power gated
MPHY CORE LANE 3                	State: Power gated
MPHY CORE LANE 4                	State: Power gated
MPHY CORE LANE 5                	State: Power gated
MPHY CORE LANE 6                	State: Power gated
MPHY CORE LANE 7                	State: Power gated
MPHY CORE LANE 8                	State: Not power gated
MPHY CORE LANE 9                	State: Power gated
MPHY CORE LANE 10               	State: Power gated
MPHY CORE LANE 11               	State: Power gated
MPHY CORE LANE 12               	State: Not power gated
MPHY CORE LANE 13               	State: Not power gated
MPHY CORE LANE 14               	State: Not power gated
MPHY CORE LANE 15               	State: Not power gated
```

/sys/kernel/debug/pmc_core/mphy_core_lanes_power_gating_status content after running `echo 'auto' > '/sys/bus/usb/devices/2-3/power/control'`:
```
MPHY CORE LANE 0                	State: Power gated
MPHY CORE LANE 1                	State: Power gated
MPHY CORE LANE 2                	State: Power gated
MPHY CORE LANE 3                	State: Power gated
MPHY CORE LANE 4                	State: Power gated
MPHY CORE LANE 5                	State: Power gated
MPHY CORE LANE 6                	State: Power gated
MPHY CORE LANE 7                	State: Power gated
MPHY CORE LANE 8                	State: Not power gated
MPHY CORE LANE 9                	State: Power gated
MPHY CORE LANE 10               	State: Power gated
MPHY CORE LANE 11               	State: Power gated
MPHY CORE LANE 12               	State: Not power gated
MPHY CORE LANE 13               	State: Not power gated
MPHY CORE LANE 14               	State: Not power gated
MPHY CORE LANE 15               	State: Not power gated
```

Without tlp after running `echo 'auto' > '/sys/bus/usb/devices/2-3/power/control'`:

```
MPHY CORE LANE 0                	State: Power gated
MPHY CORE LANE 1                	State: Power gated
MPHY CORE LANE 2                	State: Power gated
MPHY CORE LANE 3                	State: Power gated
MPHY CORE LANE 4                	State: Power gated
MPHY CORE LANE 5                	State: Power gated
MPHY CORE LANE 6                	State: Power gated
MPHY CORE LANE 7                	State: Power gated
MPHY CORE LANE 8                	State: Power gated
MPHY CORE LANE 9                	State: Power gated
MPHY CORE LANE 10               	State: Power gated
MPHY CORE LANE 11               	State: Power gated
MPHY CORE LANE 12               	State: Power gated
MPHY CORE LANE 13               	State: Power gated
MPHY CORE LANE 14               	State: Power gated
MPHY CORE LANE 15               	State: Power gated
``` 

### Steps to reproduce the problem

Install tlp on device that support S0ix (please follow [this guide](https://01.org/blogs/qwang59/2018/how-achieve-s0ix-states-linux) to verify if it's works) and check if S0ix still works after tlp installation.

linrunner, 2019-02-06:
Thanks for your research.

I can't identify the USB device concerned. Which one is it? How do you suggest changing the configuration or the code? Your pull request is welcome.

linrunner, 2019-02-11:
Had another look on the kernel bug today. Your conclusion is that the LTE modem doesn't get autosuspend thus preventing S0ix – right?

    03f0:0857 control = on, autosuspend_delay_ms = 2000 -- Hewlett-Packard (cdc_mbim, option)

WWAN blacklisting ( `USB_BLACKLIST_WWAN=1 ) seems not to be the cause, because TLP's [internal vendor based blacklist](https://github.com/linrunner/TLP/blob/1.1/tlp-functions.in#L66) does not contain 03f0.

Phone blacklisting is off ( `USB_BLACKLIST_PHONE=0` ), so I just run out of ideas.

Could you please enable trace mode ( `TLP_DEBUG="run usb"` ) and show the relevant trace output after

    tlp usb

with

    tlp-stat -T

RussianNeuroMancer, 2019-02-12:
Sorry, for delay.

> I can't identify the USB device concerned. Which one is it?

Snapdragon X12 LTE modem.

> Your conclusion is that the LTE modem doesn't get autosuspend thus preventing S0ix – right?

No, as you can see in this comment: https://bugzilla.kernel.org/show_bug.cgi?id=201579#c39 (paragraph `After disabling tlp, even without "powertop --auto-tune"`, which means modem is still not auto-suspended and MPHY Core Lane 2 is still not power gated, yet S0ix works nevertheless; Active GEN2 USB2PCIE2 PLL pll status also does not prevent S0ix). 

Main issue is: installing and running tlp with default configuration is prevent device from reaching S0ix:

1. It's seems like tlp change something which cause MPHY Core Lanes 8 and 12-15 stay not power gated, which is issue on it's own. My understanding of Rajneesh Bhardwaj comments is that not power gated MPHY Core Lanes 8 and 12-15 could prevent S0ix and I see that S0ix is indeed doesn't work with installed tlp, so probably this is because MPHY Core Lanes 8 and 12-15 is not power gated?
2. There is possibility that tlp doing something else besides item 1 which could prevent S0ix too, but to find if there is "something else" it's necessary to find what cause item 1.

> How do you suggest changing the configuration or the code? Your pull request is welcome.

I wish I know. Is there way to run tlp in some mode that doesn't touch any system settings, and then apply changes that tlp make to system configuration step by step to find which one prevent S0ix?

> Could you please enable trace mode ( `TLP_DEBUG="run usb"` ) and show the relevant trace output

Please let me know if this step is still necessary if we know that USB and MPHY Core Lane 2 is unrelated to this issue? (At least as far I understand.)

linrunner, 2019-02-13:
Disabling with TLP_ENABLE=0 is easy, but enabling only single features is not. I suggest you implement the strategy outlined in [Troubeshooting](https://linrunner.de/en/tlp/docs/tlp-troubleshooting.html).

My prime suspect would be RUNTIME_PM_ON_AC/BAT. 

You should also test with PCIE_ASPM_ON_AC/BAT  disabled (commented) because of https://github.com/linrunner/TLP/issues/344

USB_AUTOSUSPEND=0 or blacklisting devices one by one is also worth a try.

Please test with the latest code from the master branch. Packaging instructions [here](https://linrunner.de/en/tlp/docs/tlp-developer-documentation.html).

RussianNeuroMancer, 2019-02-13:
Thanks for advices, I will try to do this this week.

RussianNeuroMancer, 2019-02-23:
Sorry for delay. [I have issues with Linux 5.0 unrelated to power-saving](https://www.phoronix.com/forums/forum/phoronix/latest-phoronix-articles/1081905-the-most-interesting-highlights-to-the-linux-5-0-kernel?p=1081958#post1081958). Currently I trying to figure out how to continue testing without impact on stability of my system. I probably setup separate partition specifically for testing S0ix and tlp (I already build packages from master).

linrunner, 2019-02-23:
I made beta packages in the meantime: https://download.linrunner.de/packages/

linrunner, 2019-03-05:
Any news on this?

RussianNeuroMancer, 2019-03-05:
Unfortunately not yet.

RussianNeuroMancer, 2019-03-11:
> My prime suspect would be RUNTIME_PM_ON_AC/BAT.

Seems like this assumption turns out to be correct. I installed packages from https://download.linrunner.de/packages/ comment out RUNTIME_PM_ON_AC and RUNTIME_PM_ON_BAT in /etc/default/tlp, power cycle laptop, and find that S0ix is working even with running tlp.service: 

```
~# cat /sys/devices/system/cpu/cpuidle/low_power_idle_cpu_residency_us
0
~# cat /sys/devices/system/cpu/cpuidle/low_power_idle_system_residency_us
0
~# cat /sys/kernel/debug/pmc_core/slp_s0_residency_usec
0
~# cat /sys/devices/system/cpu/cpuidle/low_power_idle_cpu_residency_us
574503455
~# cat /sys/devices/system/cpu/cpuidle/low_power_idle_system_residency_us
103263465
~# cat /sys/kernel/debug/pmc_core/slp_s0_residency_usec
98926400
```

```
~# cat /sys/kernel/debug/pmc_core/mphy_core_lanes_power_gating_status
MPHY CORE LANE 0                	State: Power gated
MPHY CORE LANE 1                	State: Power gated
MPHY CORE LANE 2                	State: Power gated
MPHY CORE LANE 3                	State: Power gated
MPHY CORE LANE 4                	State: Power gated
MPHY CORE LANE 5                	State: Power gated
MPHY CORE LANE 6                	State: Power gated
MPHY CORE LANE 7                	State: Power gated
MPHY CORE LANE 8                	State: Power gated
MPHY CORE LANE 9                	State: Power gated
MPHY CORE LANE 10               	State: Power gated
MPHY CORE LANE 11               	State: Power gated
MPHY CORE LANE 12               	State: Power gated
MPHY CORE LANE 13               	State: Power gated
MPHY CORE LANE 14               	State: Power gated
MPHY CORE LANE 15               	State: Power gated
```
One thing I would like to double check - is that right that I doesn't see tlp in ps output? It's shouldn't be there, I guess?

Also, if S0ix really break due to tlp attempt to alter RUNTIME_PM_ON_AC/BAT what possible ways of resolving this for new users who will use default configuration? I guess as Intel get more S0ix fixes into kernel (as they do in latest releases) eventually it could became more critical issues as devices that do S0ix instead of S3 by default (or doesn't support S3 at all, like Dell Venue 11 Pro 7140) is already widespread enough.

RussianNeuroMancer, 2019-04-03:
Upon additional re-testing of previous build (1.1.909) with enabled RUNTIME_PM_ON_AC and RUNTIME_PM_ON_BAT I find that it still behave correctly (i.e. S0ix works). Yes, even with RUNTIME_PM_ON_AC and RUNTIME_PM_ON_BAT enabled. Last time I probably did mistake and start 1.1.909 test with disabled RUNTIME_PM_ON_AC and RUNTIME_PM_ON_BAT so conclusion in previous post in incorrect. 

To verify that I purged tlp installation and double checked that tlp configuration file is removed. I installed 1.2.900 build from https://download.linrunner.de/packages/ and re-tested S0ix with untouched, default tlp configuration. Result is: all MPHY Core Lanes is power gated, S0ix residency (both CPU and system) is non-zero. So it's seems like something was fixed on tlp, and I just didn't noticed that.

closed 2019-04-03 by RussianNeuroMancer

linrunner, 2019-04-03:
Glad to hear that your hardware performs well now.

There were absolutely no changes regarding power saving since 1.1.909.

RussianNeuroMancer, 2019-04-03:
The thing is, I forgot to test 1.1.909 with enabled RUNTIME_PM_ON_AC and RUNTIME_PM_ON_BAT three weeks ago, when I wrote [this commentary](https://github.com/linrunner/TLP/issues/386#issuecomment-471510300) - I disabled RUNTIME_PM_ON_AC and RUNTIME_PM_ON_BAT right after installation, so I completely missed the fact that 1.1.909  behave correctly with default configuration (enabled RUNTIME_PM_ON_AC and RUNTIME_PM_ON_BAT). Build 1.2.900 testing that I done today was just to confirm that I didn't mistake anywhere this time.

AnterCreeper, 2023-03-24:
@RussianNeuroMancer How does the problem gone? I am facing this problems too.
a x4 pcieport is not power gated, but it is unused.
And it cannot enter D3 during suspend.

RussianNeuroMancer, 2023-03-24:
> a x4 pcieport is not power gated, but it is unused.
> And it cannot enter D3 during suspend.

All of this works without TLP for you?

AnterCreeper, 2023-03-28:
> 
no

AnterCreeper, 2023-03-28:
@RussianNeuroMancer Nope. I have no idea at all. I have seen some similar issues from the bugzilla.kernel.org.
https://bugzilla.kernel.org/show_bug.cgi?id=217223

AnterCreeper, 2023-03-28:
Here it is:
https://bugzilla.kernel.org/show_bug.cgi?id=207893