linrunner/TLP #388: Returning to AC mode doesn't set the correct min freq

virtualviolet, 2019-02-17:
When returning to AC mode, the freq set in the conf file via CPU_SCALING_MIN_FREQ_ON_AC isn't applied correctly.

If I manually run tlp start the freq is applied correctly.

https://gist.github.com/virtualviolet/43fb7816f9359e282d0acd08d43a137b

linrunner, 2019-02-18:
Hi,

according to your outputs CPU_MIN_PERF_ON_AC=100 has a similar problem:

     /sys/devices/system/cpu/intel_pstate/min_perf_pct = 50 [%]

I can only speculate something weird in the kernel blocks the sysfile writes immediately after the change of power source but not at a later time.

A remark about your config: are you aware that the [kernel docs](https://www.kernel.org/doc/Documentation/cpu-freq/intel-pstate.txt) tell not to use min/max perf and scaling min/max freq at the same time with the intel_pstate driver? That's why [TLP's docs](https://linrunner.de/en/tlp/docs/tlp-configuration.html#scaling) state

> CPU_SCALING_MIN_FREQ_ON_AC ... Do not use this setting with the intel_pstate scaling driver, use CPU_MIN/MAX_PERF instead (see below).

Doesn't the performance governor alone achieve your goal of maximum performance on AC?

virtualviolet, 2019-02-19:
ok

I have disabled frequency scaling and just left min/max perf scaling

Still doesn’t return to 100%, is it possible to run a script when entering
bat or ac mode?

*From:* linrunner <notifications@github.com>
*Sent:* Monday, February 18, 2019 9:33 PM
*To:* linrunner/TLP <TLP@noreply.github.com>
*Cc:* Kevin Walker <kwalker@virtualviolet.net>; Author <
author@noreply.github.com>
*Subject:* Re: [linrunner/TLP] Returning to AC mode doesn't set the correct
min freq (#388)

Hi,

according to your outputs CPU_MIN_PERF_ON_AC=100 has a similar problem:

 /sys/devices/system/cpu/intel_pstate/min_perf_pct = 50 [%]

I can only speculate something weird in the kernel blocks the sysfile
writes immediately after the change of power source but not at a later time.

A remark about your config: are you aware that the kernel docs
<https://www.kernel.org/doc/Documentation/cpu-freq/intel-pstate.txt> tell
not to use min/max perf and scaling min/max freq at the same time with the
intel_pstate driver? That's why TLP's docs
<https://linrunner.de/en/tlp/docs/tlp-configuration.html#scaling> state

CPU_SCALING_MIN_FREQ_ON_AC ... Do not use this setting with the
intel_pstate scaling driver, use CPU_MIN/MAX_PERF instead (see below).

Doesn't the performance governor alone achieve your goal of maximum
performance on AC?

—
<https://github.com/linrunner/TLP/issues/388#issuecomment-464821447>, or mute
the thread

linrunner, 2019-02-21:
Please don't answer per mail here (spams you post with a full quote).

No external scripts, sorry.

I need trace data to proceed:

1. Install the [beta packages](https://download.linrunner.de/packages/).
1. [Enable trace mode](https://linrunner.de/en/tlp/docs/tlp-configuration.html#tracemode).
1. Post traces from a BAT to AC transition.

linrunner, 2019-03-05:
Any news on this?

linrunner, 2019-03-21:
Due to lack of response I close here. Feel free to reopen when you've got the necessary information.

closed 2019-03-21 by linrunner