linrunner/TLP #389: Recalibrate works fine till last moment, then gives up

danbst, 2019-03-10:
[x] I've read and accepted the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md)

**Describe the bug**

I have ThinkPad E470. Currently TLP outputs following stats for battery:
```
+++ ThinkPad Battery Status: BAT0 (Main / Internal)
/sys/class/power_supply/BAT0/manufacturer                   = LGC
/sys/class/power_supply/BAT0/model_name                     = 01AV412
/sys/class/power_supply/BAT0/cycle_count                    = (not supported)
/sys/class/power_supply/BAT0/energy_full_design             =  45000 [mWh]
/sys/class/power_supply/BAT0/energy_full                    =   1910 [mWh]
/sys/class/power_supply/BAT0/energy_now                     =   1910 [mWh]
/sys/class/power_supply/BAT0/power_now                      =      0 [mW]
/sys/class/power_supply/BAT0/status                         = Not charging

Charge                                                      =  100.0 [%]
Capacity                                                    =    4.2 [%]
```
Looks like battery is dead? Nope, it's just `energy_full` isn't correct. I assume `recalibrate` should fix this issue, however it doesn't work for me.

So, I do `sudo tlp recalibrate`, wait till full discharge, then `tlp` exits and battery starts charging. I do periodic dumps of stats:
```
/sys/class/power_supply/BAT0/energy_full_design             =  45000 [mWh]
/sys/class/power_supply/BAT0/energy_full                    =   1920 [mWh]
/sys/class/power_supply/BAT0/energy_now                     =    540 [mWh]
/sys/class/power_supply/BAT0/power_now                      =  55155 [mW]
/sys/class/power_supply/BAT0/status                         = Charging

tpacpi-bat.BAT0.startThreshold                              =     95 [%]
tpacpi-bat.BAT0.stopThreshold                               =    100 [%]
tpacpi-bat.BAT0.forceDischarge                              =      0

Charge                                                      =   28.1 [%]
Capacity                                                    =    4.3 [%]
```
good
```
/sys/class/power_supply/BAT0/energy_full_design             =  45000 [mWh]
/sys/class/power_supply/BAT0/energy_full                    =   1920 [mWh]
/sys/class/power_supply/BAT0/energy_now                     =   2310 [mWh]
/sys/class/power_supply/BAT0/power_now                      =  56032 [mW]
/sys/class/power_supply/BAT0/status                         = Charging

Charge                                                      =  120.3 [%]
Capacity                                                    =    4.3 [%]
```
120% charge, nice
```
/sys/class/power_supply/BAT0/energy_full_design             =  45000 [mWh]
/sys/class/power_supply/BAT0/energy_full                    =   1920 [mWh]
/sys/class/power_supply/BAT0/energy_now                     =  19790 [mWh]
/sys/class/power_supply/BAT0/power_now                      =  22948 [mW]
/sys/class/power_supply/BAT0/status                         = Charging

Charge                                                      = 1030.7 [%]
Capacity                                                    =    4.3 [%]
```
1000% charge, awesome! Looks like battery is not dead (this can also be confirmed by the time for full discharge - took several hours)
```
/sys/class/power_supply/BAT0/energy_full_design             =  45000 [mWh]                                                                                              
/sys/class/power_supply/BAT0/energy_full                    =   1920 [mWh]                                                                                              
/sys/class/power_supply/BAT0/energy_now                     =  31620 [mWh]                                                                                              
/sys/class/power_supply/BAT0/power_now                      =   2452 [mW]                                                                                               
/sys/class/power_supply/BAT0/status                         = Charging                                                                                                  

Charge                                                      = 1646.9 [%]                                                                                                
Capacity                                                    =    4.3 [%]                                                                                                
```
This is second to last measurement. The 32000 mWh is probably the cap of this battery. But when `BAT0/power_now` enters zero, things go mad again:
```
/sys/class/power_supply/BAT0/energy_full_design             =  45000 [mWh]
/sys/class/power_supply/BAT0/energy_full                    =   1910 [mWh]
/sys/class/power_supply/BAT0/energy_now                     =   1910 [mWh]
/sys/class/power_supply/BAT0/power_now                      =      0 [mW]
/sys/class/power_supply/BAT0/status                         = Not charging

Charge                                                      =  100.0 [%]
Capacity                                                    =    4.2 [%]
```

**Expected behavior**
I expect recalibration to set `energy_full` to max `energy_now` reached during charge. I also didn't expect `tlp recalibrate` to exit after full discharge - recalibration isn't complete, so don't exit untill full charge.

**To Reproduce**

This started to happen after I've set start/stop thresholds to 40%/50%. But I can't say for sure this is related, because I also have broken charging port, which may also influence things, but I'm not sure here as well. So I don't know how to reproduce. Full `tlp stat` - https://gist.github.com/danbst/9b478b50b5ab9b6338809261e8358103

My AC was plugged-in all the time.

linrunner, 2019-03-12:
Hi,

interesting symptoms. This is most likely a hardware issue e.g. broken battery pack (the integrated controller part). There's nothing TLP can do to fix it.

> I also didn't expect tlp recalibrate to exit after full discharge
This is a design decision (btw documented in the [manual](https://localhost/linrunner/tlp/docs/tlp-linux-advanced-power-management.html#recalibrate)). Some users (including me) may decide to power off the machine while charging.

closed 2019-03-12 by linrunner

marcfehling, 2024-02-17:
I experience the exact same issue with my Lenovo E470 device. Things seemed fine for years before I started using tlp a few days ago.

- When a `charge_control_end_threshold` is reached (I set it to 60% and 80%), the battery charge jumps to 100%.
- Recalibration does not seem to reliably adjust `energy_full`. In combination with thresholds, I suspect that my `energy_full` value at some point took the current charge at a 60% threshold, effectively almost halving the current capacity. At another attempt the previous capacity was restored, now it's down again. I am confused.
- Because of the incorrect capacity, I can charge beyond `energy_full` (showing charges of 150% and more). Once the battery is fully charged, the capacity and `energy_full` readings jump back to 100%.

I wonder if a firmware update of the BIOS/UEFI might help? They should affect the EC as well, correct?

Any help or hint is welcome! My [tlp stat](https://gist.github.com/marcfehling/b102db87afa5e68a98eb1fa82a678064).
@linrunner -- Have you seen these symptoms with other users?
@danbst -- Have you solved the situation in the meantime?

burner1024, 2024-02-18:
Same for me, and no, firmware updates don't help.

linrunner, 2024-02-19:
@marcfehling @burner1024 A 150% energy_full/energy_full_design quotient indicates faulty battery electronics, which can only be resolved by replacing the battery.

Please note that hardware issues are not within the scope of the TLP issue tracker.