linrunner/TLP #391: `Error: unknown command "chargeonce"`

beuerle, 2019-03-14:
**Describe the bug**

After the most recent upgrade to TLP 1.2, the `chargeonce` (sub)command is missing on my system:

```bash
$ sudo tlp chargeonce
Error: unknown command "chargeonce".
Usage: tlp start|true|bat|false|ac|usb|bayoff|discharge|setcharge|fullcharge|recalibrate|diskid
```
The `fullcharge` command works as expected and loads the battery:

```bash
$ sudo tlp fullcharge
Setting temporary charge thresholds for BAT0:
  stop  = 100
  start =  96
Charging starts now, keep AC connected.
```

System information:

```bash
$ sudo dmidecode
[..]
System Information
	Manufacturer: LENOVO
	Product Name: 20A7005KMS
	Version: ThinkPad X1 Carbon 2nd
	Wake-up Type: Power Switch
	SKU Number: LENOVO_MT_20A7_BU_Think_FM_ThinkPad X1 Carbon 2nd
	Family: ThinkPad X1 Carbon 2nd
[..]
Base Board Information
	Manufacturer: LENOVO
	Product Name: 20A7005KMS
	Version: 0B98417 WIN
	Asset Tag: Not Available
	Features:
		Board is a hosting board
		Board is replaceable
	Location In Chassis: Not Available
	Chassis Handle: 0x0000
	Type: Motherboard
	Contained Object Handles: 0
[..]
```

```bash
$ uname -a
Linux snowcrash 5.0.1-arch1-1-ARCH #1 SMP PREEMPT Sun Mar 10 10:55:53 UTC 2019 x86_64 GNU/Linux
$ pacman -Ss tlp
community/tlp 1.2-1 [installed]
    Linux Advanced Power Management
```
**Expected behavior**
According to the man page, the command should still be present:

```man
chargeonce [BAT0|BAT1] (ThinkPads only)
              Charge main (BAT0) or auxiliary (BAT1) battery to the stop charge
              threshold once (bypassing start threshold).  Configured thresholds
              are restored upon next system boot-up.
```

linrunner, 2019-03-15:
Confirmed. I'll look into that soon.

linrunner, 2019-03-15:
Thanks for your report. I fixed the issue. 

Could you please test with the master branch? There are packages in the AUR:
*  https://aur.archlinux.org/packages/tlp-git/
* https://aur.archlinux.org/packages/tlp-rdw-git/

beuerle, 2019-03-18:
Yes, working again!

Thanks for the quick fix!

closed 2019-03-18 by beuerle

linrunner, 2019-03-18:
Thanks for your feedback. Fix released with version 1.2.1.