linrunner/TLP #393: Support for laptops running the Lenovo Vantage Software with available battery conservation mode

NicolasSchuler, 2019-03-18:
Lately I'm switched from Windows to Manjaro (Linux) with my Lenovo Yoga 720. In Windows the laptop was running the Lenovo Vantage Software, with this I could turn on the "battery conservation mode" on my Laptop. Now I've searched kind a like through the whole internet trying to find a solution for Linux. The closest thing also related to Lenovo is tlp. In the documentation it is well mentioned that this feature is only available on thinkpads, but nervertheless I've tried it out, with no success.
It would be great if tlp would be able to enable the battery conservation mode on any Lenovo Laptop who can run the Lenovo Vantage Software with battery conservation mode option available.

linrunner, 2019-03-18:
Thanks for your suggestion. This is not possible without a kernel driver, see [FAQ](https://linrunner.de/en/tlp/docs/tlp-faq.html#notathinkpad).

whenov, 2021-03-20:
It is possible with the `ideapad_laptop` driver now. On my laptop (Lenovo Legion 5), I can turn on the battery conservation mode by `echo 1 > /sys/bus/platform/drivers/ideapad_acpi/VPC2004:00/conservation_mode`.

linrunner, 2021-03-20:
@whenov : thanks. Can you add the full output of 

     tlp-stat 

via https://gist.github.com/ please?

whenov, 2021-03-20:
@linrunner Sure, here it is: https://gist.github.com/whenov/72069cfc2bcd8b3e76a67e9035b9ec68

linrunner, 2021-03-21:
I would expect the file under **/sys/devices/platform/ideapad_acpi/** also. May i see

    ls -lR /sys/devices/platform/ideapad_acpi

(also via https://gist.github.com/)

whenov, 2021-03-22:
@linrunner It appears I don't have this directory: https://gist.github.com/whenov/a4c783e930bea6b8c65459c7167221a1

linrunner, 2021-04-20:
The wait is over: the stop charge threshold is implemented for 1.4. 

Please participate in the alpha test. All necessary information is [here](https://gist.github.com/linrunner/2ead4b591eed33055cf86a38ccc73949).

Have fun.

closed 2021-09-15 by linrunner