linrunner/TLP #394: Thinkpad T480 kernel 4.18 tlp 1.2 via ppa Error: cannot set threshold for BAT1 (BAT0 works)

timrichardson, 2019-03-19:
Ubuntu 18.04.2,
tim@moncrief:~$ uname -r
4.18.0-16-generic
Thinkpad T480

tlp has worked well for a year.

tlp installed via ppa

Symptom: 
```
sudo tlp setcharge 92 95 BAT1
Setting temporary charge thresholds for BAT1:
  stop  =  95 (Error: cannot set threshold)
  start =  91 (Error: cannot set threshold)

```

but this works (BAT0):

```
tim@moncrief:~$ sudo tlp setcharge 92 95 BAT0
Setting temporary charge thresholds for BAT0:
  stop  =  95
  start =  91
tim@moncrief:~$ 
```

`sudo tlp-stat -b
--- TLP 1.2.1 --------------------------------------------

+++ Battery Features
natacpi    = active (data, thresholds)
tpacpi-bat = active (discharge)
tp-smapi   = inactive (ThinkPad not supported)

+++ ThinkPad Battery Status: BAT0 (Main / Internal)
/sys/class/power_supply/BAT0/manufacturer                   = SANYO
/sys/class/power_supply/BAT0/model_name                     = 01AV419
/sys/class/power_supply/BAT0/cycle_count                    =     51
/sys/class/power_supply/BAT0/energy_full_design             =  24020 [mWh]
/sys/class/power_supply/BAT0/energy_full                    =  23430 [mWh]
/sys/class/power_supply/BAT0/energy_now                     =  18130 [mWh]
/sys/class/power_supply/BAT0/power_now                      =      0 [mW]
/sys/class/power_supply/BAT0/status                         = Unknown (threshold effective)

/sys/class/power_supply/BAT0/charge_start_threshold         =     75 [%]
/sys/class/power_supply/BAT0/charge_stop_threshold          =     80 [%]
tpacpi-bat.BAT0.forceDischarge                              =      0

Charge                                                      =   77.4 [%]
Capacity                                                    =   97.5 [%]

+++ ThinkPad Battery Status: BAT1 (Ultrabay / Slice / Replaceable)
/sys/class/power_supply/BAT1/manufacturer                   = LGC
/sys/class/power_supply/BAT1/model_name                     = 01AV492
/sys/class/power_supply/BAT1/cycle_count                    =     11
/sys/class/power_supply/BAT1/energy_full_design             =  71100 [mWh]
/sys/class/power_supply/BAT1/energy_full                    =  70940 [mWh]
/sys/class/power_supply/BAT1/energy_now                     =  59940 [mWh]
/sys/class/power_supply/BAT1/power_now                      =      0 [mW]
/sys/class/power_supply/BAT1/status                         = Unknown (threshold effective)

/sys/class/power_supply/BAT1/charge_start_threshold         =     80 [%]
/sys/class/power_supply/BAT1/charge_stop_threshold          =     85 [%]
tpacpi-bat.BAT1.forceDischarge                              =      0

Charge                                                      =   84.5 [%]
Capacity                                                    =   99.8 [%]

+++ Charge total                                            =   82.7 [%]
`

timrichardson, 2019-03-19:
`root@moncrief:/sys/class/power_supply/BAT1# echo 90 > charge_start_threshold 
bash: echo: write error: No such device
root@moncrief:/sys/class/power_supply/BAT1# 
`

timrichardson, 2019-03-19:
works in Ubuntu mainline kernel 5.0.2-050002-generic

linrunner, 2019-03-19:
Hi Tim, 

Kernel 4.18.0-17 (currently in Proposed) will fix this too. See [LP #1812099](https://bugs.launchpad.net/ubuntu/+source/linux/+bug/1812099). 

I keep this open as a notice.

linrunner, 2019-04-04:
Kernel 4.18.0-17 is available in regular updates now.

closed 2019-04-04 by linrunner

adams-family, 2021-09-29:
Interesting, I got the same problem on `4.19.0-8-amd64` (Debian GNU/Linux 10)

Examples:
```
$ echo 80 > /sys/class/power_supply/BAT1/charge_stop_threshold
-bash: echo: write error: No such device

$ cat /sys/class/power_supply/BAT1/charge_stop_threshold
cat: /sys/class/power_supply/BAT1/charge_stop_threshold: No such device
```

Anyone's got the same issue?