linrunner/TLP #395: Recalibration: 'Error: discharge BAT0 malfunction'

LinuxOnTheDesktop, 2019-03-19:
[x] I've read and accepted the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md)

**Describe the bug**

$  sudo tlp recalibrate
Setting temporary charge thresholds for BAT0:
  stop  = 100
  start =  96
Error: discharge BAT0 malfunction.

**Expected behavior**

Reclibration, with no error messages.

**To Reproduce**

Steps to reproduce the unexpected behavior:

1. Does the problem occur on battery or AC or both?

AC; the relevant command is not intended to run on battery power.

2. Actions to reproduce the behaviour

`sudo tlp recalibrate`

3. Shell commands entered and their output

$  sudo tlp recalibrate
Setting temporary charge thresholds for BAT0:
  stop  = 100
  start =  96
Error: discharge BAT0 malfunction.
4. **Full output of `tlp-stat` via https://gist.github.com/ for *all* cases of 1**

Instruction seems ambiguous. But here is [output for ac](https://gist.github.com/LinuxOnTheDesktop/e20e65685573760d2b3a58e845a99e08) and [here is output for battery](https://gist.github.com/LinuxOnTheDesktop/4954fa6d6c271220bdb6ddf24b2daacf).

**Additional context**

Kernel: 5.0.2
OS: Linux Mint 19.1 x64
DE: Cinnamon
ThinkPad X1CG6

linrunner, 2019-03-19:
Hi, 

my X1C6 (current development machine for TLP) shows the same error – welcome to the club!

It occurs with tpacpi-bat and natacpi (kernel with [patches for force_discharge](https://github.com/linrunner/TLP/issues/321)).  I guess the ACPI call 'BDSS' used for force_discharge was changed by Lenovo.

LinuxOnTheDesktop, 2019-03-19:
Much as I detest Lenovo for its near total lack of support for desktop Linux, I do find that, if I use tlp 1.1-2 (an older version), then the calibration works.

Also: may I ask whether you have the touchpad-stops-working-after-S3-sleep problem? For, tlp is one of the many ingredients in that problem? (And yes I do have the latest firmware for everything and a recent kernel and I have tried various - many - workarounds.)

linrunner, 2019-03-19:
>  I do find that, if I use tlp 1.1-2 (an older version), then the calibration works.
To my *considerable* surprise: you're right. I'll look into that. Maybe a timing issue ...

linrunner, 2019-03-19:
Well, i think i fixed it. See the above commit for details.

Please test – packages [here](https://download.linrunner.de/packages/).

LinuxOnTheDesktop, 2019-03-19:
That seems to work: recalibration functions again. I did have trouble installing the .debs, though, probably at least partly because of bugginess in the `gdebi` installer.

linrunner, 2019-03-19:
I keep this open as a notice.

linrunner, 2019-05-04:
Released with 1.2.2

closed 2019-05-04 by linrunner

ablakely, 2019-05-27:
I'm having this same issue on my w520

frauhottelmann, 2019-05-27:
I do have the same problem on my X240 with tlp 1.2.2:
[AC](https://gist.github.com/frauhottelmann/5501af4f1e2b28acd33275babbdcc88c) | [battery](https://gist.github.com/frauhottelmann/6f87867d094082b8e75b43ff62e2456b)

linrunner, 2019-05-28:
@frauhottelmann : different hardware = different problem. Please open your own issue. I also need the exact command that failed and all of it's output.