linrunner/TLP #398: Split config files if set, config.d

Thaodan, 2019-03-20:
**Describe the solution you'd like**
Please support splitting up the config/multiple config files and allow using like with something like
/etc/tlp.d
A clear and concise description of:

* Your use case(s)
keep distro defaults and allow user edits without manual merges after updates.
* What you want to happen
source configs from something like /etc/tlp.d

linrunner, 2019-03-21:
In fact TLP was written, among other things, because I don't like a lot of tiny configuration files. I also don't want users to have to look at several files to discover the actual settings.

Thaodan, 2019-03-21:
Can you provide it was an option if the users wants it?

dywisor, 2019-03-22:
Related: #335 

The configuration file is a regular shell script, you could add some glue code that loads your files from ``/etc/tlp.d``.

This is, however, strongly discouraged since the [config dump](https://github.com/linrunner/TLP/blob/903d7fa67d8da61af85d59e9bd5eac371d366dc1/tlp-stat.in#L231-L232) in your tlp-stat output will always be incomplete, which might disqualify you from getting support here.

linrunner, 2019-03-22:
Some more contras:
- Not only tlp-stat -c – it would make reading configuration unnecessarily complicated and prone to failures.
- It would complicate testing.
- It would also break TLPUI.
- It would separate parameter descriptions from parameter values.

Btw: you did not really describe your use case. Why exactly do you want to separate defaults from your settings?

koponomarenko, 2019-05-20:
I was about to ask this same question about config files.
Would be very handy to be able to use drop-in configuration files.
The default config file should be read first, then any drop-in configuration files in `/etc/tlp.d`
In case of parameter overlapping the last parameter overwrites a previous value.
This will eliminate a headache of merging configuration files in case of updates.

Thaodan, 2019-05-21:
That was my exact intend. Without it manning config changes in automated setups is impossible.
Also system defaults should go too /usr/lib/pkg/conf and be overriden by the user config that is loaded afterwards.

dywisor, 2019-05-22:
Random thoughts on this topic that'd help me to you understand your use case(s) more properly:

* *Some* distros ship a tool for merging updated files in /etc, e.g. ``etc-update`` in Gentoo -- wouldn't that be an option?

* If you wanted to go the ``/etc`` automation route, wouldn't a puppet or ansible module be much more appropriate?

koponomarenko, 2019-05-22:
I appreciate the workarounds you mentioned. I am also aware of all listed options.
My intention was to suggest a better and more flexible way of handling configuration files for `tlp`.

Personally I don't like merging configs when it is possible to eliminate this effort.
This is where "drop-in" configs come in handy.

BTW, if I am not mistaken `/etc/default` is not a standard Linux directory. It is used by Debian derivatives only.

I would suggest to use `/etc/tlp/tlp.conf` for the default configuration (updates automatically without any problems) and `/etc/tlp/conf.d` for system specific configuration "drop-in" file(s) (those are not updated automatically, but created by a user/admin). Any parameter defined in a "drop-in" config file overwrites a corresponding parameter defined in the default config. This approach will ease system setup and maintenance.

This seems to be a common best practice in Linux.
As an example: `NetworkManager`, `sudo`, `systemd`, `fontconfig`, etc.

"XDG Base Directory Specification" describes another alternative for where the default configs should be placed and how to handle different config files if there are more than one : https://specifications.freedesktop.org/basedir-spec/basedir-spec-latest.html#referencing

dywisor, 2019-05-22:
> I appreciate the workarounds you mentioned. I am also aware of all listed options.

These are not workarounds but valid solutions depending on your requirements on which you did not elaborate sufficiently, hence my questions.

> My intention was to suggest a better and more flexible way of handling configuration files for `tlp`.
>
> Personally I don't like merging configs when it is possible to eliminate this effort.
> This is [...]

This is correct, but the directory is not entirely wrong, either.

> I would suggest to use `/etc/tlp/tlp.conf` for the default configuration (updates automatically without any problems) and `/etc/tlp/conf.d` (or `/etc/tlp/tlp.conf.d`) for system specific configuration "drop-in" file(s) (those are not updated automatically, but created by a user/admin). Any parameter defined in a "drop-in" config file overwrites a corresponding parameter defined in the default config. This approach will ease system setup and maintenance.
> 

Whether ``/etc/default/tlp`` is *the* proper location for the configuration file is a wholly distinct issue, yet related since it should be taken care of before implementing split config.
Keeping compability with existing installations and content all over the web are strong points against changing this path.

When I started packaging TLP, I opted for my distro's (or rather, init system's) equivalent of ``/etc/default``, ``/etc/conf.d``. When systemd became more and more relevant, a user suggested changing the path which I agreed on since I considered it to be a *technical* issue - init system *A* should not reuse *B's* configuration (dywisor/tlp-portage#4).
The transition worked well, probably because of not-so-mainstream distro => few users, which does not apply to TLP as a whole.

I'm not really keen on going through this again. So, please, if anything, use ``/etc/tlp.conf`` plus ``/etc/tlp.conf.d``, similar to e.g. sudo.

> "XDG Base Directory Specification" describes another alternative for where the default configs should be placed and how to handle different config files if there are more than one : https://specifications.freedesktop.org/basedir-spec/basedir-spec-latest.html#referencing

XDG is mostly relevant when you have the notion of per-user configuration and/or your program's lifecycle is desktop-session alike, neither of which applies to TLP.

For example
```
$XDG_RUNTIME_DIR [...]
The lifetime of the directory MUST be bound to the user being logged in.
It MUST be created when the user first logs in
and if the user fully logs out the directory MUST be removed.
```

TLP does not start a daemon and needs to preserve runtime data in between. You might argue that script start/stop doesn't count as login/logout, but I consider this as a clear violation of above requirement.

If anything is to follow, then it's probably the [Debian policy](https://www.debian.org/doc/debian-policy/index.html) which is basically FHS plus a few extensions.

---

What I get is that you care about:
- managing configuration for a few machines (or just a single one)
- correctness as in establishing select standards

I'm not strongly against this proposal, but I'm not really convinced that the benefits outweigh the effort.

I will not spend time on implementing this myself and neither won't @linrunner, I suppose.  A pull request might help; I may be available for a review, but I can only speculate on the chances of acceptance. I **do** care about not breaking existing use cases, so a proper migration path including concerns listed in the comments above is a must.

Thaodan, 2019-05-25:
/etc/tlp.conf /etc/tlp.conf.d sounds great.
If pkg upstream config should be seperate it should be inside the pkg in /usr/
lib.
Makes editing the package config clearer.

linrunner, 2019-05-26:
> This approach will ease system setup and maintenance.

1. The current approach achieves the same goal: *only one* file in which the average user will find all the necessary information in *one* place. That you have to go further for automation is not a contradiction imho.

> This seems to be a common best practice in Linux.
> As an example: NetworkManager, sudo, systemd, fontconfig, etc.

2. Your examples illustrate what the conf.d schema was originally designed for: to allow multiple applications, e.g packages a, b, c, d, to contribute configuration elements for a central service, e.g package e, without getting in each other's way. TLP's configuration is for one application only.

> BTW, if I am not mistaken /etc/default is not a standard Linux directory. It is used by Debian derivatives only.

3. Right. But /etc/default/ is historically developed and was right at the time and context. Plus there are few advantages compared to the effort and risks involved in a migration.

> If pkg upstream config should be seperate it should be inside the pkg in /usr/lib.

4. Objection: FHS says /usr/lib is "for object files and binaries". /usr/share would be right ("read-only architecture independent data files")

Further counterarguments:

5. Config merges are not binding in any way. TLP is always backwards compatible for older configs (see [FAQ](https://linrunner.de/en/tlp/docs/tlp-faq.html#install-config) *"Do i need to edit my settings after updating TLP?"*). But even with a conf.d scheme you would have to check for new features on every update if you want them.

6. TLP has intrinsic i.e. hard coded defaults, documented in /etc/default/tlp. So you may as well run TLP with a file containing only one line `TLP_ENABLE=1`. You may add lines for the features you want to use. All the comments are redundant. You can keep this file forever or change it when it suits you. This is not a "workaround" but a supported approach.

Thaodan, 2019-05-26:
Am Sonntag, 26. Mai 2019, 18:01:57 CEST schrieb linrunner:
> > This approach will ease system setup and maintenance.
> 
> 1. The current approach achieves the same goal: *only one* file in which the
> average user will find all the necessary information in *one* place. That
> you have to go further for automation is not a contradiction imho.
But that one place is where defaults and user config is stored.
When the user overrides the default he always has to merge is his configs 
(.pacnew on arch on others different).
Putting the default in a different hierarchy prevents this.
> 2. Your examples illustrate what the conf.d schema was originally designed
> for: to allow multiple applications, e.g packages a, b, c, d, to contribute
> configuration elements for a central service, [...]
But the user may package configs and use dropins for specific config changes,
> 3. Right. But /etc/default/ is historically developed and was right at the
> time and context. Plus there are few advantages compared to the effort and
> risks involved in a migration.
> 4. Objection: FHS [...]
/usr/lib is the usual place for those style of configs many kind of script 
style files are in that place now (eg. Qt qml, python etc.).
I just cite existing uses.
> Further counterarguments:
>
> 5. Config merges are not binding in any way. TLP is always backwards
> compatible for older configs (see
> [FAQ](https://linrunner.de/en/tlp/docs/tlp-faq.html#install-config) [...]
Only if he overrides them. If you load the default first and than the user 
config everything is fine.

linrunner, 2019-10-03:
@Thaodan , @koponomarenko , @dywisor : what do you think about my proposal [here](https://github.com/linrunner/TLP/pull/435)?

linrunner, 2019-10-03:
The outcome will be to read config files in the following order:

1. **/usr/share/tlp/defaults.conf** – will contain all TLP defaults
2. **/etc/tlp.conf.d/*** – customizations (implemented by separate packages)
3. **/etc/tlp.conf** –  corresponds to today's **default** file, but all settings commented; user must uncomment his changes
4. **/etc/default/tlp** – maybe (?)  for backward compatibility

The effective settings will be written to **/run/tlp/tlp.conf** and shown by `tlp-stat -c`.

This would result in the precedence:
1. user change
2. tlp.conf.d
3. (intrinsic) defaults

Thaodan, 2019-10-03:
Sounds good great.Except that system/vendor defaults now 
go to /usr/lib these days.

The settings is that the vendor dir contains default configs 
that are by 
10-example.conf and can be overriden or deactivated by 
adding 10-example.conf to /etc/app.d/ or /run/app.d/

Look at tmpfiles.d for example.

linrunner, 2020-01-17:
Please test with 1.3.0-beta.3 and report.
Instructions: https://github.com/linrunner/TLP/issues/449

closed 2020-01-17 by linrunner