linrunner/TLP #401: charge_stop_threshold can not be set to 100

lawa42, 2019-04-02:
[x] I've read and accepted the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md)

**Describe the bug**

A clear and concise description of what the bug is.

**Expected behavior**

A clear and concise description of what you expected to happen.

**To Reproduce**

Steps to reproduce the unexpected behavior:

1. Does the problem occur on battery or AC or both?
2. Actions to reproduce the behaviour
3. Shell commands entered and their output
4. **Full output of `tlp-stat` via https://gist.github.com/ for *all* cases of 1**

**Additional context**

Add any other context about the problem here.

linrunner, 2019-04-02:
Please edit your post and provide the required info.

lawa42, 2019-04-02:
This is on arch linux:
Linux it72 5.0.5-arch1-1-ARCH #1 SMP PREEMPT Wed Mar 27 17:53:10 UTC 2019 x86_64 GNU/Linux
community/acpi_call-dkms 1.1.0-202

linrunner, 2019-04-02:
Known firmware issue (my own X1C6 is affected) --> see [FAQ](https://linrunner.de/en/tlp/docs/tlp-faq.html#stop-threshold-sticks) for a workaround..

linrunner, 2019-05-13:
@lawa42 : on the X1C6 this is fixed with BIOS 1.37. Did you recheck with the recently published 1.31 for your T480s?

closed 2019-05-28 by linrunner

lawa42, 2019-05-28:
After updating also the EC to 1.13 on the 480s it now works.

Thanks