linrunner/TLP #402: [X1C6] setcharge is OK but fullcharge don't work using dev-1.2

xiaoping378, 2019-04-03:
follow #321 

my case:  when stop_threshold is 100 will be failed.

--------

1. step-1, current status.
```bash
➜  ~ sudo tlp-stat -b 
--- TLP 1.2.900 --------------------------------------------

+++ Battery Features
natacpi    = active (data, thresholds)
tpacpi-bat = active (discharge)
tp-smapi   = inactive (ThinkPad not supported)

+++ ThinkPad Battery Status: BAT0 (Main / Internal)
/sys/class/power_supply/BAT0/manufacturer                   = LGC
/sys/class/power_supply/BAT0/model_name                     = 01AV494
/sys/class/power_supply/BAT0/cycle_count                    =     66
/sys/class/power_supply/BAT0/energy_full_design             =  57000 [mWh]
/sys/class/power_supply/BAT0/energy_full                    =  57750 [mWh]
/sys/class/power_supply/BAT0/energy_now                     =  50940 [mWh]
/sys/class/power_supply/BAT0/power_now                      =      0 [mW]
/sys/class/power_supply/BAT0/status                         = Unknown (threshold effective)

/sys/class/power_supply/BAT0/charge_start_threshold         =     75 [%]
/sys/class/power_supply/BAT0/charge_stop_threshold          =     80 [%]
tpacpi-bat.BAT0.forceDischarge                              =      0

Charge                                                      =   88.2 [%]
Capacity                                                    =  101.3 [%]
```

2. step-2, setcharge 90 99 --- **OK**
```bash
➜  ~ sudo tlp setcharge 90 99          
Setting temporary charge thresholds for BAT0:
  stop  =  99
  start =  90
➜  ~ sudo tlp-stat -b | grep _threshold
/sys/class/power_supply/BAT0/charge_start_threshold         =     90 [%]
/sys/class/power_supply/BAT0/charge_stop_threshold          =     99 [%]
```

3. step-3, setcharge 96 100 --  stop_threshold **NOK**
```bash
➜  ~ sudo tlp setcharge 96 100         
Setting temporary charge thresholds for BAT0:
  stop  = 100
  start =  96
➜  ~ sudo tlp-stat -b | grep _threshold
/sys/class/power_supply/BAT0/charge_start_threshold         =     96 [%]
/sys/class/power_supply/BAT0/charge_stop_threshold          =     99 [%]
```
4. step-4, restore to the default 75 80 -- **OK**
```bash
➜  ~ sudo tlp setcharge 75 80          
Setting temporary charge thresholds for BAT0:
  start =  75
  stop  =  80
➜  ~ sudo tlp-stat -b
--- TLP 1.2.900 --------------------------------------------

+++ Battery Features
natacpi    = active (data, thresholds)
tpacpi-bat = active (discharge)
tp-smapi   = inactive (ThinkPad not supported)

+++ ThinkPad Battery Status: BAT0 (Main / Internal)
/sys/class/power_supply/BAT0/manufacturer                   = LGC
/sys/class/power_supply/BAT0/model_name                     = 01AV494
/sys/class/power_supply/BAT0/cycle_count                    =     66
/sys/class/power_supply/BAT0/energy_full_design             =  57000 [mWh]
/sys/class/power_supply/BAT0/energy_full                    =  57750 [mWh]
/sys/class/power_supply/BAT0/energy_now                     =  52240 [mWh]
/sys/class/power_supply/BAT0/power_now                      =      0 [mW]
/sys/class/power_supply/BAT0/status                         = Unknown (threshold effective)

/sys/class/power_supply/BAT0/charge_start_threshold         =     75 [%]
/sys/class/power_supply/BAT0/charge_stop_threshold          =     80 [%]
tpacpi-bat.BAT0.forceDischarge                              =      0

Charge                                                      =   90.5 [%]
Capacity                                                    =  101.3 [%]
```

-----
the other found:
fullcharge == setcharge 96 100,  but set new_stop=100 is NOK, and

if new_start > old_stop,  both(new_stop, new_start) will be NOK,  like fullcharge.

```bash
➜  ~ sudo tlp fullcharge
Setting temporary charge thresholds for BAT0:
  stop  = 100
  start =  96
Charging starts now, keep AC connected.
➜  ~ sudo tlp-stat -b | grep _threshold
/sys/class/power_supply/BAT0/charge_start_threshold         =     75 [%]
/sys/class/power_supply/BAT0/charge_stop_threshold          =     80 [%]

```

linrunner, 2019-04-03:
Thanks for testing. This is a duplicate of #401 , see [FAQ](https://linrunner.de/en/tlp/docs/tlp-faq.html#stop-threshold-sticks).

> Cause: after applying a stop threshold value < 100, Lenovo's system firmware will not accept values higher than the previously set value.

You found a new aspect however: the second `tlp setcharge 77 82` goes through after the first one failed. This contradicts my above statement. 

Can you go back to 100 in small increments?

xiaoping378, 2019-04-03:
Can't go back to 100 in small increments:
```bash
➜  ~ for ((i=80; i<=100; i+=2))
sudo tlp setcharge 75 $i && sleep 1 && sudo tlp-stat -b | grep stop_threshold
Setting temporary charge thresholds for BAT0:
  start =  75 (no change)
  stop  =  80 (no change)
/sys/class/power_supply/BAT0/charge_stop_threshold          =     80 [%]
Setting temporary charge thresholds for BAT0:
  start =  75 (no change)
  stop  =  82
/sys/class/power_supply/BAT0/charge_stop_threshold          =     82 [%]
Setting temporary charge thresholds for BAT0:
  start =  75 (no change)
  stop  =  84
/sys/class/power_supply/BAT0/charge_stop_threshold          =     84 [%]
Setting temporary charge thresholds for BAT0:
  start =  75 (no change)
  stop  =  86
/sys/class/power_supply/BAT0/charge_stop_threshold          =     86 [%]
Setting temporary charge thresholds for BAT0:
  start =  75 (no change)
  stop  =  88
/sys/class/power_supply/BAT0/charge_stop_threshold          =     88 [%]
Setting temporary charge thresholds for BAT0:
  start =  75 (no change)
  stop  =  90
/sys/class/power_supply/BAT0/charge_stop_threshold          =     90 [%]
Setting temporary charge thresholds for BAT0:
  start =  75 (no change)
  stop  =  92
/sys/class/power_supply/BAT0/charge_stop_threshold          =     92 [%]
Setting temporary charge thresholds for BAT0:
  start =  75 (no change)
  stop  =  94
/sys/class/power_supply/BAT0/charge_stop_threshold          =     94 [%]
Setting temporary charge thresholds for BAT0:
  start =  75 (no change)
  stop  =  96
/sys/class/power_supply/BAT0/charge_stop_threshold          =     96 [%]
Setting temporary charge thresholds for BAT0:
  start =  75 (no change)
  stop  =  98
/sys/class/power_supply/BAT0/charge_stop_threshold          =     98 [%]
Setting temporary charge thresholds for BAT0:
  start =  75 (no change)
  stop  = 100
/sys/class/power_supply/BAT0/charge_stop_threshold          =     98 [%]
➜  ~ 
```

linrunner, 2019-04-03:
Only the last step (98 --> 100) fails. Interesting. Did you retry?

Btw: unfortunately your truncated tlp-stat -b outputs hide the actual charging level (SOC). Maybe there is a connection with it?

xiaoping378, 2019-04-08:
retry, failed....
```bash
/sys/class/power_supply/BAT0/status                         = Charging
```

xiaoping378, 2019-05-11:
upgrade X1C6 BIOS version to 1.37, **fullcharage** is worked.

closed 2019-05-12 by xiaoping378

linrunner, 2019-05-13:
Thanks for your feedback. Solution added to the FAQ.

On my X1C6 it works again too. However, i think it might as well have been the [EC firmware 1.12 update](https://fwupd.org/lvfs/device/com.lenovo.ThinkPadN23HT.firmware) published recently. 

I'm not sure when exactly I did my retests. I installed the 1.37 update on 10th of march and didn't look if the 1.12 EC was contained in the 1.37  cab.