linrunner/TLP #404: RESTORE_DEVICE_STATE_ON_STARTUP=1 doesn't work after updade of TLP

pindos74, 2019-04-08:
Hello!
After update of TLP from 1.1 to 1.2.1 I noticed that parameter RESTORE_DEVICE_STATE_ON_STARTUP=1 doesn't work anymore on ThinkPad x270 running LinuxMint Cinnamon 19.1.
I turn off bluetooth pressing hotkey Fn+F8 or disabling it in the panel.
Them I reboot my system, and I see that bluetooth is enabled again.
This thing worked fine on previous version (1.1) of TLP.
This is my system:
--- TLP 1.2.1 --------------------------------------------
+++ Configured Settings: /etc/default/tlp
TLP_ENABLE=1
TLP_DEFAULT_MODE=BAT
TLP_PERSISTENT_DEFAULT=0
DISK_IDLE_SECS_ON_AC=0
DISK_IDLE_SECS_ON_BAT=2
MAX_LOST_WORK_SECS_ON_AC=15
MAX_LOST_WORK_SECS_ON_BAT=60
CPU_HWP_ON_AC=balance_performance
CPU_HWP_ON_BAT=balance_power
SCHED_POWERSAVE_ON_AC=0
SCHED_POWERSAVE_ON_BAT=1
NMI_WATCHDOG=0
ENERGY_PERF_POLICY_ON_AC=performance
ENERGY_PERF_POLICY_ON_BAT=power
DISK_DEVICES="nvme0n1 sda"
DISK_APM_LEVEL_ON_AC="254 254"
DISK_APM_LEVEL_ON_BAT="128 128"
SATA_LINKPWR_ON_AC="med_power_with_dipm max_performance"
SATA_LINKPWR_ON_BAT="med_power_with_dipm min_power"
AHCI_RUNTIME_PM_TIMEOUT=15
RADEON_POWER_PROFILE_ON_AC=default
RADEON_POWER_PROFILE_ON_BAT=default
RADEON_DPM_STATE_ON_AC=performance
RADEON_DPM_STATE_ON_BAT=battery
RADEON_DPM_PERF_LEVEL_ON_AC=auto
RADEON_DPM_PERF_LEVEL_ON_BAT=auto
WIFI_PWR_ON_AC=off
WIFI_PWR_ON_BAT=on
WOL_DISABLE=Y
SOUND_POWER_SAVE_ON_AC=0
SOUND_POWER_SAVE_ON_BAT=1
SOUND_POWER_SAVE_CONTROLLER=Y
BAY_POWEROFF_ON_AC=0
BAY_POWEROFF_ON_BAT=0
BAY_DEVICE="sr0"
RUNTIME_PM_ON_AC=on
RUNTIME_PM_ON_BAT=auto
USB_AUTOSUSPEND=1
USB_BLACKLIST_BTUSB=0
USB_BLACKLIST_PHONE=0
USB_BLACKLIST_PRINTER=1
USB_BLACKLIST_WWAN=0
RESTORE_DEVICE_STATE_ON_STARTUP=1
START_CHARGE_THRESH_BAT0=75
STOP_CHARGE_THRESH_BAT0=80
START_CHARGE_THRESH_BAT1=75
STOP_CHARGE_THRESH_BAT1=80
RESTORE_THRESHOLDS_ON_BAT=1
NATACPI_ENABLE=0
TPACPI_ENABLE=1
TPSMAPI_ENABLE=0

+++ System Info
System         = LENOVO ThinkPad X270 20HN0069RT
BIOS           = R0IET56W (1.34 )
Release        = Linux Mint 19.1 Tessa
Kernel         = 4.15.0-47-generic #50-Ubuntu SMP Wed Mar 13 10:44:52 UTC 2019 x86_64
/proc/cmdline  = BOOT_IMAGE=/boot/vmlinuz-4.15.0-47-generic root=UUID=da7668c0-71c6-4e30-9a2f-8c40cf06ed53 ro quiet splash resume=UUID=817f72cf-430a-4998-8bb5-d98165491892 vt.handoff=7
Init system    = systemd v237
Boot mode      = UEFI

+++ TLP Status
State          = enabled
RDW state      = enabled
Last run       = 22:13:35,     23 sec(s) ago
Mode           = AC
Power source   = AC

linrunner, 2019-04-09:
Hi,

this has already been reported by another user. Please test the fix and report.

[Package download](https://download.linrunner.de/packages/)

pindos74, 2019-04-09:
Thank you very much! 
Your fix work fine! Bluetooth stays disabled after reboot.

closed 2019-04-09 by pindos74