linrunner/TLP #405: [Feature request] DISK_IDLE_SECS_ON description imrpovement

keybreak, 2019-04-11:
**Is your feature request related to a problem? Please describe.**

Currently description of **DISK_IDLE_SECS_ON_** is not really clear:
> Seconds laptop mode has to wait after the disk goes idle before doing a sync.
Non-zero value enables, zero disables laptop mode.

**Describe the solution you'd like**

> Seconds laptop mode has to wait after the disk goes idle before doing a **RAM / cache data sync**.
Non-zero value enables, zero disables laptop mode.

**Additional context**

I had to ask on forum to make sure what exactly it means, to be sure, so figured it would be better to just expand description a little to avoid further questions)

linrunner, 2019-04-12:
Hi, the configuration file is not intended to replace a textbook on operating systems. I have adapted the documentation instead --> https://linrunner.de/en/tlp/docs/tlp-configuration.html#filesystem

closed 2019-04-12 by linrunner

keybreak, 2019-04-12:
@linrunner 
Great, thx! :)

There's typo though:
> from RAM **do** disk again.

I think you mean **to** :wink: