linrunner/TLP #407: Laptop overheating (high CPU load, TLP not

janez33, 2019-04-29:
[x] I've read and accepted the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md)

**Describe the bug**

Hello, I have a new computer (HP G5 i7 laptop)  at my job and I'm using ElementaryOS. 
The problem I have when laptop has a power cable plugged in cpu governor is set to performance (which is causing overheating) so I want to set my cpu governor to be always on powersave but I can't get it to work. Even if TLP is showing that powersave settings are ON, the laptop is on high cpu load (loud fan and high temperature) with power cable plugged in. When I take out power cable the fan gets quite in about 5 min and its working as it should be. It's like the computer doesn't switch cpu_governor even regardless of TLP settings.

**Expected behavior**

The CPU governor is set to powersave so the fan should be quiet and laptop should not overheat regardless of powercable plugged in or plugged out.

**To Reproduce**

Steps to reproduce the unexpected behavior:

1. Does the problem occur on battery or AC or both? Only on AC.
2. Actions to reproduce the behaviour--> I changed my settings with this command: sudo nano /etc/default/tlp
3. Shell commands entered and their output: sudo tlp start
4. **Full output of `tlp-stat` via https://gist.github.com/ for *all*
   matching cases of 1.** LINK: [https://gist.github.com/janez33/d6b4f0c27e44402df27023c58067b385](url)

**Additional context**

Add any other context about the problem here.

linrunner, 2019-04-29:
Hi, you said it yourself already

> TLP is showing that powersave settings are ON

and your tlp-stat outputs prove that all settings are applied correctly. 

This is not an issue with TLP.

I guess it's something between the kernel and the laptop hardware (or BIOS/firmware).

linrunner, 2019-04-29:
One more hint at a kernel problem (from the tlp-stat output):

    CPU model      = Intel(R) Core(TM) i7-8550U CPU @ 1.80GHz
    /sys/devices/system/cpu/cpu0/cpufreq/scaling_driver    = intel_pstate
    /sys/devices/system/cpu/cpu0/cpufreq/scaling_governor  = powersave
    *** /sys/devices/system/cpu/cpu0/cpufreq/scaling_min_freq  =  1800000 [kHz]
    *** /sys/devices/system/cpu/cpu0/cpufreq/scaling_max_freq = 1800000 [kHz]

Your CPU seems locked at max frequency. TLP doesn't touch it because you didn't configure the freqs.

My prediction is: when you try to configure CPU_SCALING_MIN_FREQ* in TLP the setting will not apply because the kernel doesn't accept it because something (ACPI BIOS) tells it not to do so.

janez33, 2019-04-29:
Ok man, thank you very much for this :) I'm gonna try to find a way to fix this issue, I've already looked bios settings and there isn't really much you can do there, but I will keep searching 🔎 😉 Ok then, close the issue :) 

_Sent from my OnePlus5T using [FastHub](https://play.google.com/store/apps/details?id=com.fastaccess.github)_

closed 2019-04-29 by linrunner