linrunner/TLP #408: [TLP 1.2.2 Beta] Call for testers

linrunner, 2019-05-01:
Hi @all,

I need a little help with the upcoming buxfix release.

**What's changed?**
Details can be found in the [changelog](https://github.com/linrunner/TLP/blob/master/changelog). The changes essentially revolve around two topics:

(1) Detection of the operating mode – i.e. what tlp-stat -s shows as _"Mode"_ or _"Power source"_:
-  MacBookPro 5.3: circumvent false AC offline status

(2) Battery Features, tlp-stat -b:
-  Issue #395: ThinkPad X1C6 discharge malfunction
-  Explain "Battery Features: Charge Thresholds and Recalibrate"
-  Intelligible recommendation for tp-smapi on X220/T420 et al.
-  Show data for non-featured batteries != BAT0/BAT1 too

**Where do I get packages?**
See the download page for Arch Linux, Fedora Debian/Ubuntu. If your distro is not contained, you may install the master branch (see ["Installing from Source"](https://linrunner.de/en/tlp/docs/tlp-developer-documentation.html#sourcecode)).

**Test Objectives**
The correct operating mode detection is a critical functionality of TLP. "Battery Features" are important to many users as well. With your help I want to make sure that no new problems have been introduces by the changes.

**What do I need from the testers?**
I'm expecting your outputs of

    sudo tlp-stat

on battery and AC (via https://gist.github.com/ please). 

**Thanks for testing!**

kageurufu, 2019-05-01:
I'll give it a go - Thinkpad P1, Intel 8850H, Quadro P2000

https://gist.github.com/kageurufu/5b3c92e77f8c0ad5132cd692fed51134

bubbleguuum, 2019-05-01:
No issue whatsoever, but nothing in the changelog seems to be relevant for my laptop model.

[Thinkpad P72: AC and battery](https://gist.github.com/bubbleguuum/4eb95f5c6bef2efe22470d236430783a)

sodaplayer, 2019-05-02:
Here's the [output](https://gist.github.com/sodaplayer/40c1f9cc6be8601d82326a5782ef82b1) from my T430.

KJoke70, 2019-05-02:
[ac](https://gist.github.com/KJoke70/f35548352c898086ef9ef6e82f3d6e5e#file-tlp-ac), [bat](https://gist.github.com/KJoke70/f35548352c898086ef9ef6e82f3d6e5e#file-tlp-bat) on my T480s.

Tandrial, 2019-05-02:
[T450s](https://gist.github.com/Tandrial/a6200e26a9ae47cea106a238874cd931)
Is it correct that the version from AUR shows `TLP 1.2.1.901`?

ReekyMarko, 2019-05-02:
[ac](https://hastebin.com/vinanecogu) [bat](https://hastebin.com/owavuduhim)

Dell Inspiron 7375 w/ Ryzen 2700U
Arch Linux, tlp-git 1.2.1.r31.1004acb-1

linrunner, 2019-05-02:
> Is it correct that the version from AUR shows `TLP 1.2.1.901`?
Yes. Current version no added in the first post.

linrunner, 2019-05-02:
@ALL: thanks for your outputs :)

linrunner, 2019-05-02:
@sodaplayer : BAT output?

BarbUk, 2019-05-03:
ThinkPad X220
Arch Linux x86_64
5.0.11-1-ck-sandybridge

[tlp stat output](https://gist.github.com/BarbUk/1bbb2c1c3f0ee3ee4df0e6e64f48b77f)

Chronophylos, 2019-05-03:
[AC and BAT](https://git.io/fjnT8) on my T470

parkerlreed, 2019-05-03:
[tlp.txt](https://github.com/linrunner/TLP/files/3143022/tlp.txt)

Razer Blade late 2016

linrunner, 2019-05-04:
1.2.2 Beta test is finished. Thank you very much for your help :).

closed 2019-05-04 by linrunner

linrunner, 2020-01-08:
1.3 Beta phase has started. Please help: https://github.com/linrunner/TLP/issues/449

linrunner, 2021-09-10:
1.4 Beta 2 is [out](https://linrunner.de/tlp/news.html#news-top-1). Enjoy!

linrunner, 2021-12-21:
TLP 1.5 Beta 1 is out. Testers welcome -> https://github.com/linrunner/TLP/issues/609

linrunner, 2023-07-22:
Hi @ALL : TLP 1.6 Beta 1 is out. Testers wanted -> https://github.com/linrunner/TLP/issues/700