linrunner/TLP #409: Documentation: must secure boot really be disabled for the ThinkPad modules to work?

LinuxOnTheDesktop, 2019-05-04:
[x] I've read and accepted the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md)

**Describe the bug**

The [installation part of the documentation webpage](https://linrunner.de/en/tlp/docs/tlp-linux-advanced-power-management.html#installation) says:

> ThinkPad notes:
> 
>     [. . ]
>     You must disable Secure Boot to use the ThinkPad specific packages

I do have secure boot disabled. Yet, I am pretty sure that I used tlp for six months with secure boot **enabled** and with `tp-smapi-dkms` and `acpi-call-dkms` installed - without any problems that I noticed.

**Expected behavior**

N. A.

**To Reproduce**

Steps to reproduce the unexpected behavior:

1. Does the problem occur on battery or AC or both?
2. Actions to reproduce the behaviour
3. Shell commands entered and their output

4. **Full output of `tlp-stat` via https://gist.github.com/ for *all*
   matching cases of 1.**

https://gist.github.com/LinuxOnTheDesktop/2cc5d0a97e29282a2aca4274e3d948a8

**Additional context**

Linux Mint 19.1 x64 Cinnamon
Kernel 5.0.10
ThinkPad X1CG6

linrunner, 2019-05-09:
> I am pretty sure 

May I ask you to prove your assumption? Please:
 * Enable secure boot
 * Show the output of `tlp-stat -s -b`

LinuxOnTheDesktop, 2019-05-09:
I am afraid that I cannot oblige: ever since resetting my BIOS, secure boot no longer works. I think that reinstalling my OS would make it work again; but I am not prepared to do that. Sorry.

linrunner, 2019-05-10:
Well yes, we can! But it's a royal pita to activate after installation. [Docs](https://linrunner.de/en/tlp/docs/tlp-linux-advanced-power-management.html#ubuntu) changed.

closed 2019-05-10 by linrunner