linrunner/TLP #410: CPU FIFO underrun due to tlp-sleep.service

SwampFalc, 2019-05-09:
[x] I've read and accepted the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md)

**Describe the bug**

Normally, I suspend my computer every night, but after installing TLP and enabling both services, my computer ended up crashing when I suspend it. The error message: "CPU pipe A/B FIFO underrun".

Following a tidbit of advice in other bugs related to sleep/suspend, I decided to disable the tlp-sleep.service, and now everything works just like before again.

**To Reproduce**

Steps to reproduce the unexpected behavior:

1. Does the problem occur on battery or AC or both?

I haven't tried it on battery, because I do not need that right now. I could try it if needed to find the source of the bug.

2. Actions to reproduce the behaviour

Install TLP. Enable and start both services. Suspend my computer.

3. Shell commands entered and their output

sudo systemctl suspend
=> CPU pipe A FIFO underrun
=> CPU pipe B FIFO underrun

4. **Full output of `tlp-stat` via https://gist.github.com/ for *all*
   matching cases of 1.**

https://gist.github.com/SwampFalc/7535d4097f1da3f5f60195d34d34996f

linrunner, 2019-05-09:
Hi, 

some background 1st: tlp-sleep.service does the following ([code](https://github.com/linrunner/TLP/blob/master/tlp.in#L397))
* Apply AC settings before the suspend
* Apply settings according to the power source AC/BAT after resume

These are the very same settings applied when booting on AC (or connecting the charger).

I suggest you try to isolate the offending setting, see the [Troubleshooting Guide](https://linrunner.de/en/tlp/docs/tlp-troubleshooting.html).

linrunner, 2019-05-09:
ps. It could be helpful to also try to reproduce in BAT mode, because of the differing settings.

SwampFalc, 2019-05-09:
So I had planned on upgrading my computer from Ubuntu 18.10 to 19.04. I have now done so and the problem seems to have vanished.

So you were probably spot on in labeling this a kernel quirk.

I've made a new gist with the current state (https://gist.github.com/SwampFalc/c90a6659fe93d1d3598348b4a349075c) in case that helps you figure out what was wrong with that kernel. Unfortunately, now that I have upgraded and it all works, I personally can no longer actually help in tracking the error down...

linrunner, 2019-05-09:
I see. A great deal of TLP issues are caused by the kernel:
* https://github.com/linrunner/TLP/issues?utf8=%E2%9C%93&q=is%3Aissue+label%3A%22kernel+quirk%22+
* https://github.com/linrunner/TLP/issues?utf8=%E2%9C%93&q=is%3Aissue+label%3A%22hardware+quirk%22+

closed 2019-05-09 by linrunner