linrunner/TLP #411: Kernel "dazed and confused" error

perronet, 2019-05-13:
After installing TLP I started noticing two types of kernel errors in the console:
```
Message from syslogd@turing-machine at Mar 17 18:49:06 ...
 kernel:[ 4277.021601] Uhhuh. NMI received for unknown reason 2c on CPU 0.

Message from syslogd@turing-machine at Mar 17 18:49:06 ...
 kernel:[ 4277.021613] Do you have a strange power saving mode enabled?

Message from syslogd@turing-machine at Mar 17 18:49:06 ...
 kernel:[ 4277.021619] Dazed and confused, but trying to continue
```

```
Message from syslogd@turing-machine at Apr  7 11:45:11 ...
kernel:[ 7929.693129] Uhhuh. NMI received for unknown reason 3c on CPU 0.

Message from syslogd@turing-machine at Apr  7 11:45:11 ...
kernel:[ 7929.693132] Do you have a strange power saving mode enabled?

Message from syslogd@turing-machine at Apr  7 11:45:11 ...
kernel:[ 7929.693133] Dazed and confused, but trying to continue
```
These errors started to appear just after I installed TLP, and if i uninstall it they won't appear even after weeks: this is why I am assuming that this is caused by TLP. The fact that the error says "Do you have a strange power saving mode enabled?" also suggests that it might be the cause.

Unfortunately, after four months in which I used TLP, I wasn't able to find exactly what triggers this behavior. These errors appear almost randomly and I observed that it doesn't matter if I'm on AC or battery: both types of errors can appear.

I'm on Debian 9.9 (stretch) x86_64 and my kernel version is 4.9.0-9-amd64.
Here is the output of `tlp-stat`.
On AC: https://gist.github.com/perronet/dc7111a943a40ececf35e0523ee8fa01
On battery: https://gist.github.com/perronet/b7c65883ed47dbb1347be6af89298b50

linrunner, 2019-05-14:
Hi, 

you can try to isolate the causing setting by following the [Troubleshooting Guide](https://linrunner.de/en/tlp/docs/tlp-troubleshooting.html). Focus on settings that don't differ between AC and BAT first, e.g. USB. 

I'd also try a newer kernel from Debian backports.

linrunner, 2019-07-04:
No feedback --> closed.

closed 2019-07-04 by linrunner