linrunner/TLP #413: [RFC] Improved versioning scheme

linrunner, 2019-05-23:
Dear distribution package maintainers,

i'd like to propose an improved versioning scheme for TLP.

**Goal**
1. Allow git tag-ging of beta releases
2. Enable beta packages in separate repositories (not mandatory)
3. Release versions must supersede beta (and development) versions

**Requirements**
5. Beta / development releases must be sorted before regular releases
6. Scheme should be (largely) compatible to distribution's packaging policies

**Proposal**
7. Based on https://semver.org/

_Examples:_
8. Development version: 1.3.0-alpha.[0..n]
9. Beta / RC version 1.3.0-beta.[1..n]
10. Release version 1.3.0
11. Bugfix version 1.3.[1..n]

**Questions**
12. Does the scheme fit your policy or can it be easily adapted?
13. Does tagging of beta releases disrupt your workflow e.g. flagging the package as outdated?

Thanks in advance for your feedback!

MoonSweep, 2019-05-23:
Hi,

With a "dumb" numeric sort, 1.3.0-alpha (or any other letters) would be sorted after 1.3.0, which could be problematic. In Debian, this is resolved by using several separators which all have a certain precedence over the others, explained [here](https://www.debian.org/doc/manuals/maint-guide/first.html#namever) (at the end of section 2.6, just before 2.7).

Moreover, in Debian, the hyphen is reserved for separating the upstream version and the Debian revision.

So, with -alpha or -beta, I'd just replace the hyphen with a tilde when importing the tarball. It would be (slightly) more work, but I don't mind. For a concrete example, see [LibreOffice Debian packages](https://snapshot.debian.org/package/libreoffice/).

I don't know about other distributions, so I'm not sure the tilde would suit, say, Arch or Fedora. I guess the hyphen is much more common, so I won't mind if you use it, I'll adapt.

To answer the second question (again from a Debian point of view), I guess the right thing to do would be to upload the alpha and beta versions to experimental, and the release candidates, releases and bugfix versions to unstable (again, this is how LibreOffice maintainers work; it's open to discussion, as it depends on how "stable" you consider your alpha/beta releases; keep in mind that people usually don't install packages from experimental, unless they want to test something specific, so uploading TLP versions to experimental may not be useful at all, since the chance of receiving bug reports for those versions would be extremely low; also, Ubuntu don't sync packages from experimental, but only from unstable).

This, on the other hand, may represent a lot more work for me, depending on the frequency of your releases. In that case, I can't promise to keep up with every tagged release.

XRevan86, 2019-05-23:
In agreement with @MoonSweep I will post this:
```
$ zypper vcmp '1.3.0-alpha' '1.3.0'
1.3.0-alpha is newer than 1.3.0
```

alucryd, 2019-05-24:
Arch Linux doesn't support hyphens in version numbers, so we usually replace those with a dot (or a plus sign). Unfortunately our version comparator thinks alphas and betas are newer, but this shouldn't be a problem because we have a policy saying no alpha, beta, development version, pre-release and whatnot can be packaged in our official repositories unless absolutely necessary. As long as the git tag is marked pre-release, arch users won't flag the package as out-of-date.

```
$ vercmp
vercmp (pacman) v5.1.3

Compare package version numbers using pacman's version comparison logic.

Usage: vercmp <ver1> <ver2>

Output values:
  < 0 : if ver1 < ver2
    0 : if ver1 == ver2
  > 0 : if ver1 > ver2

$ vercmp 1.3.0.alpha 1.3.0
1
```

We sometimes have betas in AUR with a `-devel` suffix added to the regular package name where this could become a problem, but most of the time we only have a VCS version (in this case tlp-git) that will always build the HEAD of the repository. In that case the version is computed from `git describe` whenever possible, or just `r<commit_number>.<commit_hash>` and I have yet to encounter versioning problems with any vcs package.

In any case I am all for semver, it's a good and widely used versioning scheme, and I wish vercmp and pacman could work with it natively.

ArchangeGabriel, 2019-05-24:
Regarding Arch, I must correct that point: since we sometime have alpha/beta/rc software for $reasons, there has been thinking about how to publish the version, and the answer is `1.3.0alpha` or `1.3.0a`. And this works:
```
$ vercmp 1.3.0alpha 1.3.0
-1
```

alucryd, 2019-05-24:
> Regarding Arch, I must correct that point: since we sometime have alpha/beta/rc software for $reasons, there has been thinking about how to publish the version, and the answer is `1.3.0alpha` or `1.3.0a`. And this works:
> 
> ```
> $ vercmp 1.3.0alpha 1.3.0
> -1
> ```

Oh nice! Well, that takes care of it then!

dywisor, 2019-05-25:
semver would be a clear improvement over the current ``.900`` versioning scheme for prereleases.

> Questions
> (11) Does the scheme fit your policy or can it be easily adapted?

Gentoo supports various version suffixes and sorts them properly, including ``alpha`` and ``beta`` ([src](https://devmanual.gentoo.org/ebuild-writing/file-format/index.html)). The syntax is slightly different, for example ``_alphaK`` instead of ``-alpha.K``, but than can be translated with some shell-foo when determining the uri of the release tarball.

> (12) Does tagging of beta releases disrupt your workflow e.g. flagging the package as outdated?

Doubt it. I can push an arbitrary number of versions for the same package to the repo and mark them as stable/testing/experimental individually, but, ultimately, it's up to the user to pick a one. For the hasty ones, there's always the live version, which lets them build against virtually any git commit/tag (default: HEAD) as long as the build instructions fit.

Mystro256, 2019-05-29:
Personally, I've always liked the idiom of "1.2.90-alpha" "1.2.91-Beta1" "1.2.92-Beta2" etc, but it doesn't matter too much to me how it's tagged in git.

For Fedora/EL, if you went with 1.3.0-aplha for example, in a spec file I would do something like this:
```
...
Name: tlp
Version: 1.3.0
Release: 0.1.alpha
...
```
Then I would keep bumping 0.1 until 1.3.0 release.
For example the version history could be something like this:
tlp-1.2.2-1
tlp-1.3.0-0.1.alpha
tlp-1.3.0-0.2.beta1
tlp-1.3.0-0.3.beta2
tlp-1.3.0-0.4.RC1
tlp-1.3.0-0.5.RC2
tlp-1.3.0-1

Note that generally I would only create pre-release builds in rawhide, although I'm not opposed to pushing beta and RC's to the updates-testing repositories if needed. Similar to other distros, I would avoid pushing non-release versions to any repositories enabled by default.

linrunner, 2019-06-02:
@all: thank you so much for your extensive feedback.

Conclusion: my proposed scheme works for you. It cannot perfectly fit all package tools, small adaptions are inevitable but do not pose an obstacle for any of you.

_Clarification: it is not my intention that you should provide beta packages for your distributions._

linrunner, 2019-07-04:
Implemented.

closed 2019-07-04 by linrunner