linrunner/TLP #415: [Thinkpad X240] Error: discharge BAT1 malfunction.

frauhottelmann, 2019-05-28:
[x] I've read and accepted the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md)

**Describe the bug**

sudo tlp recalibrate should discharge battery 1 (since I only have the external battery)

**Expected behavior**

It starts discharging and stops with error after a couple of seconds.

**To Reproduce**

Steps to reproduce the unexpected behavior:

1. Does the problem occur on battery or AC or both?

Only on AC, command doesn't work on battery.

2. Actions to reproduce the behaviour

run sudo tlp recalibrate (or sudo tlp recalibrate bat1)

3. Shell commands entered and their output

`robert@robert-ThinkPad-X240:~$ sudo tlp recalibrate
Setting temporary charge thresholds for BAT1:
  start =  96 (no change)
  stop  = 100 (no change)
Initiating discharge of battery BAT1 ...............
Error: discharge BAT1 malfunction.
`

4. **Full output of `tlp-stat` via https://gist.github.com/ for *all*
   matching cases of 1.**

[AC](https://gist.github.com/frauhottelmann/5501af4f1e2b28acd33275babbdcc88c)

linrunner, 2019-05-28:
Please enable [Trace Mode](https://linrunner.de/en/tlp/docs/tlp-configuration.html#tracemode), execute your command and show the output of

  tlp-stat -T

frauhottelmann, 2019-05-28:
[tlp-stat -T](https://gist.github.com/frauhottelmann/2ecebb5d12d93858f995ceab2c2cbab2)

linrunner, 2019-05-28:
OK. I also need

    tlp-stat --psup

frauhottelmann, 2019-05-28:
[tlp-stat --psup](https://gist.github.com/frauhottelmann/119cd90a24cd76d1bd1286b4ca108177)

linrunner, 2019-05-28:
Thanks. 

The current logic interprets "BAT0 not installed" plus "BAT1 present" as a special case for the E series models, where BAT1 is the first (and only) battery. I'll have to think about that, this will take some time. 

Please stay tuned for testing a fix.

linrunner, 2019-06-11:
Done. Please test the fix. Packages are [here](https://download.linrunner.de/packages/).

frauhottelmann, 2019-06-13:
Will try. But I cna only do it after the weekend because I have to disconnect my new internal battery :smile:

frauhottelmann, 2019-06-22:
Works, it starts recharging correctly.

`
Currently discharging battery BAT1:
voltage            =  11797 [mV]
remaining capacity =  38220 [mWh]
remaining percent  =     90 [%]
remaining time     =    138 [min]
power              =  16504 [mW]
state              = Discharging
force discharge    = 1
Press Ctrl+C to cancel.
`

linrunner, 2019-06-22:
Thank you for testing.

KristerV, 2019-07-24:
Just to confirm - current master works on arch linux on my T470 which also has internal battery missing for some reason.

linrunner, 2019-07-29:
@KristerV : thank you very much for your feedback.

dinokov, 2019-09-14:
Also confirming that with current master 1.3.0-alpha.0 installed from AUR , recalibrate and discharge now works on manjaro linux on my T440 which has internal battery missing.

linrunner, 2019-09-16:
Thanks @dinokov

closed 2019-09-16 by linrunner