linrunner/TLP #416: change CPU_SCALING documentation

Coleej, 2019-07-02:
Add a sentence to make it clear that the defaults don't apply to battery power, if only the CPU_SCALING for only AC is set.

In other words, if  you set:

```
CPU_SCALING_GOVERNOR_ON_AC=performance
#CPU_SCALING_GOVERNOR_ON_BAT=powersave
```

The governor on battery power is also performance. To correctly have performance on AC and powersave on BAT, you must set:

 ```
CPU_SCALING_GOVERNOR_ON_AC=performance
CPU_SCALING_GOVERNOR_ON_BAT=powersave
```

linrunner, 2019-07-03:
Hi, 

values for commented i.e. disabled parameters are not the default. The default is documented above the parameter lines and there's a hint about this circumstancein the leading notes:

    # Notes:
    # - Some parameters are disabled, remove the leading '#' to enable # them;
    #   shown values are suggestions not defaults
    # - Default *: intrinsic default that is effective when the parameter is missing
    #   or disabled by a leading '#'; use PARAM="" to disable intrinsic defaults for
    #   parameters with text string values

In your case there is no default:

    # Default: <none>
    #CPU_SCALING_GOVERNOR_ON_AC=powersave
    #CPU_SCALING_GOVERNOR_ON_BAT=powersave

Besides i find it obvious that you have to uncomment and configure AC and BAT parameters to get distinct values for AC and BAT.

Coleej, 2019-07-03:
Ahh, sorry. I opened this issue hastily.

However, I disagree that it's obvious to a new user that setting an option for AC, but not BAT, will change both to the AC setting. It seems to me that the more typical behavior would be to leave BAT at its previous value as if nothing was altered, which in my case was powersave as configured by Ubuntu.

closed 2019-07-03 by Coleej

linrunner, 2019-07-04:
TLP has to write a value on every change of power source, so the AC value will stick when switching to BAT.