linrunner/TLP #417: tlp setcharge BAT1 actually sets BAT0

timrichardson, 2019-07-07:
[x] I've read and accepted the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md)

**Describe the bug**
tim@moncrief:/etc/default$ sudo tlp setcharge BAT1 95
Setting temporary charge thresholds for BAT0:
  stop  =  95 (no change)

tlp-stat verifies that the threshold for BAT0 is changed. 

A clear and concise description of what the bug is.

**Expected behavior**
sudo tlp setcharge BAT1 95
should change BAT1, not BAT0

A clear and concise description of what you expected to happen.

**To Reproduce**

Steps to reproduce the unexpected behavior:

1. Does the problem occur on battery or AC or both?
both
2. Actions to reproduce the behaviour
execute
tlp setcharge BAT1 95

3. Shell commands entered and their output
4. **Full output of `tlp-stat` via https://gist.github.com/ for *all*
   matching cases of 1.**

https://gist.github.com/timrichardson/e08ed9343b5542426c8cebba7385a9bf

**Additional context**

Add any other context about the problem here.
Thinkpad T480. 

clear

timrichardson, 2019-07-07:
oh:
in setcharge_battery 

    2) # assume $1,$2 are thresholds
            start_thresh=$1
            stop_thresh=$2
            bat=DEF # use default(1st) battery
            ;;

so providing two arguments only means I am setting BAT0 to start at threshold BAT1 and finish at 95
Perhaps this should be a usage error?

timrichardson, 2019-07-07:
No, this is just stupid user error.

closed 2019-07-07 by timrichardson

linrunner, 2019-07-07:
OK. I concede the command should throw an error msg when 1st + 2nd parameters are non numeric. I've made a note.

linrunner, 2019-10-02:
Checks implemented. See commit.