linrunner/TLP #419: Wifi card put on sleep but impossible to wake up

froom, 2019-07-22:
[x] I've read and accepted the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md)

**Describe the bug**
After my wifi card has been put down it never came back, even after restart.

A clear and concise description of what the bug is.
I use Linux Mint 19 x64 with Wfce on my Sony VAIO (vpcf11e4e) laptop.
I have installed TLP to lower the global temperature because it made my computer shut down.
After a short time my wifi card has gone to sleep but since it doesn't wake up.
I have uninstalled TLP and TLP-rdw because I thought all would wome back like before installation but it doesn't.

**Expected behavior**
I didn't expect my card to go to bed indefinitly. I tried to launch firefox, I tried to enable wifi with network-manager but nothing worked.

A clear and concise description of what you expected to happen.
I expected it to be simpler to revert the state it put my hardware on.

**To Reproduce**
My wifi card is a Centrino Advanced-N 6200. I have just installed tlp and tlp-rdw and I didn't go on the web during 10 minutes and that's it.

Steps to reproduce the unexpected behavior:

1. Does the problem occur on battery or AC or both?
it happened on AC

2. Actions to reproduce the behaviour
see what I explained before

3. Shell commands entered and their output

# lshw
*-network DÉSACTIVÉ
                description: Interface réseau sans fil
                produit: Centrino Advanced-N 6200
                fabriquant: Intel Corporation
                identifiant matériel: 0
                information bus: pci@0000:02:00.0
                nom logique: wlp2s0
                version: 35
                numéro de série: 00:23:14:45:f2:b8
                bits: 64 bits
                horloge: 33MHz
                fonctionnalités: pm msi pciexpress bus_master cap_list ethernet physical wireless
                configuration: broadcast=yes driver=iwlwifi driverversion=4.15.0-54-generic firmware=9.221.4.1 build 25532 latency=0 link=no multicast=yes wireless=IEEE 802.11
                ressources: irq:26 mémoire:e7a00000-e7a01fff

# iwconfig
wlp2s0    IEEE 802.11  ESSID:off/any  
          Mode:Managed  Access Point: Not-Associated   Tx-Power=off   
          Retry short limit:7   RTS thr:off   Fragment thr:off
          Power Management:off
enp4s0    no wireless extensions.
lo        no wireless extensions.

# dmesg | tail
[ 1230.835577] ata1.00: status: { DRDY }
[ 1230.835584] ata1: hard resetting link
[ 1231.931599] ata1: SATA link up 1.5 Gbps (SStatus 113 SControl 300)
[ 1231.941410] ata1.00: ACPI cmd f5/00:00:00:00:00:00 (SECURITY FREEZE LOCK) filtered out
[ 1231.951402] ata1.00: ACPI cmd f5/00:00:00:00:00:00 (SECURITY FREEZE LOCK) filtered out
[ 1231.951572] ata1.00: configured for UDMA/133
[ 1231.951613] ata1: EH complete
[ 1638.024366] IPv6: ADDRCONF(NETDEV_UP): enp4s0: link is not ready
[ 1638.028364] IPv6: ADDRCONF(NETDEV_UP): wlp2s0: link is not ready

other things interesting in # dmesg
[   12.085826] iwlwifi 0000:02:00.0: CONFIG_IWLWIFI_DEBUG disabled
[   12.085829] iwlwifi 0000:02:00.0: CONFIG_IWLWIFI_DEBUGFS enabled
[   12.085830] iwlwifi 0000:02:00.0: CONFIG_IWLWIFI_DEVICE_TRACING enabled
[   12.085832] iwlwifi 0000:02:00.0: Detected Intel(R) Centrino(R) Advanced-N 6200 AGN, REV=0x74
[   12.098407] iwlwifi 0000:02:00.0: reporting RF_KILL (radio disabled)
[   12.098429] iwlwifi 0000:02:00.0: RF_KILL bit toggled to disable radio.
[   12.279549] ieee80211 phy0: Selected rate control algorithm 'iwl-agn-rs'
[   19.307622] IPv6: ADDRCONF(NETDEV_UP): enp4s0: link is not ready
[   19.310818] sky2 0000:04:00.0 enp4s0: enabling interface
[   19.311303] IPv6: ADDRCONF(NETDEV_UP): enp4s0: link is not ready
[   19.428283] IPv6: ADDRCONF(NETDEV_UP): wlp2s0: link is not ready

# dmesg | grep wifi
[   11.286061] iwlwifi 0000:02:00.0: can't disable ASPM; OS doesn't have ASPM control
[   11.313399] iwlwifi 0000:02:00.0: Direct firmware load for iwlwifi-6000-6.ucode failed with error -2
[   11.315131] iwlwifi 0000:02:00.0: Direct firmware load for iwlwifi-6000-5.ucode failed with error -2
[   11.331306] iwlwifi 0000:02:00.0: loaded firmware version 9.221.4.1 build 25532 op_mode iwldvm
[   12.085826] iwlwifi 0000:02:00.0: CONFIG_IWLWIFI_DEBUG disabled
[   12.085829] iwlwifi 0000:02:00.0: CONFIG_IWLWIFI_DEBUGFS enabled
[   12.085830] iwlwifi 0000:02:00.0: CONFIG_IWLWIFI_DEVICE_TRACING enabled
[   12.085832] iwlwifi 0000:02:00.0: Detected Intel(R) Centrino(R) Advanced-N 6200 AGN, REV=0x74
[   12.098407] iwlwifi 0000:02:00.0: reporting RF_KILL (radio disabled)
[   12.098429] iwlwifi 0000:02:00.0: RF_KILL bit toggled to disable radio.
[   12.727109] iwlwifi 0000:02:00.0 wlp2s0: renamed from wlan0
[ 1927.125217] iwlwifi 0000:02:00.0: can't disable ASPM; OS doesn't have ASPM control
[ 1927.125715] iwlwifi 0000:02:00.0: Direct firmware load for iwlwifi-6000-6.ucode failed with error -2
[ 1927.125732] iwlwifi 0000:02:00.0: Direct firmware load for iwlwifi-6000-5.ucode failed with error -2
[ 1927.126242] iwlwifi 0000:02:00.0: loaded firmware version 9.221.4.1 build 25532 op_mode iwldvm
[ 1927.157322] iwlwifi 0000:02:00.0: CONFIG_IWLWIFI_DEBUG disabled
[ 1927.157325] iwlwifi 0000:02:00.0: CONFIG_IWLWIFI_DEBUGFS enabled
[ 1927.157327] iwlwifi 0000:02:00.0: CONFIG_IWLWIFI_DEVICE_TRACING enabled
[ 1927.157330] iwlwifi 0000:02:00.0: Detected Intel(R) Centrino(R) Advanced-N 6200 AGN, REV=0x74
[ 1927.169965] iwlwifi 0000:02:00.0: reporting RF_KILL (radio disabled)
[ 1927.169992] iwlwifi 0000:02:00.0: RF_KILL bit toggled to disable radio.
[ 1927.204861] iwlwifi 0000:02:00.0 wlp2s0: renamed from wlan0

# iwlist wlp2s0 power
wlp2s0    Current mode:off

# ifconfig wlp2s0 up
SIOCSIFFLAGS: Opération impossible du fait de RF-kill

# iwlist wlp2s0 scan
wlp2s0    Interface doesn't support scanning : Network is down

If you want me to try other commands don't hesitate as I really would like to find a solution.

4. **Full output of `tlp-stat` via https://gist.github.com/ for *all*
   matching cases of 1.**
Unfortunately it isn't anymore on my system.

**Additional context**
Add any other context about the problem here.

I don't know if it's a bug or if I did something wrong but I have change or done nothing after the installation, just shut down and restart my laptop once.
RFKILL seems to do something in this process but is it on iys own or because of something else, I don't know.
That's all I can tell about it for the moment. At this moment I use an ethernet cable that runs in the middle of my room so I would be really pleased if it wouldn't last too long.
Thank you.

linrunner, 2019-07-23:
With its default configuration TLP doesn't disable wifi. None of the settings is permanent. Especially not after you've uninstalled TLP. 

So you must have done something else you don't remember. Possibly some hardware kill switch (or button) on your laptop? 

Anyway: this is not a TLP issue, please seek help in an appropriate Linux or hardware forum.

closed 2019-07-23 by linrunner

froom, 2019-07-23:
Thanks for your answer.

I have resolved this issue: wifi (and bluetooth) were hard blocked and I don't know how it happened but I managed to put these on by pressing button after button with the FN function on my laptop keyboard.
Maybe I will try to reinstall TLP when I will have time because it was possible to put my laptop cpu on conservative energy management mode and that was the only time I managed to do that on my Mint 19.1 distro. So in one way I'm happy that you are sure it wasn't a TLP issue so I can test it again and resolve my "too hot computer" issues. The positive is that I have learned some little things.

Hope I haven't taken too much of your time. Thanks again!