linrunner/TLP #420: [T470] Undock in clamshell mode, NetworkManager doesn't enable wifi

javiertury, 2019-07-29:
[x] I've read and accepted the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md)

**Describe the bug**

The laptop is docked with the lid closed. The dock is connected to the internet through an ethernet cable and has an external monitor attached. Then the laptop is taken away from the dock. Because the lid is closed, the laptop suspends.

When the device is awaken, the wifi device is unblocked correctly but NetworkManager still shows wifi as disabled.

**Expected behavior**

NetworkManager shows wifi as enabled

**To Reproduce**

Steps to reproduce the unexpected behavior:
1. Dock laptop with the lid closed(clamshell mode). Dock has external display. Dock is connect to internet through a ethernet cable.
2. Take away laptop from dock, it will suspend itself.
3. Resume laptop from sleep

**Additional context**

If the laptop is undocked with the laptop lid open, then network manager correctly enables wifi.

Config

```
// config
WIFI_PWR_ON_AC=off
WIFI_PWR_ON_BAT=on
DEVICES_TO_DISABLE_ON_LAN_CONNECT="wifi"
DEVICES_TO_ENABLE_ON_LAN_DISCONNECT="wifi"
```

Before undocking

```
$ tlp-stat
+++ Wireless
bluetooth = off (software)
wifi      = off (software)
wwan      = none (no device)

wlp4s0(iwlwifi)               : wifi, not connected, power management = off

$ rfkill
ID TYPE      DEVICE                 SOFT      HARD
 0 bluetooth tpacpi_bluetooth_sw blocked unblocked
 1 wlan      phy0                blocked unblocked

$ nmcli radio
WIFI-HW  WIFI      WWAN-HW  WWAN
enabled  disabled  enabled  enabled
```

After undocking

```
$ tlp-stat
+++ Wireless
bluetooth = off (software)
wifi      = on
wwan      = none (no device)

wlp4s0(iwlwifi)               : wifi, not connected, power management = on

$ rfkill
ID TYPE      DEVICE                   SOFT      HARD
 0 bluetooth tpacpi_bluetooth_sw   blocked unblocked
 1 wlan      phy0                unblocked unblocked

$ nmcli radio
WIFI-HW  WIFI      WWAN-HW  WWAN
enabled  disabled  enabled  enabled
```

linrunner, 2019-07-29:
> [x] I've read and accepted the Bug Reporting Howto

You must have overlooked this line in the bug report template:

> Full output of tlp-stat via https://gist.github.com/ for all matching cases of 1.

Please provide the full output before undocking and after resume (wakeup).

javiertury, 2019-07-29:
> Full output of tlp-stat via https://gist.github.com/ for all matching cases of 1.

I'm sorry, I didn't understand that line correctly. I thought it referred to only AC/battery related bugs.

[Here](https://gist.github.com/javiertury/ea2522bfb6829f08846aaa44c82e4c2b) is the output of `tlp-stat` before and after undocking.

linrunner, 2019-07-30:
OK, please enable [trace mode](https://linrunner.de/en/tlp/docs/tlp-configuration.html#tracemode) and show me the period from undocking to resume (waking up):

    tlp-stat -T

javiertury, 2019-07-31:
[Here](https://gist.github.com/javiertury/f17f47180798630cc8e41ef1a41dd5fd) is the output for the before-undock to after-resume period.

linrunner, 2019-07-31:
The trace shows that wifi is turned on immediately *after* ethernet disconnects and *before* suspending. Extract form the trace:

    Jul 31 15:05:16 localhost.localdomain tlp[23993]: +++ rdw_udev(usb_dock).undock dev=/sys/devices/pci0000:00/0000:00:14.0/usb1/1-1 syspwr=1
    Jul 31 15:05:17 localhost.localdomain tlp[24148]: +++ rdw_nm(enp0s31f6).down
    Jul 31 15:05:17 localhost.localdomain tlp[24156]: rdw_nm(enp0s31f6).down: type=ethernet [saved]
    Jul 31 15:05:17 localhost.localdomain tlp[24159]: device_switch(wifi, on, rdw_nm_wifi, 2)
    Jul 31 15:05:17 localhost.localdomain tlp[24168]: get_devc(wifi) = /sys/class/rfkill/rfkill1/state
    Jul 31 15:05:17 localhost.localdomain tlp[24175]: get_devs(wifi) = 0
    Jul 31 15:05:17 localhost.localdomain tlp[24201]: device_switch(wifi, on).rfkill
    Jul 31 15:05:17 localhost.localdomain tlp[24210]: get_devs(wifi) = 1Jul 31 15:05:17 localhost.localdomain tlp[24274]: invoke_nmcli(wifi, on).radio: rc=0
    Jul 31 15:05:17 localhost.localdomain tlp[24280]: get_devs(wifi) = 1
    Jul 31 15:05:17 localhost.localdomain tlp[24281]: device_switch(wifi, on).ok: rc=0
    Jul 31 15:05:17 localhost.localdomain tlp[24354]: +++ suspend (1.2.1) ++++++++++++++++++++++++++++++++++++++++
    Jul 31 15:05:45 localhost.localdomain tlp[24524]: +++ resume (1.2.1) ++++++++++++++++++++++++++++++++++++++++

I also shows that tlp-rdw doesn't touch wifi state after resume. 

My conclusion is that NM doesn't initiate the reconnect after resume.

javiertury, 2019-07-31:
With that information I made some tests. These are the results:

- Disconnect ethernet cable from docked laptop (works well)
  - Logs are similar to undocking case until suspend
- Suspend undocked laptop connected to wifi (works well)
  - Logs are different from undocking case after suspend

Suspend undocked laptop connected to wifi (complete logs)

```
Jul 31 22:54:13 localhost.localdomain tlp[27247]: parse_args4config: tlp-stat -T
Jul 31 22:54:27 localhost.localdomain tlp[27737]: check_run_flag(rdw_kill): rc=1
Jul 31 22:54:27 localhost.localdomain tlp[27737]: +++ rdw_nm(wlp4s0).down
Jul 31 22:54:27 localhost.localdomain tlp[27737]: PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin
Jul 31 22:54:27 localhost.localdomain tlp[27737]: rdw_nm(wlp4s0).down: type=wifi [saved]
Jul 31 22:54:27 localhost.localdomain tlp[27737]: check_timed_lock(rdw_nm_wifi).not_locked: 1564606467
Jul 31 22:54:33 localhost.localdomain tlp[27985]: check_run_flag(rdw_kill): rc=1
Jul 31 22:54:33 localhost.localdomain tlp[27985]: +++ rdw_nm(wlp4s0).up
Jul 31 22:54:33 localhost.localdomain tlp[27985]: PATH=/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin
Jul 31 22:54:33 localhost.localdomain tlp[27985]: rdw_nm(wlp4s0).up: type=wifi [nmcli]
Jul 31 22:54:33 localhost.localdomain tlp[27985]: check_timed_lock(rdw_nm_wifi).not_locked: 1564606473
Jul 31 22:54:38 localhost.localdomain tlp[28104]: parse_args4config: tlp-stat -T
```

Undocking laptop in clamshell mode (summarized logs from [previous report](https://github.com/linrunner/TLP/issues/420#issuecomment-516847344))
```
Jul 31 15:05:17 localhost.localdomain tlp[24354]: +++ suspend (1.2.1) ++++++++++++++++++++++++++++++++++++++++
Jul 31 15:05:18 localhost.localdomain tlp[24413]: set_wifi_power_mode(0, wlp4s0).iw: off; rc=0
Jul 31 15:05:18 localhost.localdomain tlp[24419]: disable_wake_on_lan: enp0s31f6; rc=0
Jul 31 15:05:45 localhost.localdomain tlp[24524]: +++ resume (1.2.1) ++++++++++++++++++++++++++++++++++++++++
Jul 31 15:05:45 localhost.localdomain tlp[24532]: restore_device_states: /run/tlp/rfkill_saved
Jul 31 15:05:45 localhost.localdomain tlp[24552]: compare_and_save_power_state(1).equal
Jul 31 15:05:45 localhost.localdomain tlp[24565]: set_nmi_watchdog: 0; rc=0
Jul 31 15:05:45 localhost.localdomain tlp[24589]: set_wifi_power_mode(1, wlp4s0).iw: on; rc=0
Jul 31 15:05:45 localhost.localdomain tlp[24595]: disable_wake_on_lan: enp0s31f6; rc=0
```

linrunner, 2019-08-02:
You may use "undocking suspended laptop" as a workaround. But i can't do anything against the original issue, because TLP has no influence on the Network Manager.

Btw: please don't summarize logs yourself. I suggest to use 

    TLP_DEBUG="lock nm rf run udev" 

instead (for this issue),

javiertury, 2019-08-02:
I understand, thanks for your help in diagnosing the problem. I will close the issue since TLP is not at fault.

closed 2019-08-02 by javiertury