linrunner/TLP #421: TLP is not able to change CPU Governor on Secure Boot machines

53c70r, 2019-07-29:
This bug is related to machines using UEFI Secure Boot and kernel lockdown mode enabled.
TLP does not change the CPU Governor depending on AC plugged in or not.
All kernels in lockdown mode will not support direct MSR access for security reasons and therefore can't change the CPU Governor directly by manipulating the module-specific register using x86_energy_perf. [link](https://lwn.net/Articles/750730/)
A possible fix could be to edit the /sys/ fs directly to change the Governor instead of using x86_energy_perf.

**Reproducing the bug**
1. Booting into system with kernel in lockdown mode
2. `[root@a ~]# x86_energy_perf_policy --all performance`
`x86_energy_perf_policy: pwrite(cpu0, offset 0x770, 0x1) = -1: Operation not permitted`
3. `[root@a ~]# dmesg | grep MSR`
`[   55.058941] Lockdown: x86_energy_perf: Direct MSR access is restricted; see man kernel_lockdown.7`
4. `tlp ac` or `tlp bat` will result in new kernel log entry's like `msr: Direct access to MSR XXX` for each call which indicates TLP tryes to use MSR

linrunner, 2019-07-30:
Thanks for the hint. In fact using the new sysfs interface of kernel 5.2 is already on my list. Better than using an external tool. 

Btw: this is not about the "CPU governor" but about [EPB](https://www.kernel.org/doc/html/latest/admin-guide/pm/intel_epb.html) – and i wouldn't call this a bug in TLP, it's more a kernel change that breaks user space ... ;-)

linrunner, 2019-08-13:
Done. Please test and report.

Debian/Ubuntu packages provided [here](https://download.linrunner.de/packages/).

linrunner, 2019-08-18:
I did some refactoring and merged HWP.EPP and EPB features. Please test and report.

53c70r, 2019-08-21:
I can confirm it works as intended. Btw my platform is Fedora.

linrunner, 2019-08-21:
Thank you for testing.

closed 2019-09-26 by linrunner

ljrk0, 2021-06-21:
JFYI: The kernel-tools got updated with 5.11 and now work properly again:

https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/tools/power/x86/x86_energy_perf_policy/x86_energy_perf_policy.c?h=v5.11&id=9c70f046787505127f2f8bb60ad24bb3c8bfc61a

To set the EPB:

```
# x86_energy_perf_policy --epb balance-performance
```

Unfortunately `--hwp-enable`, `--hwp-epp`, `--hwp-max max_val` and `--turbo-enable 0|1` still poke at MSRs, but that's not what was wanted here anyway.