linrunner/TLP #422: Trace mode: logger --id fails with "Operation not permitted"

javiertury, 2019-07-31:
[x] I've read and accepted the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md)

**Describe the bug**

In trace mode, `systemctl status tlp` and `tlp-stat -T` record logging failures instead of trace information. They display the following message multiple times:

```
logger: send message failed: Operation not permitted
```

**To Reproduce**

Every time [this line](https://github.com/linrunner/TLP/blob/master/tlp-func-base.in#L69) and [this other line](https://github.com/linrunner/TLP/blob/master/tlp-func-base.in#L102) are executed.

**Additional context**

The first result on google is [this issue](https://github.com/jonathanio/update-systemd-resolved/issues/25). Folowing their advice, I edited `/usr/share/tlp/tlp-func-base`, removed the `--id` arg and managed to work around this issue.

linrunner, 2019-07-31:
In order to be able to assess the problem, I am missing essential data:

    tlp-stat -s

Can you please also verify if [this](https://github.com/linrunner/TLP/commit/2805a85a33aa019fa3dd1fef1de5bd00ef91ba80) and the next commit solve the problem?

javiertury, 2019-07-31:
[Here](https://gist.github.com/javiertury/9bb6935e5b4970ccdb344530818dc902) is the output of `tlp-stat -s`.

After those two commits, the errors disappear. However, the logging output is much more scarce.

I will use issue #420 as an example. Compare the [logs](https://gist.github.com/javiertury/f17f47180798630cc8e41ef1a41dd5fd) before applying any patch, to the logs after applying both [patches](https://gist.github.com/javiertury/75ea65c9814c92471ca2e1393cccf4a5).

All the logs are for the before-undock to the after-resume period.

javiertury, 2019-08-01:
I found the culprit, SELinux.

```
type=AVC msg=audit(1564652336.513:9741): avc:  denied  { sys_admin } for  pid=24166 comm="logger" capability=21  scontext=system_u:system_r:tlp_t:s0-s0:c0.c1023 tcontext=system_u:system_r:tlp_t:s0-s0:c0.c1023 tclass=capability permissive=0
```

linrunner, 2019-08-02:
SELinux is a PITA. suggest to file a Fedora Bug against **selinux-policy**.

javiertury, 2019-08-02:
[Done](https://bugzilla.redhat.com/show_bug.cgi?id=1737106)

closed 2019-08-02 by javiertury