linrunner/TLP #423: Website availability over Tor network.

affinityv, 2019-08-03:
Accessing http://linrunner.de/ via the Tor network fails with 503 error.

Please provide .onion resource and/or make the website fully available via the Tor network if you care about user privacy.

Thank you

linrunner, 2019-08-03:
[http://linrunner.de/](http://linrunner.de/) will redirect to [https://linrunner.de/](https://linrunner.de/). Did you try [https://linrunner.de/](https://linrunner.de/) directly?

affinityv, 2019-08-03:
Yes, I've had the redirection work, not every try of HTTP resulted in 503 failure.  But HTTPS seems to fail similarly with timeouts ... unless you've now fixed that.

affinityv, 2019-08-03:
Restarting Tor to get a new circuit helps... sometimes.

linrunner, 2019-08-12:
I was able to reproduce your issue the day after your report. My web hoster's home page https://all-inkl.com/ also had timeouts. Today the accessibility seems to be normal for me. For you too?

affinityv, 2019-08-13:
Both seem to be timing out again; you can try with new Tor circuits, but it is hit and miss.

closed 2020-01-20 by linrunner