linrunner/TLP #424: Battary protection for Huawei Matebooks?

ylxdzsw, 2019-08-09:
TLP supports setting battery charging thresholds, but only for ThinkPads. Huawei Matebooks also support a similar feature, and is quite easy with [this driver](https://github.com/aymanbagabas/Huawei-WMI) (Just write to `/sys/devices/platform/huawei-wmi/charge_thresholds`, after confirming this file exists). Are you interested in adding this to TLP?

linrunner, 2019-08-12:
Hi,

does it have separate start/stop thresholds ranging 0..100?

To be honest, adding another case to the threshold logic in TLP makes the code even more complicated than it already is with the three cases for ThinkPads. I'd prefer a kernel interface like https://github.com/linrunner/TLP/issues/321 which is supported out of the box:

    /sys/class/power_supply/BAT0/charge_start_threshold         =     66 [%]
    /sys/class/power_supply/BAT0/charge_stop_threshold          =    100 [%]

Maybe you ask the kernel module author if he adapts the current interface to the new framework?

ylxdzsw, 2019-08-13:
Hi,

Yes, the file looks like this:
```
> cat /sys/devices/platform/huawei-wmi/charge_thresholds 
70 90
```
and they are the start and stop thresholds (%) respectively.

I agree that a uniform interface would be the ideal solution, but it looks very similar to the current one and I think a simple adapter could support both. To be honest I was shocked by the length of the ThinkPad logic. In my opinion supporting Huawei WMI should as simple as something like
```
if [ -f /sys/devices/platform/huawei-wmi/charge_thresholds ]; then
    echo "$START $STOP" > /sys/devices/platform/huawei-wmi/charge_thresholds
fi
```

linrunner, 2019-08-13:
The logic contains a lot of checking for prerequisites, like do we run on supported hardware, is the kernel module there and so on. This is also necessary to properly support Matebooks.

So please don't assume a simple 

    if [ -f /sys/devices/platform/huawei-wmi/charge_thresholds ]; then
         echo "$START $STOP" > /sys/devices/platform/huawei-wmi/charge_thresholds
    fi

is enough for inclusion into TLP. 

All case distinctions in [func.d/35-tlp-func-batt](https://github.com/linrunner/TLP/blob/master/func.d/35-tlp-func-batt), [func.d/tlp-func-stat](https://github.com/linrunner/TLP/blob/master/func.d/tlp-func-stat#L458) and [tlp-stat](https://github.com/linrunner/TLP/blob/master/tlp-stat.in#L811) must be served.

You have the necessary hardware to conduct the tests. Feel free to make a pull request.

validatedev, 2020-01-13:
That feature was implemented on https://github.com/nekr0z/linux-on-huawei-matebook-13-2019/blob/master/README.md#battery-protection , so I don’t think TLP should have that feature, because the feature is there with other implementation.

linrunner, 2020-01-14:
@validatedev : no objection from my side :D

linrunner, 2020-01-14:
@validatedev no objection from my side.

ylxdzsw, 2020-01-14:
Also huawei-wmi driver added support for an API similar (just different in name, should be solved by symlinks) to #321 recently so no special support is needed here now. I think this can be closed.

closed 2020-01-14 by ylxdzsw

awehrfritz, 2020-01-18:
@ylxdzsw, the changes discussed in #321 will not support the interface that the huawei-wmi driver ([drivers/platform/x86/huawei-wmi.c](https://github.com/torvalds/linux/blob/13b2668d6fc95dcecd11e4f86f47be24fda7da1f/drivers/platform/x86/huawei-wmi.c#L467)) is using, as the interface used in #321 is specific to the driver for ThinkPads (i.e. using [custom attributes](https://github.com/torvalds/linux/blob/13b2668d6fc95dcecd11e4f86f47be24fda7da1f/drivers/platform/x86/thinkpad_acpi.c#L9642)). For a few more details see: https://github.com/linrunner/TLP/issues/321#issuecomment-575653531