linrunner/TLP #425: "powersupersave" not listed as available ASPM method

adrfantini, 2019-08-12:
The default configuration for TLP in Arch states:
```
# PCI Express Active State Power Management (PCIe ASPM):
#   default(*), performance, powersave.
# (*) keeps BIOS ASPM defaults (recommended)
# Default: <none>
#PCIE_ASPM_ON_AC=default
#PCIE_ASPM_ON_BAT=default
```
On my system, the setting `powersupersave` (which is not included in the possible values) also works for me. Is there any specific reason why it is not listed?

Thanks for the awesome tool by the way!

linrunner, 2019-08-12:
Because nobody told me about it? ;-)

adrfantini, 2019-08-12:
> Because nobody told me about it? ;-)
> 
> ps. for many parameters TLP doesn't check for allowed values, because the kernel already does.

Excellent, happy to have been of some help!

linrunner, 2019-08-13:
Done. Plus tlp-stat displays available methods now.

adrfantini, 2019-08-13:
Neat, thanks! Feel free to close.

closed 2019-09-26 by linrunner