linrunner/TLP #426: Runtime PM for Intel 9260 disables internet access

adrfantini, 2019-08-12:
I have a Dell XPS15 9550 in which I installed an Intel 9260 wifi card, using the iwlwifi driver and NetworkManager and kernel 5.2.8.
If using TLP, It needs to be explicitly blacklisted in the runtime power management variable:
```
# Runtime Power Management for PCI(e) bus devices: on=disable, auto=enable.
# Default: <none>
RUNTIME_PM_ON_AC=on
RUNTIME_PM_ON_BAT=auto
#...

RUNTIME_PM_BLACKLIST="01:00.0 02:00.0"
#01:00.0 is the GPU
#02:00.0 is the Intel 9260 wifi card
```

Otherwise I get extremely erratic internet: mostly no internet access, but also wifi disconnects and usually it is impossible to reconnect. The wifi applet in my Plasma panel also becomes very unresponsive.

I doubt this is a TLP issue, but due to the fact that this wifi card is very common I thought I would make this known. Maybe `iwlwifi` could be added to `RUNTIME_PM_DRIVER_BLACKLIST=`. I'm not sure if this is specific to my laptop or to this software combo or what.

linrunner, 2019-08-12:
Hi, 

as you're the first to report, I consider this specific to your machine. `iwlwifi`is the driver for all modern Intel wifi chipsets, don't want to blacklist it in general atm.

I'll add your case to the FAQ instead.

adrfantini, 2019-08-13:
ok, thanks! If there is any specific test I can try to figure out the exact source of the problem, let me know!

linrunner, 2019-08-13:
You may file a kernel bug report for this.

adrfantini, 2019-08-13:
Will do. I found the issue. It was the combination of enabled runtime PM and `d0i3_disable=0` module option.
It was already reported on the kernel bug tracker, and it looks like the d0i3_disable feature is going to be removed anyway: https://bugzilla.kernel.org/show_bug.cgi?id=204113

Feel free to close.

linrunner, 2019-08-13:
OK. No FAQ entry then.

closed 2019-08-13 by linrunner