linrunner/TLP #427: "ThinkPad battery features not available" on Thinkpad T490

jerabaul29, 2019-08-20:
This is a bug report opened following recommendation form the discussion here: https://askubuntu.com/questions/1166885/protect-battery-on-thinkpad-t490-ubuntu-18-04 .

On Ubuntu 18.04, with a thinkpad T490:

```
~$ sudo tlp setcharge START_CHARGE 70
Error: ThinkPad battery features not available.
```

I think this is due to problems with tp-sampi and / or tpacpi-bat:

```
~$ sudo tlp stat
...
+++ ThinkPad Battery Features
tp-smapi   = inactive (kernel module 'tp_smapi' load error)
tpacpi-bat = inactive (kernel module 'acpi_call' load error)
```

All I want is protect my battery ie let it rest between 70 and 80 percents when on external power supply.

linrunner, 2019-08-20:
Hi,

please read the [Issue template](https://github.com/linrunner/TLP/blob/master/.github/ISSUE_TEMPLATE/bug_report.md) and the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/ISSUE_TEMPLATE/bug_report.md) *before* actually submitting an issue and give all information required by the template.

You'll find suggestions für module "load error" in the FAQ. In your case I guess secure boot is the culprit.

This is not an issue with TLP. Please seek help in adequate forums.

Btw: opening an issue was not suggested at Reddit.

linrunner, 2019-08-20:
You're running on kernel 5.0 --> Install TLP 1.2.2 from the PPA and get charge thresholds without the hassle of external kernel modules :-)

jerabaul29, 2019-08-20:
Sorry, is it better description? :)

linrunner, 2019-08-20:
The problem is not the issue title. Did you actually read what I've written?

jerabaul29, 2019-08-20:
Yes, and I am sorry but it was confusing for me which point you wanted some update on / where I was not following the guidelines :) I know very little about those low level things and I must recognize I am a bit afraid of damaging some stuff on my installs.

Thank you for your help. So you mean I should do a:

```
sudo add-apt-repository ppa:linrunner/tlp
sudo apt-get update 
 sudo apt-get install tp-smapi-dkms acpi-call-dkms 
```

right?

jerabaul29, 2019-08-20:
This has worked, many thanks :) Now I get:

```
~$ sudo tlp-stat
--- TLP 1.2.2 --------------------------------------------

...

+++ Battery Features: Charge Thresholds and Recalibrate
natacpi    = active (data, thresholds)
tpacpi-bat = inactive (kernel module 'acpi_call' load error)
tp-smapi   = inactive (ThinkPad not supported)

...

```

And I can do:

```
~$ sudo tlp setcharge START_CHARGE 70
Setting temporary charge thresholds for BAT0:
  stop  =  70

~$ sudo tlp setcharge STOP_CHARGE 80
Setting temporary charge thresholds for BAT0:
  stop  =  80
```

full output: https://gist.github.com/jerabaul29/62f642ebd79340271e16b1b35c0a02b5

jerabaul29, 2019-08-20:
However, as you see this is strange: the ```START``` tlp command still seems to set up the ```stop``` value. Should I do something else here / open a bug report?

linrunner, 2019-08-20:
The correct command syntax is

     sudo tlp setcharge 70 80

See the example in the [docs](https://linrunner.de/en/tlp/docs/tlp-linux-advanced-power-management.html#thresholds). And remember to enter the thresholds into the [configuration](https://linrunner.de/en/tlp/docs/tlp-configuration.html#chargethresholds).

jerabaul29, 2019-08-20:
Thanks, I had mis-understood the man page.

closed 2019-08-20 by jerabaul29

jerabaul29, 2019-08-20:
FYI @linrunner , I hope now that no other n00bs like me will disturb you :)

https://askubuntu.com/questions/1166885/protect-battery-on-thinkpad-t490-ubuntu-18-04/1167078#1167078

linrunner, 2019-08-20:
> In your case I guess secure boot is the culprit.

If you want recalibration too, try to disable Secure Boot and the acpi_call module will probably load,

jerabaul29, 2019-08-20:
Hmm, I would loke to stay away from those Secure Boot parameter changes... Is it a lot I loose if not disabling it? :)

Thank you again for your help :) .