linrunner/TLP #429: No BAT0 detected on AC power and no switch to BAT0

julpec, 2019-08-20:
Hello

BAT0 (internal) is only detected on battery mode. If I plug the AC power, BAT0 disappear (from tlp stat and from Gnome-power-statistics).
So, I can't recalibrate BAT0 since it need to be on AC power…
Furthermore, when BAT1 is low, the laptop shutdown unexpectedly (it doesn't switch to BAT0).

**Expected behavior**
Both battery should be visible and monitored on AC power. When BAT1 is at 5%, it should switch to BAT0

[tlp-stat on battery](https://gist.github.com/julpec/30e38d9664320bb001fb991e10365c00)
[tlp-stat on AC](https://gist.github.com/julpec/d203232e4fdc46b28af0ee9a7da9e8b9)

linrunner, 2019-08-21:
Hi, 

i'm afraid this is probably a hardware issue. 
> If I plug the AC power, BAT0 disappear
> when BAT1 is low, the laptop shutdown unexpectedly (it doesn't switch to BAT0).

If you can't turn on the ThinkPad with BAT1 removed or it cuts off when you unplug the power supply without BAT1, that would be a sure indication.

I suggest you seek advice in a (ThinkPad) hardware oriented forum.

julpec, 2019-08-21:
> If you can't turn on the ThinkPad with BAT1 removed or it cuts off when you unplug the power supply without BAT1, that would be a sure indication.

Yes it is :(
Holly shit, it a almost new battery…

Thanks for the advice and sorry for the inconvenience

closed 2019-08-21 by linrunner