linrunner/TLP #430: Errors in tlp-stat output in battery section

inksnnov, 2019-08-22:
--- TLP 1.2.2 --------------------------------------------

+++ Configured Settings: /etc/default/tlp
TLP_ENABLE=1
TLP_DEFAULT_MODE=AC
TLP_PERSISTENT_DEFAULT=0
DISK_IDLE_SECS_ON_AC=0
DISK_IDLE_SECS_ON_BAT=2
MAX_LOST_WORK_SECS_ON_AC=15
MAX_LOST_WORK_SECS_ON_BAT=60
CPU_HWP_ON_AC=balance_performance
CPU_HWP_ON_BAT=balance_power
SCHED_POWERSAVE_ON_AC=0
SCHED_POWERSAVE_ON_BAT=1
NMI_WATCHDOG=0
ENERGY_PERF_POLICY_ON_AC=performance
ENERGY_PERF_POLICY_ON_BAT=power
DISK_DEVICES="nvme0n1 sda"
DISK_APM_LEVEL_ON_AC="254 254"
DISK_APM_LEVEL_ON_BAT="128 128"
SATA_LINKPWR_ON_AC="med_power_with_dipm max_performance"
SATA_LINKPWR_ON_BAT="med_power_with_dipm min_power"
AHCI_RUNTIME_PM_TIMEOUT=15
RADEON_POWER_PROFILE_ON_AC=default
RADEON_POWER_PROFILE_ON_BAT=default
RADEON_DPM_STATE_ON_AC=performance
RADEON_DPM_STATE_ON_BAT=battery
RADEON_DPM_PERF_LEVEL_ON_AC=auto
RADEON_DPM_PERF_LEVEL_ON_BAT=auto
WIFI_PWR_ON_AC=off
WIFI_PWR_ON_BAT=on
WOL_DISABLE=Y
SOUND_POWER_SAVE_ON_AC=0
SOUND_POWER_SAVE_ON_BAT=1
SOUND_POWER_SAVE_CONTROLLER=Y
BAY_POWEROFF_ON_AC=0
BAY_POWEROFF_ON_BAT=0
BAY_DEVICE="sr0"
RUNTIME_PM_ON_AC=on
RUNTIME_PM_ON_BAT=auto
USB_AUTOSUSPEND=1
USB_BLACKLIST_BTUSB=0
USB_BLACKLIST_PHONE=0
USB_BLACKLIST_PRINTER=1
USB_BLACKLIST_WWAN=0
RESTORE_DEVICE_STATE_ON_STARTUP=0
NATACPI_ENABLE=1
TPACPI_ENABLE=0
TPSMAPI_ENABLE=0

+++ System Info
System         = LENOVO ThinkPad L380 20M5001YRT
BIOS           = R0RET39W (1.22 )
Release        = Ubuntu 18.04.3 LTS
Kernel         = 5.0.0-25-generic #26~18.04.1-Ubuntu SMP Thu Aug 1 13:51:02 UTC 2019 x86_64
/proc/cmdline  = BOOT_IMAGE=/boot/vmlinuz-5.0.0-25-generic root=UUID=f73754ee-1e18-4db8-afbe-6d6562796968 ro quiet splash vt.handoff=1
Init system    = systemd v237
Boot mode      = UEFI

+++ TLP Status
State          = enabled
RDW state      = enabled
Last run       = 22:30:51,     24 sec(s) ago
Mode           = AC
Power source   = AC

+++ Processor
CPU model      = Intel(R) Core(TM) i5-8250U CPU @ 1.60GHz

/sys/devices/system/cpu/cpu0/cpufreq/scaling_driver    = intel_pstate
[... 185 lines cut ...]
/sys/class/power_supply/BAT0/energy_now                     =  37700 [mWh]
/sys/class/power_supply/BAT0/power_now                      =      0 [mW]
/sys/class/power_supply/BAT0/status                         = Unknown (threshold effective)

/sys/class/power_supply/BAT0/charge_start_threshold         =     80 [%]
/sys/class/power_supply/BAT0/charge_stop_threshold          =     85 [%]

Charge                                                      =   83.9 [%]
Capacity                                                    =   99.9 [%]

+++ ThinkPad Battery Status: hid-00:1f:20:e4:e8:df-battery
/sys/class/power_supply/hid-00:1f:20:e4:e8:df-battery/manufacturer = (not available)
/sys/class/power_supply/hid-00:1f:20:e4:e8:df-battery/model_name = Bluetooth Mouse M557
/sys/class/power_supply/hid-00:1f:20:e4:e8:df-battery/cycle_count = (not supported)
/sys/class/power_supply/hid-00:1f:20:e4:e8:df-battery/status = Discharging

/usr/bin/tlp-stat: 349: [: -eq: unexpected operator
                                                            =      0 [%]
/usr/bin/tlp-stat: 349: [: -eq: unexpected operator
                                                            =      0 [%]

+++ Charge total                                            =   83.9 [%]

+++ Recommendations
* Install smartmontools for disk drive health info

linrunner, 2019-08-26:
Hi, thanks for your report. Be so kind as to give me the files in that directory:

    grep '.*' /sys/class/power_supply/hid-00:1f:20:e4:e8:df-battery/*

inksnnov, 2019-08-29:
Hi
grep '.*' /sys/class/power_supply/hid-00:1f:20:e4:e8:df-battery/*
/sys/class/power_supply/hid-00:1f:20:e4:e8:df-battery/capacity:35
grep: /sys/class/power_supply/hid-00:1f:20:e4:e8:df-battery/device: Это каталог
/sys/class/power_supply/hid-00:1f:20:e4:e8:df-battery/model_name:Bluetooth Mouse M557
/sys/class/power_supply/hid-00:1f:20:e4:e8:df-battery/online:1
grep: /sys/class/power_supply/hid-00:1f:20:e4:e8:df-battery/power: Это каталог
grep: /sys/class/power_supply/hid-00:1f:20:e4:e8:df-battery/powers: Это каталог
/sys/class/power_supply/hid-00:1f:20:e4:e8:df-battery/present:1
/sys/class/power_supply/hid-00:1f:20:e4:e8:df-battery/scope:Device
/sys/class/power_supply/hid-00:1f:20:e4:e8:df-battery/status:Discharging
grep: /sys/class/power_supply/hid-00:1f:20:e4:e8:df-battery/subsystem: Это каталог
/sys/class/power_supply/hid-00:1f:20:e4:e8:df-battery/type:Battery
/sys/class/power_supply/hid-00:1f:20:e4:e8:df-battery/uevent:POWER_SUPPLY_NAME=hid-00:1f:20:e4:e8:df-battery
/sys/class/power_supply/hid-00:1f:20:e4:e8:df-battery/uevent:POWER_SUPPLY_PRESENT=1
/sys/class/power_supply/hid-00:1f:20:e4:e8:df-battery/uevent:POWER_SUPPLY_ONLINE=1
/sys/class/power_supply/hid-00:1f:20:e4:e8:df-battery/uevent:POWER_SUPPLY_CAPACITY=35
/sys/class/power_supply/hid-00:1f:20:e4:e8:df-battery/uevent:POWER_SUPPLY_MODEL_NAME=Bluetooth Mouse M557
/sys/class/power_supply/hid-00:1f:20:e4:e8:df-battery/uevent:POWER_SUPPLY_STATUS=Discharging
/sys/class/power_supply/hid-00:1f:20:e4:e8:df-battery/uevent:POWER_SUPPLY_SCOPE=Device

linrunner, 2019-08-30:
Thanks, fixed.

Would you please test and report? You may download packages [here](https://download.linrunner.de/packages/).

linrunner, 2020-01-17:
Please re-test with 1.3.0-beta.3 and report.
Instructions: https://github.com/linrunner/TLP/issues/449

closed 2020-01-17 by linrunner