linrunner/TLP #431: T440p battery not discharging

mightyBroccoli, 2019-08-30:
[x] I've read and accepted the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md)

**Describe the bug**
My T440p with the latest ArchLinux is not discharging at all. I installed tlp, tlp-rdw and tpacpi-bat without problems.
tlp-stat -b is showing that natacpi and tpacpi-bat are loaded but none of them list discharge as a feature.
Thus the stop_charge thresholds are not adopted. If the stop threshold is reached the status switches to Unknown.

**Expected behavior**
I would like my T440p to utilize the configured start/stop thresholds to not burn to yet another laptop battery.

**To Reproduce**
- T440p
- ArchLinux
- install tlp, tlp-rdw and tpacpi-bat
- profit?

1. Does the problem occur on battery or AC or both?
This could only occur while the laptop is connected to the AC.

2. Actions to reproduce the behavior
nothing specific, tlp is started normally.

3. Shell commands entered and their output
[BAT0 charging](https://gist.github.com/mightyBroccoli/2d5b03c7008eff1a94c191dd78ebf4ea)
[stop_threshold reached](https://gist.github.com/mightyBroccoli/b70c1583f145a35493e0dc38e7bfabc8)

linrunner, 2019-08-30:
I think you've fundamentally misunderstood the purpose of charge thresholds.

Charging stops at
> Charge                                                      =   84.5 [%]
e.g. approx. 85%. That's exactly what the stop threshold should do. The output of tlp-stat indicates why: 
> /sys/class/power_supply/BAT0/status                         = Unknown (threshold effective)

For comprehension please search help in a ThinkPad or hardware oriented forum, the issue tracker is intended for bug reports only. Thanks for your understanding.

closed 2019-08-30 by linrunner

mightyBroccoli, 2019-08-30:
The whole question was not about charge thresholds. It was about the fact that somehow discharging is not happening and natacpi not listing "discharge" in its feature set.
```
natacpi    = active (data, thresholds)
```

linrunner, 2019-08-30:
Discharge is not just "happening". It's initiated by

    tlp discharge

or 

   tlp recalibrate

which you didn't issue (or did not tell about)

tlp-stat indeed indicates the feature:

> tpacpi-bat = active (**recalibrate**)

mightyBroccoli, 2019-08-30:
Am I correct about the fact that tlp instructs the system when to start charging ( start_threshold) and when to stop charging (stop_threshold) until the start charging percentage is reached again?

I know that I can manually invoke this, and they both work but why doesn't it work automatically. That is the reason why one would install something like TLP.

My problem is that the stop_charging threshold is reached, but then the device holds that charge stating the the status is unknown.
I can manually discharge the battery with `tlp discharge` and ccyle the battery with `tlp recalibrate` just the automatic behavior is nonfunctional.

linrunner, 2019-08-30:
OMG: *severe* misunderstanding. Why should the battery commence discharging after reaching the stop threshold? Do you really suppose the battery should constanty wear in an infinite loop between start --> stop --> start thresholds? Please rethink this. The purpose of the stop threshold is to limit max. charge, the start threshold prevents small charge cycles.

And again: the issue tracker is not a support forum, nor do I want to teach the basics here.

mightyBroccoli, 2019-08-30:
Yes you are exactly right here. You could describe this in your FAQ and there is no reason to be that conceited in a friendly conversation.

But besides that, on Windows Lenovo actually defaults to this behavior, to cycle the battery while attached to the AC.
Additionally the word threshold implicates that something happens up until said threshold. Thus one could assume this something stops when the threshold is reached.

I though this is not the intentional behavior, thus posting this here.

linrunner, 2019-08-30:
> But besides that, on Windows Lenovo actually defaults to this behavior, to cycle the battery while attached to the AC.

It doesn't on Windows. Period. The behaviour you expect would cause absurd wear, as explained above. Besides, the FAQ already states

> **Do charge thresholds work even when Linux/TLP is not running?** Yes. The charging process is not controlled by software, but by hardware. TLP just writes the thresholds to the hardware registers (via tp-smapi, tpacpi-bat or natacpi). Once stored in the hardware the charge thresholds stay effective permanently.

The behaviour is hardcoded into the EC firmware and the only parameters you can change are the thresholds.

linrunner, 2019-09-09:
[FAQ done](https://linrunner.de/en/tlp/docs/tlp-faq.html#discharge-misconception).