linrunner/TLP #434: tlp-sleep service fails in 1.2.2 with Fedora 30

mkowalski, 2019-09-20:
### Summary
Using Fedora 30 and `tlp-1.2.2-1.fc30`, execution of `/usr/sbin/tlp suspend` fails. The same command works correctly when using `tlp-1.2.2-1.fc30`. Linked below are straces for respectively old/working [1] and new/broken [2] versions.

Please note the following difference - in the new version a path where tlp looks for `rfkill_saved` seems to be changed from `/run/tlp/...` to `/var/lib/tlp/...`, however the latter does not exist in the underlying OS.

This kinda makes sense given the recent [3] introduction of the `TLP_VAR` defaulting to `/var/lib/tlp`.

### Debug
#### TLP 1.2.1
```
openat(AT_FDCWD, "/run/tlp/rfkill_saved", O_WRONLY|O_CREAT|O_TRUNC, 0666) = 3
```

#### TLP 1.2.2
```
openat(AT_FDCWD, "/var/lib/tlp/rfkill_saved", O_WRONLY|O_CREAT|O_TRUNC, 0666) = -1 ENOENT (No such file or directory)
```

### Links
[1] https://bugzilla.redhat.com/attachment.cgi?id=1617294
[2] https://bugzilla.redhat.com/attachment.cgi?id=1617295
[3] https://linrunner.de/en/tlp/docs/tlp-developer-documentation.html

WillNilges, 2019-09-21:
I'm getting this issue too. For clarification, what do you mean by
> Using Fedora 30 and tlp-1.2.2-1.fc30, execution of /usr/sbin/tlp suspend fails. The same command works correctly when using tlp-1.2.2-1.fc30.

What version was working correctly again?

mkowalski, 2019-09-21:
Obviously a typo, 1.2.2 is broken whereas 1.2.1 is working correctly

linrunner, 2019-09-21:
Hi, thanks for your extensive analysis. From what it looks this could be two problems.

(1) Directory **/var/lib/tlp** 

TLP assumes that this is created by the package install, as `make install` does. I guess Fedora's  [specfile](https://gist.github.com/linrunner/7cf7a6b2ea901595990bb91820ba12c1#file-tlp-1-2-2-2-fc31-spec-L82) lacks a line like

    %{_sharedstatedir}/tlp

in the %files section. This must be fixed in the Fedora package. I'll prompt Jeremy about this.

But this doesn't explain why **tlp-sleep.service** fails. **tlp-sleep.service** doesn't touch **/var/lib/tlp/rfkill_saved** (btw the only file in that directory), this is done upon system boot / shutdown only e.g. when tlp.service is called. tlp-sleep.service writes in **/var/run/tlp/** only. 

@WillNilges post points in this direction too.

So I guess there is a

(2) SELinux issue with **/var/run/tlp** (or elsewhere)

I suggest you create **/var/lib/tlp manually**, check for AVCs and file a Fedora bug against **selinux-policy** if necessary. There's been quite a few of that.

linrunner, 2019-09-21:
ps. **tlp-sleep.service** in fact writes to **/var/lib/tlp/rfkill_saved** – unnecessarily.

linrunner, 2019-09-21:
I prompted Jeremy the maintainer of the Fedora package about this

linrunner, 2019-09-21:
The [openSUSE package](https://build.opensuse.org/package/view_file/openSUSE:Factory/tlp/tlp.spec) has line #119 for **/var/lib/tlp**:

    %{_localstatedir}/lib/%{name}/

ondrejbudai, 2019-10-30:
When I manually create /var/lib/tlp I got the following errors in `journalctl` on suspend:
```
localhost.localdomain audit[8839]: AVC avc:  denied  { write } for  pid=8839 comm="tlp" name="rfkill_saved" dev="dm-1" ino=655667 scontext=system_u:system_r:tlp_t:s0 tcontext=unconfined_u:object_r:var_lib_t:s0 tclass=file permissive=0
localhost.localdomain audit[8839]: AVC avc:  denied  { write } for  pid=8839 comm="tlp" name="rfkill_saved" dev="dm-1" ino=655667 scontext=system_u:system_r:tlp_t:s0 tcontext=unconfined_u:object_r:var_lib_t:s0 tclass=file permissive=0
```

adatum, 2019-11-17:
@ondrejbudai Thanks, I was getting the `tlp-sleep.service` failed state too, and https://github.com/linrunner/TLP/issues/434#issuecomment-547766196 looks like it solves the problem. I'm running Fedora 31 on a Dell E7250.

eraser215, 2019-12-11:
I also had the same problem on F31 running on a Lenovo T470s, and it was also resolved by https://github.com/linrunner/TLP/issues/434#issuecomment-547766196

Thanks for the work-around!

Mystro256, 2020-01-03:
Hey sorry, looks like my email junk filter has been filtering a lot of these emails, along with life getting in the way. I'll add %{_sharedstatedir}/tlp and push a build out ASAP.

linrunner, 2020-01-20:
Solved with package update tlp-1.2.2-4.

closed 2020-01-20 by linrunner