linrunner/TLP #436: New USB devices not recognized on T495 in battery mode

Jakeler, 2019-09-22:
**Describe the bug**
When my Thinkpad T495 runs on battery (or TLP is in manual battery mode, set with `tlp bat`) and i plug in a new USB device, it does not get recognized. Also there is nothing visible in `lsusb` or `journalctl`/`dmesg`. The USB device only charges then, like on a simple power supply.
Already connected devices keep working though, but they are also gone when shortly disconnected and reconnected, same as described above.
This only happens if all (external) USB devices are disconnected at some point. If at least one is kept connected all the time, then new devices (on other ports) show up and work fine.

USB autosuspend was enabled first, but it makes no difference when it is disabled (with `USB_AUTOSUSPEND=0`).

On AC everything works as expected, that is the only way how i can currently use USB, without disabling TLP completely.

All USB ports (USB-C 3.1, USB-A 3.0) on the laptop are affected.

**To Reproduce**
- Make TLP go into battery mode
- Unplug all USB devices
- Plug in an USB device again
- Nothing gets recognized anymore...

Full output of `tlp-stat` for AC and battery:
https://gist.github.com/Jakeler/5bbe260898e7920f99c689a954299b75
You can see that my Logitech mouse is missing in the battery stat (i replugged it to test it). 
Looking over both with `icdiff` shows that many PCIe device power controls change from on to auto. Maybe it has something to do with the PCIe USB controller going into sleep mode?

linrunner, 2019-09-22:
Hi,

you may isolate the cause by following the [Troubleshooting Guide](https://linrunner.de/en/tlp/docs/tlp-troubleshooting.html).

My prime suspect would be the PCIe devices e.g. Runtime PM.

Jakeler, 2019-09-22:
Thanks, it was indeed PCIe Runtime PM. I blacklisted the USB controllers with:
```
RUNTIME_PM_BLACKLIST="06:00.3 06:00.4"
```
and it works!

closed 2019-09-22 by Jakeler

gsalgado, 2021-03-22:
Would it make sense to mention this issue in the docs somewhere?  This has caused me a lot of headaches but since I hadn't noticed it would only happen when running on battery my google searches never led me here...

linrunner, 2021-03-22:
@gsalgado : your wish is granted. Thanks for the hint. 

https://linrunner.de/tlp/faq/usb.html#newly-inserted-usb-devices-not-recognized-in-battery-mode

ShashaankHari, 2022-02-10:
> Thanks, it was indeed PCIe Runtime PM. I blacklisted the USB controllers with:
> 
> RUNTIME_PM_BLACKLIST="06:00.3 06:00.4"
> 
> and it works!

Does this work on other Lenovo devices like IdeaPad? I am having the same issues if i turn on TLP, with usb auto suspend off, still usb doesn't work on battery

SohamG, 2024-04-29:
Hello, sorry to bump an old thread but this just bit me right now at a not-so-convenient time. I am on NixOS 23.11, and I think the commit above hasn't made it to me yet. Do you think its appropriate to suggest this config change as a sane-default till a new tlp release is out? I wouldn't mind opening a PR with my distro for this.

linrunner, 2024-04-30:
@SohamG Why don't you just add it to _your_ configuration?

SohamG, 2024-04-30:
Oh i was just hoping to make a contribution so that others don't have this
issue. I've submitted packages there before so I know the workflow.

Soham Gumaste
***@***.***

linrunner, 2024-05-01:
@SohamG From an upstream support perspective, it is not very helpful when distributions prematurely include individual changes outside of releases. Then I don't know exactly what I'm dealing with in the issue tracker.

If you want to contribute more extensively, then a NixOS package that reflects the HEAD of the main branch would be great. Something like this exists for Arch Linux: 

https://aur.archlinux.org/packages/tlp-git
https://aur.archlinux.org/packages/tlp-rdw-git

I should probably do another release ...