linrunner/TLP #438: Cannot recalibrate battery

mindrunner, 2019-10-05:
[x] I've read and accepted the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md)

**Describe the bug**

I ma trying to recalibrate a battery. TLP starts with the discharging, but after a couple of seconds stops with the message, that the battery was fully discharged (which is not the case):
```
Currently discharging battery BAT0:
voltage            =  11970 [mV]
remaining capacity =  47110 [mWh]
remaining percent  =     53 [%]
remaining time     = not discharging [min]
state              = Discharging
force discharge    = 0
Press Ctrl+C to cancel.

Done: battery BAT0 was completely discharged.
Charging starts now, for a complete recalibration
keep AC connected until the battery is fully charged.
```

**Expected behavior**

Battery fully discharges as explained in TLP docs.

**To Reproduce**

Steps to reproduce the unexpected behavior:

1. sudo tlp recalibrate
2. wait

**Additional context**

```
[le@w530]: ~>$ sudo tlp-stat -v -b 
--- TLP 1.2.2 --------------------------------------------

+++ Battery Features: Charge Thresholds and Recalibrate
natacpi    = active (data, thresholds)
tpacpi-bat = active (recalibrate)
tp-smapi   = inactive (ThinkPad not supported)

+++ ThinkPad Battery Status: BAT0 (Main / Internal)
/sys/class/power_supply/BAT0/manufacturer                   = SANYO
/sys/class/power_supply/BAT0/model_name                     = 45N1023
/sys/class/power_supply/BAT0/cycle_count                    = (not supported)
/sys/class/power_supply/BAT0/energy_full_design             =  73260 [mWh]
/sys/class/power_supply/BAT0/energy_full                    =  87430 [mWh]
/sys/class/power_supply/BAT0/energy_now                     =  46010 [mWh]
/sys/class/power_supply/BAT0/power_now                      =  30150 [mW]
/sys/class/power_supply/BAT0/status                         = Charging

/sys/class/power_supply/BAT0/voltage_min_design             =  11100 [mV]
/sys/class/power_supply/BAT0/voltage_now                    =  11477 [mV]

/sys/class/power_supply/BAT0/charge_start_threshold         =     96 [%]
/sys/class/power_supply/BAT0/charge_stop_threshold          =    100 [%]
tpacpi-bat.BAT0.forceDischarge                              =      0

Charge                                                      =   52.6 [%]
Capacity                                                    =  119.3 [%]
```

linrunner, 2019-10-05:
The actual discharge process is only started by `tlp discharge` and then controlled by the Thinkpad's charging circuit, which also decides when discharge ends and then resets the control register. `tlp discharge` just waits for this reset and then terminates. TLP can't force discharge when the charging circuit has a different opinion.

Conclusion: I think this is a hardware issue, either the battery or the system board. I can't help you with that. I suggest you try another battery

mindrunner, 2019-10-05:
Tried with anothter battery and got this:
```
[le@w530]: ~>$ sudo tlp recalibrate
Setting temporary charge thresholds for BAT0:
  start =  96 (no change)
  stop  = 100 (no change)
Initiating discharge of battery BAT0 ...............
Error: discharge BAT0 malfunction.
```

So you are probably right. Time to replace my good old friend...

closed 2019-10-06 by mindrunner

tadawson, 2024-09-02:
A belated (but seemingly valid) data point on this.

I have a W520, and while on Slackware 14.2 (4.x kernel, TLP 1.6.0 or 1.6.1, iirc) this had always worked, either from Linux or Windows.

I updated to Slackware 15 (5.15.161 kernel, TLP 1.6.1) and now getting this failure. (Firmware terminates discharge, seemingly at 20%)

Updated to 5.17.19 kernel and TLP 1.7.0 alpha, still get this error . . . was thinking bad battery . . .

*BUT*

Windows (with the Lenovo tool) can still recalibrate this battery fine, and can discharge it to zero consistently without error, clearly screaming that this is *NOT* a battery or EC issue, but a change in kernel/TLP issue.

I have not had a chance, but may revert to my SL14.2 drive and verify there as well.

FWIW . . .

tadawson, 2024-09-03:
An additional insight that tests to verify that this is *NOT* a battery issue or error . . . 

I note that on this device Lenovo W520) with 1.7.0a and natacpi, that the recall is triggered by simply writing "force-discharge" to /sys/class/power_devices/BAT0/charge_behaviour with a fully charged battery. After seeing 100% fail of recal using tlp, I triggered the recall manually, and sat back and watched.

The recal ran *perfectly* with no glitches or issues of any kind! 

This screams to me that something in the monitoring loop is getting bad data and focing the abort in tlp - because the EC certainly is not aborting!

(I did note that when I had tested in Windows, that when things got lower, that occasiinally some parameters reporting would be absent for a polling cycle . . . I wonder if this is a factor? The monitor fails a read and forces an abort.)

Watching things run, I'm frankly unclear as to why the recal monitor watches anything *but* charge_behaviour . . . . If that doesn't toggle back to "auto", it's still running, and the percentage is irrelevant . . .

tadawson, 2024-09-03:
FYI, I appear to have found a cause and (potential) solution for this, documented in issje #759.