linrunner/TLP #441: Question regarding T480s charge threshold limitation

grssnbchr, 2019-10-30:
I don't have a bug report, merely a question. I have a T480s and apparently suffer from the limitation described [here](https://linrunner.de/en/tlp/docs/tlp-faq.html#stop-threshold-sticks). However, my BIOS already *is* at 1.31, as the output from `tlp-stat` shows:

```
System         = LENOVO ThinkPad T480s 20L7001NMZ
BIOS           = N22ET54W (1.31 )
Release        = Ubuntu 18.04.3 LTS
```

So the BIOS that is mentioned on the troubleshooting page is the one I have:

```
Solution: update EC firmware (contained in BIOS update)

    T480s: ECP 1.13 (BIOS 1.31)
    X1 Carbon 6th: ECP 1.12 (BIOS 1.37)
```

Does this mean I have to update to one *above 1.31* or should it theoretically already work with 1.31? If the latter is the case, I need to go into debugging, but if not, I will update my firmware of course.

linrunner, 2019-10-30:
These versions are specified as **solution**, isn't that clear enough?

The important part is the EC firmware (ECP) version. Did you check it on the first BIOS page?

ECP 1.13 was [reported](https://github.com/linrunner/TLP/issues/401#issuecomment-496374950) by a user, Lenovo says it's contained in BIOS 1.31. 

Alas, I wouldn't fool around for long and update straight to BIOS 1.34 w/ ECP 1.17 instead. Eith fwupdmgr you can do it in the twinkling of an eye.

grssnbchr, 2019-10-31:
Updating to BIOS 1.34 did the trick. I suspect that 1.31 is not enough, that's why I was confused. Anyway, thank you for the hint with `fwupdmgr`.

closed 2019-10-31 by grssnbchr