linrunner/TLP #444: Error: battery charge thresholds not available.

yusyel, 2019-11-19:
I tried the thresholds for latitude 5400 battery. This is the output after "sudo tlp setcharge 70 95 bat0"
Same result, TLP start ac/battery mode.
I followed the Installation & Configuration.
fedora 31

linrunner, 2019-11-19:
Battery thresholds are for ThinkPads only. That fact is stated all over the documentation, dear.

closed 2019-11-19 by linrunner

yusyel, 2019-11-19:
yeah. I should read more carefuly the FAQ. There is a Bat0 and Bat1. I I thought that was for thinkpad only. Thinkpads comes with two battery.

rien333, 2020-01-23:
@linrunner My laptop is not a thinkpad, but I do have `/sys/class/power_supply/BAT0/charge_control_{start,end}_threshold` available. Should tlp support this? (I understand if it currently doesn't, but it seems reasonable to support it at one point) 

Currently, setting the threshold through tlp outputs the following:
```
# tlp setcharge 20 90 BAT0
Error: battery charge thresholds not available.
```

linrunner, 2020-01-31:
@rien333 : hardware support does not develop by itself, no matter how often you ask. Follow the discussion [here](https://github.com/linrunner/TLP/issues/321) and help with testing – or contribute the necessary code and send a pull request.

rien333, 2020-01-31:
Thanks, I wasn't asking whether you would implement it though, but more if it's in the scope  of what tlp aims to support. From your answer, I gather it is.

awehrfritz, 2020-01-31:
> @linrunner My laptop is not a thinkpad, but I do have `/sys/class/power_supply/BAT0/charge_control_{start,end}_threshold` available. Should tlp support this?

@rien333 sounds like you have a Huawei laptop. These sysfs attributes have only been recently implemented in the kernel for Huawei and Asus laptop, following the initial work by Google for their Wilco EC (see my posts [here](https://github.com/linrunner/TLP/issues/321#issuecomment-575653531) and [here](https://github.com/linrunner/TLP/issues/321#issuecomment-576059832) for details).

Thinkpads have had battery kernel drivers for probably more than a decade now, even my Thinkpad T500 from around 2008 has had support for setting battery charging thresholds and other things already back then. And they continue to have the best Linux support in this regard even nowadays, though Dell and others are closing in slowly. (Dell does not have kernel support for their batteries, but they provide this in their [libsmbios](https://github.com/dell/libsmbios) library.)

TLP is mostly shell scripts, so implementing and testing support for these new sysfs attributes is not overly difficult (i.e. it doesn’t require you to use a compiler or anything the like). Though I have to say that TLP is a bit involved at times, especially when it comes to determining what kernel drive to pick or what that drive supports and has a lot of safeguards around that - this makes it probably a bit daunting for newbies.

linrunner, 2020-02-01:
@non-Thinkpads: *please* stop posting here. Join the discussion: [here](https://github.com/linrunner/TLP/issues/321)