linrunner/TLP #446: TLP do not change mode on powerchange

nnmlss, 2019-12-07:
**Describe the bug**

When I plug in the adapter the result of  `sudo tlp stat -s` is:

>+++ TLP Status
State          = enabled
RDW state      = enabled
Last run       = 18:38:15,   1148 sec(s) ago
Mode           = AC
Power source   = AC

When I plug out 
>+++ TLP Status
State          = enabled
RDW state      = enabled
Last run       = 18:58:25,      5 sec(s) ago
Mode           = AC
Power source   = battery

**Expected behavior**

When there is no power adapter the tlp mode to be battery. It happens only if I type `sudo tlp start` while the adapter is unplugged. 
>TLP started in battery mode.

>sudo tlp-stat -s
--- TLP 1.2.2 --------------------------------------------

>+++ System Info
System         = HP Type1ProductConfigId HP Pavilion Power Laptop 15-cb0xx
BIOS           = F.18
Release        = elementary OS 5.1 Hera
Kernel         = 5.0.0-37-generic #40~18.04.1-Ubuntu SMP Thu Nov 14 12:06:39 UTC 2019 x86_64
/proc/cmdline  = BOOT_IMAGE=/boot/vmlinuz-5.0.0-37-generic root=UUID=xxxxxxxxxxxxxx ro quiet splash
Init system    = systemd v237
Boot mode      = UEFI

>+++ TLP Status
State          = enabled
RDW state      = enabled
Last run       = 19:01:03,      9 sec(s) ago
Mode           = battery
Power source   = battery

but when I plug the power adapter in again mode is set to AC and do not change when I unplug it again. I was searching for an option to use the OS power events and start a script with `sudo tlp start` when the power adapter is off, but as I see the tlp application detects the Power source but does not change the Mode. 

Information about my OS and hardware you can see above, the same problem was also with the older 4.x kernels.  My TLP settings at the moment are (serial numbers replaced with xxx, I can share if you need):

TLP_ENABLE=1

TLP_DEFAULT_MODE=BAT

TLP_PERSISTENT_DEFAULT=0

DISK_IDLE_SECS_ON_AC=0
DISK_IDLE_SECS_ON_BAT=2

MAX_LOST_WORK_SECS_ON_AC=15
MAX_LOST_WORK_SECS_ON_BAT=60

CPU_SCALING_GOVERNOR_ON_AC=performance
CPU_SCALING_GOVERNOR_ON_BAT=powersave

CPU_SCALING_MIN_FREQ_ON_AC=800000
CPU_SCALING_MAX_FREQ_ON_AC=3800000
CPU_SCALING_MIN_FREQ_ON_BAT=800000
CPU_SCALING_MAX_FREQ_ON_BAT=1900000

CPU_HWP_ON_AC=balance_performance
CPU_HWP_ON_BAT=balance_power

CPU_MIN_PERF_ON_AC=0
CPU_MAX_PERF_ON_AC=100
CPU_MIN_PERF_ON_BAT=0
CPU_MAX_PERF_ON_BAT=30

CPU_BOOST_ON_AC=1
CPU_BOOST_ON_BAT=0

SCHED_POWERSAVE_ON_AC=0
SCHED_POWERSAVE_ON_BAT=1

NMI_WATCHDOG=0

ENERGY_PERF_POLICY_ON_AC=performance
ENERGY_PERF_POLICY_ON_BAT=power

DISK_DEVICES="nvme-KXG50ZNV256G_TOSHIBA_xxxxxxxxxxxx ata-CT500BX100SSD1_xxxxxxxxxxxx"

DISK_APM_LEVEL_ON_AC="254 254"
DISK_APM_LEVEL_ON_BAT="128 128"

DISK_SPINDOWN_TIMEOUT_ON_AC="0 0"
DISK_SPINDOWN_TIMEOUT_ON_BAT="0 0"

SATA_LINKPWR_ON_AC="med_power_with_dipm max_performance"
SATA_LINKPWR_ON_BAT="med_power_with_dipm min_power"

SATA_LINKPWR_BLACKLIST="host1"

AHCI_RUNTIME_PM_TIMEOUT=15

PCIE_ASPM_ON_AC=performance
PCIE_ASPM_ON_BAT=powersave

INTEL_GPU_MIN_FREQ_ON_AC=350
INTEL_GPU_MIN_FREQ_ON_BAT=350
INTEL_GPU_MAX_FREQ_ON_AC=1100
INTEL_GPU_MAX_FREQ_ON_BAT=1100
INTEL_GPU_BOOST_FREQ_ON_AC=1100
INTEL_GPU_BOOST_FREQ_ON_BAT=1100

WIFI_PWR_ON_AC=off
WIFI_PWR_ON_BAT=on

WOL_DISABLE=Y

SOUND_POWER_SAVE_ON_AC=0
SOUND_POWER_SAVE_ON_BAT=0

SOUND_POWER_SAVE_CONTROLLER=Y

BAY_POWEROFF_ON_AC=0
BAY_POWEROFF_ON_BAT=0

RUNTIME_PM_ON_AC=on
RUNTIME_PM_ON_BAT=auto

USB_AUTOSUSPEND=1

USB_BLACKLIST="046d:c534"

USB_BLACKLIST_BTUSB=0

USB_BLACKLIST_PHONE=1

USB_BLACKLIST_PRINTER=1

USB_BLACKLIST_WWAN=0

USB_AUTOSUSPEND_DISABLE_ON_SHUTDOWN=1

RESTORE_DEVICE_STATE_ON_STARTUP=0

DEVICES_TO_ENABLE_ON_AC="bluetooth wifi wwan"

DEVICES_TO_DISABLE_ON_BAT="bluetooth wwan"

DEVICES_TO_DISABLE_ON_BAT_NOT_IN_USE="bluetooth wwan"

NATACPI_ENABLE=1
TPACPI_ENABLE=1
TPSMAPI_ENABLE=1

linrunner, 2019-12-07:
Your laptop apparently does not generate the necessary ACPI events when unplugging/plugging the charger. You may monitor this with

    tlp-stat --pev

Just show the output after (a) unplugging and (b) plugging AC. Please also show the output of

    tlp-stat --psup

when (a) running on battery and (b) on AC.

nnmlss, 2019-12-07:
First command cannot run without sudo - Error: missing root privilage. The second one is ok. 
======== sudo tlp-stat --pev ======
------------a----------------------
UDEV  [514.333918] change   /devices/LNXSYSTM:00/LNXSYBUS:00/PNP0C0A:00/power_supply/BAT0 (power_supply)
ACTION=change
DEVPATH=/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0C0A:00/power_supply/BAT0
POWER_SUPPLY_CAPACITY=94
POWER_SUPPLY_CAPACITY_LEVEL=Normal
POWER_SUPPLY_CHARGE_FULL=4512000
POWER_SUPPLY_CHARGE_FULL_DESIGN=4512000
POWER_SUPPLY_CHARGE_NOW=4256000
POWER_SUPPLY_CURRENT_NOW=253000
POWER_SUPPLY_CYCLE_COUNT=0
POWER_SUPPLY_MANUFACTURER=HP
POWER_SUPPLY_MODEL_NAME=Primary
POWER_SUPPLY_NAME=BAT0
POWER_SUPPLY_PRESENT=1
POWER_SUPPLY_SERIAL_NUMBER= 
POWER_SUPPLY_STATUS=Unknown
POWER_SUPPLY_TECHNOLOGY=Li-ion
POWER_SUPPLY_VOLTAGE_MIN_DESIGN=15400000
POWER_SUPPLY_VOLTAGE_NOW=17014000
SEQNUM=4424
SUBSYSTEM=power_supply
USEC_INITIALIZED=513781023

UDEV  [514.351438] change   /devices/LNXSYSTM:00/LNXSYBUS:00/ACPI0003:00/power_supply/ADP1 (power_supply)
ACTION=change
DEVPATH=/devices/LNXSYSTM:00/LNXSYBUS:00/ACPI0003:00/power_supply/ADP1
POWER_SUPPLY_NAME=ADP1
POWER_SUPPLY_ONLINE=0
SEQNUM=4425
SUBSYSTEM=power_supply
USEC_INITIALIZED=513801414
---------b-----------
DEV  [561.474144] change   /devices/LNXSYSTM:00/LNXSYBUS:00/PNP0C0A:00/power_supply/BAT0 (power_supply)
ACTION=change
DEVPATH=/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0C0A:00/power_supply/BAT0
POWER_SUPPLY_CAPACITY=94
POWER_SUPPLY_CAPACITY_LEVEL=Normal
POWER_SUPPLY_CHARGE_FULL=4512000
POWER_SUPPLY_CHARGE_FULL_DESIGN=4512000
POWER_SUPPLY_CHARGE_NOW=4256000
POWER_SUPPLY_CURRENT_NOW=117000
POWER_SUPPLY_CYCLE_COUNT=0
POWER_SUPPLY_MANUFACTURER=HP
POWER_SUPPLY_MODEL_NAME=Primary
POWER_SUPPLY_NAME=BAT0
POWER_SUPPLY_PRESENT=1
POWER_SUPPLY_SERIAL_NUMBER= 
POWER_SUPPLY_STATUS=Unknown
POWER_SUPPLY_TECHNOLOGY=Li-ion
POWER_SUPPLY_VOLTAGE_MIN_DESIGN=15400000
POWER_SUPPLY_VOLTAGE_NOW=17125000
SEQNUM=4426
SUBSYSTEM=power_supply
USEC_INITIALIZED=561459370

UDEV  [561.484662] change   /devices/LNXSYSTM:00/LNXSYBUS:00/ACPI0003:00/power_supply/ADP1 (power_supply)
ACTION=change
DEVPATH=/devices/LNXSYSTM:00/LNXSYBUS:00/ACPI0003:00/power_supply/ADP1
POWER_SUPPLY_NAME=ADP1
POWER_SUPPLY_ONLINE=1
SEQNUM=4427
SUBSYSTEM=power_supply
USEC_INITIALIZED=561469479

UDEV  [564.504611] change   /devices/LNXSYSTM:00/LNXSYBUS:00/PNP0C0A:00/power_supply/BAT0 (power_supply)
ACTION=change
DEVPATH=/devices/LNXSYSTM:00/LNXSYBUS:00/PNP0C0A:00/power_supply/BAT0
POWER_SUPPLY_CAPACITY=94
POWER_SUPPLY_CAPACITY_LEVEL=Normal
POWER_SUPPLY_CHARGE_FULL=4512000
POWER_SUPPLY_CHARGE_FULL_DESIGN=4512000
POWER_SUPPLY_CHARGE_NOW=4256000
POWER_SUPPLY_CURRENT_NOW=0
POWER_SUPPLY_CYCLE_COUNT=0
POWER_SUPPLY_MANUFACTURER=HP
POWER_SUPPLY_MODEL_NAME=Primary
POWER_SUPPLY_NAME=BAT0
POWER_SUPPLY_PRESENT=1
POWER_SUPPLY_SERIAL_NUMBER= 
POWER_SUPPLY_STATUS=Charging
POWER_SUPPLY_TECHNOLOGY=Li-ion
POWER_SUPPLY_VOLTAGE_MIN_DESIGN=15400000
POWER_SUPPLY_VOLTAGE_NOW=17141000
SEQNUM=4428
SUBSYSTEM=power_supply
USEC_INITIALIZED=564491746

=============== tlp-stat --psup =============
----------------------a----------------------
+++ Power supply diagnostic
/sys/class/power_supply/ADP1/type:Mains
/sys/class/power_supply/ADP1/online:0
/sys/class/power_supply/ADP1/device/path:\_SB_.ADP1
/sys/class/power_supply/BAT0/type:Battery
/sys/class/power_supply/BAT0/present:1
/sys/class/power_supply/BAT0/status:Discharging
/sys/class/power_supply/BAT0/device/path:\_SB_.BAT0

----------------------b----------------------
+++ Power supply diagnostic
/sys/class/power_supply/ADP1/type:Mains
/sys/class/power_supply/ADP1/online:1
/sys/class/power_supply/ADP1/device/path:\_SB_.ADP1
/sys/class/power_supply/BAT0/type:Battery
/sys/class/power_supply/BAT0/present:1
/sys/class/power_supply/BAT0/status:Charging
/sys/class/power_supply/BAT0/device/path:\_SB_.BAT0

nnmlss, 2019-12-09:
Probably I miss something, but the results of the tests are good, aren't they? When I plug and unplug the AC my os produce a sound also and the batery icon in the system tray changes. There is a delay of second or two, but usually it works.

linrunner, 2019-12-09:
Maybe the delay you mention is the culprit. 

Please configure [Trace Mode](https://localhost/linrunner/tlp/docs/tlp-linux-advanced-power-management.html#trace) with

     TLP_DEBUG="ps run"

to keep it brief. And show the output of

    sudo tlp-stat -T

for (a) unplugging and (b) plugging AC.

nnmlss, 2019-12-09:
Hmm, I have set the TLP_DEBUG variable, but I cannot do execute the command and unplug the cable at one and the same time, because the command returns one and the same information and exits quickly after that. Also the output I paste here has line through in my preview window, but I did not add such a styling, don't no why it is applied.

b0@bo-pc:~$ echo $TLP_DEBUG
ps run
b0@bo-pc:~$ sudo tlp-stat -T
--- TLP 1.2.2 --------------------------------------------

-- Logs begin at Mon 2019-12-09 19:35:21 EET, end at Mon 2019-12-09 21:48:28 EET. --
дек 09 19:35:30 bo-pc tlp[2099]: Applying power save settings...done.
дек 09 19:35:30 bo-pc tlp[2099]: Setting battery charge thresholds...done.

===== unplug the AC and execute again:
b0@bo-pc:~$ sudo tlp-stat -T
--- TLP 1.2.2 --------------------------------------------

-- Logs begin at Mon 2019-12-09 19:35:21 EET, end at Mon 2019-12-09 21:48:39 EET. --
дек 09 19:35:30 bo-pc tlp[2099]: Applying power save settings...done.
дек 09 19:35:30 bo-pc tlp[2099]: Setting battery charge thresholds...done.

===== Plugging AC and execute again 
b0@bo-pc:~$ sudo tlp-stat -T
--- TLP 1.2.2 --------------------------------------------

-- Logs begin at Mon 2019-12-09 19:35:21 EET, end at Mon 2019-12-09 21:48:46 EET. --
дек 09 19:35:30 bo-pc tlp[2099]: Applying power save settings...done.
дек 09 19:35:30 bo-pc tlp[2099]: Setting battery charge thresholds...done.
b0@bo-pc:~$

linrunner, 2019-12-09:
You've got me wrong:

1. You must add the line `TLP_DEBUG="ps run"` to **/etc/default/tlp**
2. Trace info is written to the syslogs, you may very well issue `tlp-stat -T` *after* the event

nnmlss, 2019-12-09:
yes, I just figure out this. 

--- TLP 1.2.2 --------------------------------------------

-- Logs begin at Mon 2019-12-09 19:35:21 EET, end at Mon 2019-12-09 22:01:28 EET. --
дек 09 19:35:30 bo-pc tlp[2099]: Applying power save settings...done.
дек 09 19:35:30 bo-pc tlp[2099]: Setting battery charge thresholds...done.
дек 09 22:00:43 bo-pc tlp[13759]: get_sys_power_supply(BAT0).bat_idle: syspwr=0
дек 09 22:00:43 bo-pc tlp[13759]: +++ auto (1.2.2) ++++++++++++++++++++++++++++++++++++++++
дек 09 22:00:43 bo-pc tlp[13759]: power_source=ac
дек 09 22:00:43 bo-pc tlp[13759]: manual_mode=none
дек 09 22:00:43 bo-pc tlp[13759]: power_mode=ac
дек 09 22:00:43 bo-pc tlp[13773]: get_sys_power_supply(BAT0).bat_idle: syspwr=0
дек 09 22:00:43 bo-pc tlp[13773]: +++ auto (1.2.2) ++++++++++++++++++++++++++++++++++++++++
дек 09 22:00:43 bo-pc tlp[13773]: power_source=ac
дек 09 22:00:43 bo-pc tlp[13773]: manual_mode=none
дек 09 22:00:43 bo-pc tlp[13773]: power_mode=ac
дек 09 22:00:45 bo-pc tlp[13819]: get_sys_power_supply(ADP1).ac_online: syspwr=0
дек 09 22:00:45 bo-pc tlp[13819]: +++ auto (1.2.2) ++++++++++++++++++++++++++++++++++++++++
дек 09 22:00:45 bo-pc tlp[13819]: power_source=ac
дек 09 22:00:45 bo-pc tlp[13819]: manual_mode=none
дек 09 22:00:45 bo-pc tlp[13819]: power_mode=ac
дек 09 22:00:45 bo-pc tlp[13827]: get_sys_power_supply(ADP1).ac_online: syspwr=0
дек 09 22:00:45 bo-pc tlp[13827]: +++ auto (1.2.2) ++++++++++++++++++++++++++++++++++++++++
дек 09 22:00:45 bo-pc tlp[13827]: power_source=ac
дек 09 22:00:45 bo-pc tlp[13827]: manual_mode=none
дек 09 22:00:45 bo-pc tlp[13827]: power_mode=ac
дек 09 22:00:47 bo-pc tlp[13861]: get_sys_power_supply(ADP1).ac_online: syspwr=0
дек 09 22:00:47 bo-pc tlp[13861]: +++ auto (1.2.2) ++++++++++++++++++++++++++++++++++++++++
дек 09 22:00:47 bo-pc tlp[13861]: power_source=ac
дек 09 22:00:47 bo-pc tlp[13861]: manual_mode=none
дек 09 22:00:47 bo-pc tlp[13861]: power_mode=ac
дек 09 22:00:50 bo-pc tlp[13889]: get_sys_power_supply(ADP1).ac_online: syspwr=0
дек 09 22:00:50 bo-pc tlp[13889]: +++ start (1.2.2) ++++++++++++++++++++++++++++++++++++++++
дек 09 22:00:50 bo-pc tlp[13889]: power_source=ac
дек 09 22:00:50 bo-pc tlp[13889]: manual_mode=none
дек 09 22:00:50 bo-pc tlp[13889]: power_mode=ac
дек 09 22:01:04 bo-pc tlp[14230]: get_sys_power_supply(BAT0).bat_idle: syspwr=0
дек 09 22:01:04 bo-pc tlp[14230]: +++ auto (1.2.2) ++++++++++++++++++++++++++++++++++++++++
дек 09 22:01:04 bo-pc tlp[14230]: power_source=ac
дек 09 22:01:04 bo-pc tlp[14230]: manual_mode=none
дек 09 22:01:04 bo-pc tlp[14230]: power_mode=ac
дек 09 22:01:04 bo-pc tlp[14244]: get_sys_power_supply(BAT0).bat_idle: syspwr=0
дек 09 22:01:04 bo-pc tlp[14244]: +++ auto (1.2.2) ++++++++++++++++++++++++++++++++++++++++
дек 09 22:01:04 bo-pc tlp[14244]: power_source=ac
дек 09 22:01:04 bo-pc tlp[14244]: manual_mode=none
дек 09 22:01:04 bo-pc tlp[14244]: power_mode=ac
дек 09 22:01:10 bo-pc tlp[14292]: get_sys_power_supply(ADP1).ac_online: syspwr=0
дек 09 22:01:10 bo-pc tlp[14292]: +++ auto (1.2.2) ++++++++++++++++++++++++++++++++++++++++
дек 09 22:01:10 bo-pc tlp[14292]: power_source=ac
дек 09 22:01:10 bo-pc tlp[14292]: manual_mode=none
дек 09 22:01:10 bo-pc tlp[14292]: power_mode=ac
дек 09 22:01:10 bo-pc tlp[14303]: get_sys_power_supply(ADP1).ac_online: syspwr=0
дек 09 22:01:10 bo-pc tlp[14303]: +++ auto (1.2.2) ++++++++++++++++++++++++++++++++++++++++
дек 09 22:01:10 bo-pc tlp[14303]: power_source=ac
дек 09 22:01:10 bo-pc tlp[14303]: manual_mode=none
дек 09 22:01:10 bo-pc tlp[14303]: power_mode=ac
дек 09 22:01:13 bo-pc tlp[14335]: get_sys_power_supply(ADP1).ac_online: syspwr=0
дек 09 22:01:13 bo-pc tlp[14335]: +++ auto (1.2.2) ++++++++++++++++++++++++++++++++++++++++
дек 09 22:01:13 bo-pc tlp[14335]: power_source=ac
дек 09 22:01:13 bo-pc tlp[14335]: manual_mode=none
дек 09 22:01:13 bo-pc tlp[14335]: power_mode=ac

linrunner, 2019-12-09:
Could you mark (a) and (b)? Watch the exact time please.

nnmlss, 2019-12-09:
Sorry but I do not get it.. I added the line into /etc/default/tlp and plugged and unplugged the cable few times then execute the `sud tlp-stat -T` and this is the output. 

What you want me to do is to unplug, execute the command, paste the trace and then plugin, execute the command and paste the trace again, am I correct?

nnmlss, 2019-12-09:
========== a ========
дек 09 22:10:57 bo-pc tlp[15119]: get_sys_power_supply(BAT0).bat_idle: syspwr=0
дек 09 22:10:57 bo-pc tlp[15119]: +++ auto (1.2.2) ++++++++++++++++++++++++++++++++++++++++
дек 09 22:10:57 bo-pc tlp[15119]: power_source=ac
дек 09 22:10:57 bo-pc tlp[15119]: manual_mode=none
дек 09 22:10:57 bo-pc tlp[15119]: power_mode=ac
дек 09 22:10:57 bo-pc tlp[15131]: get_sys_power_supply(BAT0).bat_idle: syspwr=0
дек 09 22:10:57 bo-pc tlp[15131]: +++ auto (1.2.2) ++++++++++++++++++++++++++++++++++++++++
дек 09 22:10:57 bo-pc tlp[15131]: power_source=ac
дек 09 22:10:57 bo-pc tlp[15131]: manual_mode=none
дек 09 22:10:57 bo-pc tlp[15131]: power_mode=ac

==========b=========
дек 09 22:11:47 bo-pc tlp[15273]: get_sys_power_supply(ADP1).ac_online: syspwr=0
дек 09 22:11:47 bo-pc tlp[15273]: +++ auto (1.2.2) ++++++++++++++++++++++++++++++++++++++++
дек 09 22:11:47 bo-pc tlp[15273]: power_source=ac
дек 09 22:11:47 bo-pc tlp[15273]: manual_mode=none
дек 09 22:11:47 bo-pc tlp[15273]: power_mode=ac
дек 09 22:11:47 bo-pc tlp[15283]: get_sys_power_supply(ADP1).ac_online: syspwr=0
дек 09 22:11:47 bo-pc tlp[15283]: +++ auto (1.2.2) ++++++++++++++++++++++++++++++++++++++++
дек 09 22:11:47 bo-pc tlp[15283]: power_source=ac
дек 09 22:11:47 bo-pc tlp[15283]: manual_mode=none
дек 09 22:11:47 bo-pc tlp[15283]: power_mode=ac
b0@bo-pc:~$

linrunner, 2019-12-09:
Sorry. I don't need separate outputs, the trace carries timestamps already. 

What I'd like to know is at which time exactly unplugging (a) and plugging (b) actually happened. Maybe you set your desktop's clock to show seconds and watch it when plugging / unplugging. And another trace of the whole thing of course.

nnmlss, 2019-12-09:
The trace above is extraction of (a) after unplugged, wait about a minute and (b) after plugged the power adapter. I've kept an eye of the clock and copy/paste only the output  of the time of unplugging and plugging, these are the last lines of the log, there is nothing in between.

linrunner, 2019-12-10:
Thanks. Your logs and description say:
* Unplugging happened at 22:10:57
* The corresponding udev event occured without delay
* Battery status at event time is "Unknown":
    POWER_SUPPLY_NAME=BAT0
    POWER_SUPPLY_STATUS=Unknown

TLP's interpretation is:
* [get_sys_power_supply()](https://github.com/linrunner/TLP/blob/1.2.2/tlp-func-base.in#L512) interprets "Unknown" as battery not discharging e.g. AC mode
* In this case battery status has precedence over the AC offline status because we have to work around quirky ACPI BIOS or kernel implementations of the AC/charger device
* TLP concludes laptop is in AC mode

Possible workaround: add `X_SIMULATE_BAT_QUIRK= 2` to config file ([see](https://github.com/linrunner/TLP/blob/1.2.2/tlp-func-base.in#L557)) so AC offline status will take precedence

I am currently missing ideas how to solve this correctly.

Can you please (with `tlp-stat --psup`) check how long it takes after unplugging before the battery status changes from "Unknown" to "Discharging?

nnmlss, 2019-12-10:
There are no any stats using `tlp-stat --psup` about timestamp, but I may say less then a second, because if I execute it quickly after unplug on the first execution of the command is unknown, but on the next cycle is discharging. Now I hear the sound of unplugged charger immediately but sometimes it takes more time and probably the result will be different.  

Adding the line `X_SIMULATE_BAT_QUIRK=2` into config file works - now it changes the mode when I plug in and plug out the charger properly and I can see that the bluetooth icon changes (on battery is disabled, on ac enabled). **Thanks for your time**, if you need some logs or other tests, I will make it.  
Respect!

linrunner, 2019-12-12:
Fine. I think I have an idea how to solve it now. Stay tuned.

linrunner, 2019-12-17:
Done. I've uploaded packages [here](https://download.linrunner.de/packages/).

Please test. 

Remember to remove the line `X_SIMULATE_BAT_QUIRK=2` from your config file.
Important: with 1.3 the user config moved to **/etc/tlp.conf**.

nnmlss, 2019-12-17:
Before I start testing I have edited the config file with a comment of the line, installed (update) both of the deb packages and reboot. 

The system started with **100% battery power on AC**.   `tlp-stat -s` 
Mode - AC 
Power source - AC 

Unplugged the power cord and no sound of the OS about about changing the power source, also the battery icon remain the same - 100% charged. `tlp-stat -s` otutput was
Mode - AC
Power source - battery

A few more tests - Power source changes properly, but mode is always AC

The battery discharded on 97% and then I hear the sound of the OS, also the icon changed to Discahrging, but the output of `tlp-stat -s` remain the same:
Mode - AC
Power source - battery

Plug in the charger, the OS (sound and icon) detected the power change and the output of `tlp-stat -s`
Mode - AC
Power source - AC

Unplugged the charger, OS detected the change, `tlp-stat -s` 
Mode - AC
Power source - battery

~~Tried this a few times with no change - power source changes but mode always is AC. 

Reboot the system on **battery**, `tlp-stat -s`:
Mode - battery
Power source - battery. 

Plug in the power cord, OS detected the change `tlp-stat -s`
Mode - AC
power source - AC

Unplugged the power cord, OS detected the change `tlp-stat -s`
Mode - battery
Power source - battery

Tried a few times, everything works as it should. 

Wait unitl the **battery charged to 100%**, unplugged the power source,  `tlp-stat -s`
Mode AC
Power source - battery.

Tried all above 2 times, everything works well **until the battery is 100% charged**. After this, the power source is properly detected, but the mode remain AC even after some discharge of the battery no matter that the OS detect the change or not. In order to fix it, I have to reboot the system on a battery,  probably if I run `sudo tlp start` on a battery will fix it without reboot until the battery change the status to 100% charged again. 
---edit----
It took me about 10 minutes until I edit the text above to make it more readable than the draft version, the battery now is 99%, I did not restarted on battery, neighter run `sudo tlp start` but everything works as it should - power source and battery changes properly when I plug and unplug the power adapter. So it is not necessary to restart tlp on a battery in order to work properly. I will wait until the 100% and will share the test results in another comment.

nnmlss, 2019-12-17:
So, again after 100% charge the Mode keep staying into AC but the power source is changing properly. Tired this like 10 times with no change, no matter that the os detects or not the power change.

Rebooted into windows 10, if I unplug the power source the icon and help text into system tray changes immediately no matter is it 100% charged or less.  

Tried on original Ubuntu 18.04 with latest kernel updates - results are the same as testing into elementary OS.

linrunner, 2019-12-29:
To be honest, i can't keep track of all the details of your description. 

To me the bottom line is: at the point in time of the udev event, the kernel's battery status is incorrect. So i have created – and uploaded – a new revision, which in this case weights the AC offline (unplugged) status higher.

Please re-test without the workaround configuration.

nnmlss, 2019-12-29:
Donwloaded and updated (I have reverted to the 1.2 version and uncomment the fix) from the same url - https://download.linrunner.de/packages/ ?

nnmlss, 2019-12-29:
I don't know if it is the new package but it behaves the same like the previous one - when battery is 100% on AC and if I unplug the charger, the mode do not change - it is AC, but the Power source changes to battery. It remains AC even after the battery discharges to 98% and the OS detects that the power source is changed.

linrunner, 2019-12-30:
The packages are new (despite the identical version number). Can you please provide traces?

linrunner, 2019-12-30:
Btw: TLP necessarily requires the udev event from the OS to detect the unplugging (at 98% in your case).

linrunner, 2019-12-30:
It would be best to trace the entire period:

- 100%, plugged
- unplugging
- 98%, OS detects change

nnmlss, 2019-12-30:
Installed both of the updates, comment the workaround line, poweroff. Turn on the laptop with 100% percent battery and a chrger connected

tlp-stat -s
--- TLP 1.3.0-alpha.3 --------------------------------------------

+++ System Info
System         = HP Type1ProductConfigId HP Pavilion Power Laptop 15-cb0xx
BIOS           = F.18
Release        = elementary OS 5.1 Hera
Kernel         = 5.0.0-37-generic #40~18.04.1-Ubuntu SMP Thu Nov 14 12:06:39 UTC 2019 x86_64
/proc/cmdline  = BOOT_IMAGE=/boot/vmlinuz-5.0.0-37-generic root=UUID=3b678b4c-1e36-4b1f-abce-728fc72282a9 ro quiet splash pcie_aspm=off
Init system    = systemd v237
Boot mode      = UEFI

+++ TLP Status
State          = enabled
RDW state      = enabled
Last run       = 15:25:55,     23 sec(s) ago
Mode           = AC
Power source   = AC`

**unplugged the adapter, not detected from the os:**

tlp-stat -s
--- TLP 1.3.0-alpha.3 --------------------------------------------

+++ System Info
System         = HP Type1ProductConfigId HP Pavilion Power Laptop 15-cb0xx
BIOS           = F.18
Release        = elementary OS 5.1 Hera
Kernel         = 5.0.0-37-generic #40~18.04.1-Ubuntu SMP Thu Nov 14 12:06:39 UTC 2019 x86_64
/proc/cmdline  = BOOT_IMAGE=/boot/vmlinuz-5.0.0-37-generic root=UUID=3b678b4c-1e36-4b1f-abce-728fc72282a9 ro quiet splash pcie_aspm=off
Init system    = systemd v237
Boot mode      = UEFI

+++ TLP Status
State          = enabled
RDW state      = enabled
Last run       = 15:26:32,      4 sec(s) ago
Mode           = AC
Power source   = battery

**wait until the OS detects the battery:**

tlp-stat -s
--- TLP 1.3.0-alpha.3 --------------------------------------------

+++ System Info
System         = HP Type1ProductConfigId HP Pavilion Power Laptop 15-cb0xx
BIOS           = F.18
Release        = elementary OS 5.1 Hera
Kernel         = 5.0.0-37-generic #40~18.04.1-Ubuntu SMP Thu Nov 14 12:06:39 UTC 2019 x86_64
/proc/cmdline  = BOOT_IMAGE=/boot/vmlinuz-5.0.0-37-generic root=UUID=3b678b4c-1e36-4b1f-abce-728fc72282a9 ro quiet splash pcie_aspm=off
Init system    = systemd v237
Boot mode      = UEFI

+++ TLP Status
State          = enabled
RDW state      = enabled
Last run       = 15:26:32,    130 sec(s) ago
Mode           = AC
Power source   = battery

b0@bo-pc:~/Desktop$ acpi
Battery 0: Discharging, 98%, 03:43:06 remaining

So the results are the same as before. Something to point - I do not know if it is important or not, but because of a lot of acpie error logs all the time running `dmesg` or even into tty I put an option into grub `pcie_aspm=off`. Should I remove it? 

Also, when I have updated to  1.3alpha, most of times running `acpi` I see this result: 
`Battery 0: Charging, 97%, charging at zero rate - will never fully charge.`
which is not correct, because after 5-10 minutes from 97% suddenly it is 100% charged. Yestarday after the tests I reverted back to 1.2 version with the workaround and there was not such fake reports from the acpi command.

linrunner, 2019-12-30:
Thanks, but what I need is a trace of the whole period --> https://github.com/linrunner/TLP/issues/446#issuecomment-563398930

nnmlss, 2019-12-30:
sudoedit /etc/tlp.conf  **uncomment debug line**            
sudo tlp start 
TLP started in AC mode.
b0@bo-pc:~/Desktop$ sudo tlp-stat -T
--- TLP 1.3.0-alpha.3 --------------------------------------------

-- Logs begin at Mon 2019-12-30 15:25:54 EET, end at Mon 2019-12-30 19:57:12 EET. --
дек 30 15:25:55 bo-pc tlp[1329]: Applying power save settings...done.
дек 30 15:25:55 bo-pc tlp[1329]: Setting battery charge thresholds...done.
дек 30 19:56:17 bo-pc tlp[1561]: get_sys_power_supply(ADP1).ac_online: syspwr=0
дек 30 19:56:17 bo-pc tlp[1561]: +++ start (1.3.0-alpha.3) ++++++++++++++++++++++++++++++++++++++++
дек 30 19:56:17 bo-pc tlp[1561]: power_source=ac
дек 30 19:56:17 bo-pc tlp[1561]: manual_mode=none
дек 30 19:56:17 bo-pc tlp[1561]: power_mode=ac
дек 30 19:56:17 bo-pc tlp[1561]: compare_and_save_power_state(0).equal
дек 30 19:56:35 bo-pc tlp[1927]: get_sys_power_supply(ADP1).ac_online: syspwr=0
дек 30 19:56:35 bo-pc tlp[1927]: +++ auto (1.3.0-alpha.3) ++++++++++++++++++++++++++++++++++++++++
дек 30 19:56:35 bo-pc tlp[1927]: power_source=ac
дек 30 19:56:35 bo-pc tlp[1927]: manual_mode=none
дек 30 19:56:35 bo-pc tlp[1927]: power_mode=ac
дек 30 19:56:35 bo-pc tlp[1927]: compare_and_save_power_state(0).equal
дек 30 19:57:07 bo-pc tlp[1967]: get_sys_power_supply(ADP1).ac_online: syspwr=0
дек 30 19:57:07 bo-pc tlp[1967]: +++ auto (1.3.0-alpha.3) ++++++++++++++++++++++++++++++++++++++++
дек 30 19:57:07 bo-pc tlp[1967]: power_source=ac
дек 30 19:57:07 bo-pc tlp[1967]: manual_mode=none
дек 30 19:57:07 bo-pc tlp[1967]: power_mode=ac
дек 30 19:57:07 bo-pc tlp[1967]: compare_and_save_power_state(0).equal

**unplug the charger**
b0@bo-pc:~/Desktop$ sudo tlp-stat -T
--- TLP 1.3.0-alpha.3 --------------------------------------------

-- Logs begin at Mon 2019-12-30 15:25:54 EET, end at Mon 2019-12-30 19:57:28 EET. --
дек 30 15:25:55 bo-pc tlp[1329]: Applying power save settings...done.
дек 30 15:25:55 bo-pc tlp[1329]: Setting battery charge thresholds...done.
дек 30 19:56:17 bo-pc tlp[1561]: get_sys_power_supply(ADP1).ac_online: syspwr=0
дек 30 19:56:17 bo-pc tlp[1561]: +++ start (1.3.0-alpha.3) ++++++++++++++++++++++++++++++++++++++++
дек 30 19:56:17 bo-pc tlp[1561]: power_source=ac
дек 30 19:56:17 bo-pc tlp[1561]: manual_mode=none
дек 30 19:56:17 bo-pc tlp[1561]: power_mode=ac
дек 30 19:56:17 bo-pc tlp[1561]: compare_and_save_power_state(0).equal
дек 30 19:56:35 bo-pc tlp[1927]: get_sys_power_supply(ADP1).ac_online: syspwr=0
дек 30 19:56:35 bo-pc tlp[1927]: +++ auto (1.3.0-alpha.3) ++++++++++++++++++++++++++++++++++++++++
дек 30 19:56:35 bo-pc tlp[1927]: power_source=ac
дек 30 19:56:35 bo-pc tlp[1927]: manual_mode=none
дек 30 19:56:35 bo-pc tlp[1927]: power_mode=ac
дек 30 19:56:35 bo-pc tlp[1927]: compare_and_save_power_state(0).equal
дек 30 19:57:07 bo-pc tlp[1967]: get_sys_power_supply(ADP1).ac_online: syspwr=0
дек 30 19:57:07 bo-pc tlp[1967]: +++ auto (1.3.0-alpha.3) ++++++++++++++++++++++++++++++++++++++++
дек 30 19:57:07 bo-pc tlp[1967]: power_source=ac
дек 30 19:57:07 bo-pc tlp[1967]: manual_mode=none
дек 30 19:57:07 bo-pc tlp[1967]: power_mode=ac
дек 30 19:57:07 bo-pc tlp[1967]: compare_and_save_power_state(0).equal
дек 30 19:57:22 bo-pc tlp[2048]: get_sys_power_supply(BAT0).bat_not_discharging_re-checked: syspwr=0
дек 30 19:57:22 bo-pc tlp[2046]: get_sys_power_supply(BAT0).bat_not_discharging_re-checked: syspwr=0
дек 30 19:57:22 bo-pc tlp[2046]: +++ auto (1.3.0-alpha.3) ++++++++++++++++++++++++++++++++++++++++
дек 30 19:57:22 bo-pc tlp[2048]: +++ auto (1.3.0-alpha.3) ++++++++++++++++++++++++++++++++++++++++
дек 30 19:57:22 bo-pc tlp[2046]: power_source=ac
дек 30 19:57:22 bo-pc tlp[2048]: power_source=ac
дек 30 19:57:22 bo-pc tlp[2046]: manual_mode=none
дек 30 19:57:22 bo-pc tlp[2048]: manual_mode=none
дек 30 19:57:22 bo-pc tlp[2046]: power_mode=ac
дек 30 19:57:22 bo-pc tlp[2048]: power_mode=ac
дек 30 19:57:22 bo-pc tlp[2046]: compare_and_save_power_state(0).equal

**os detects there's no charger**
b0@bo-pc:~/Desktop$ sudo tlp-stat -T
--- TLP 1.3.0-alpha.3 --------------------------------------------

-- Logs begin at Mon 2019-12-30 15:25:54 EET, end at Mon 2019-12-30 19:59:28 EET. --
[... 6 lines cut ...]
дек 30 19:56:17 bo-pc tlp[1561]: power_mode=ac
дек 30 19:56:17 bo-pc tlp[1561]: compare_and_save_power_state(0).equal
дек 30 19:56:35 bo-pc tlp[1927]: get_sys_power_supply(ADP1).ac_online: syspwr=0
дек 30 19:56:35 bo-pc tlp[1927]: +++ auto (1.3.0-alpha.3) ++++++++++++++++++++++++++++++++++++++++
дек 30 19:56:35 bo-pc tlp[1927]: power_source=ac
дек 30 19:56:35 bo-pc tlp[1927]: manual_mode=none
дек 30 19:56:35 bo-pc tlp[1927]: power_mode=ac
дек 30 19:56:35 bo-pc tlp[1927]: compare_and_save_power_state(0).equal
дек 30 19:57:07 bo-pc tlp[1967]: get_sys_power_supply(ADP1).ac_online: syspwr=0
дек 30 19:57:07 bo-pc tlp[1967]: +++ auto (1.3.0-alpha.3) ++++++++++++++++++++++++++++++++++++++++
дек 30 19:57:07 bo-pc tlp[1967]: power_source=ac
дек 30 19:57:07 bo-pc tlp[1967]: manual_mode=none
дек 30 19:57:07 bo-pc tlp[1967]: power_mode=ac
дек 30 19:57:07 bo-pc tlp[1967]: compare_and_save_power_state(0).equal
дек 30 19:57:22 bo-pc tlp[2048]: get_sys_power_supply(BAT0).bat_not_discharging_re-checked: syspwr=0
дек 30 19:57:22 bo-pc tlp[2046]: get_sys_power_supply(BAT0).bat_not_discharging_re-checked: syspwr=0
дек 30 19:57:22 bo-pc tlp[2046]: +++ auto (1.3.0-alpha.3) ++++++++++++++++++++++++++++++++++++++++
дек 30 19:57:22 bo-pc tlp[2048]: +++ auto (1.3.0-alpha.3) ++++++++++++++++++++++++++++++++++++++++
дек 30 19:57:22 bo-pc tlp[2046]: power_source=ac
дек 30 19:57:22 bo-pc tlp[2048]: power_source=ac
дек 30 19:57:22 bo-pc tlp[2046]: manual_mode=none
дек 30 19:57:22 bo-pc tlp[2048]: manual_mode=none
дек 30 19:57:22 bo-pc tlp[2046]: power_mode=ac
дек 30 19:57:22 bo-pc tlp[2048]: power_mode=ac
дек 30 19:57:22 bo-pc tlp[2046]: compare_and_save_power_state(0).equal

linrunner, 2019-12-30:
Well.

1. When unplugging a udev event is issued but battery status is neither "Discharging" nor "Unknown" at that time
2. When the OS actually detects there's no charger, no udev event is issued

So we have to catch at 1. I made another change, same download location, same version number. 
Please install and give me the trace again.

nnmlss, 2019-12-30:
I have unplugged in the middle of 23:10 system time on the log below. There was no delay to become 98%, the os detected the change almost immediately, so it was the first time it happens with 100% battery. Before I plug it back after a minute, it was 100% discharging, but once I plug it in, it changed to 97% charging. Tried this like 3 times, with the same result, only % differs, sometimes it is 98, sometimes it is 97.  But it looks like mode changes properly

дек 30 23:09:01 bo-pc tlp[17348]: get_sys_power_supply(ADP1).ac_online: syspwr=0
дек 30 23:09:01 bo-pc tlp[17348]: +++ auto (1.3.0-alpha.3) ++++++++++++++++++++++++++++++++++++++++
дек 30 23:09:01 bo-pc tlp[17348]: power_source=ac
дек 30 23:09:01 bo-pc tlp[17348]: manual_mode=none
дек 30 23:09:01 bo-pc tlp[17348]: power_mode=ac
дек 30 23:09:01 bo-pc tlp[17348]: compare_and_save_power_state(0).equal
дек 30 23:09:33 bo-pc tlp[17483]: get_sys_power_supply(ADP1).ac_online: syspwr=0
дек 30 23:09:33 bo-pc tlp[17483]: +++ auto (1.3.0-alpha.3) ++++++++++++++++++++++++++++++++++++++++
дек 30 23:09:33 bo-pc tlp[17483]: power_source=ac
дек 30 23:09:33 bo-pc tlp[17483]: manual_mode=none
дек 30 23:09:33 bo-pc tlp[17483]: power_mode=ac
дек 30 23:09:33 bo-pc tlp[17483]: compare_and_save_power_state(0).equal
дек 30 23:10:05 bo-pc tlp[17517]: get_sys_power_supply(ADP1).ac_online: syspwr=0
дек 30 23:10:05 bo-pc tlp[17517]: +++ auto (1.3.0-alpha.3) ++++++++++++++++++++++++++++++++++++++++
дек 30 23:10:05 bo-pc tlp[17517]: power_source=ac
дек 30 23:10:05 bo-pc tlp[17517]: manual_mode=none
дек 30 23:10:05 bo-pc tlp[17517]: power_mode=ac
дек 30 23:10:05 bo-pc tlp[17517]: compare_and_save_power_state(0).equal
дек 30 23:10:37 bo-pc tlp[17553]: get_sys_power_supply(ADP1).ac_online: syspwr=0
дек 30 23:10:37 bo-pc tlp[17553]: +++ auto (1.3.0-alpha.3) ++++++++++++++++++++++++++++++++++++++++
дек 30 23:10:37 bo-pc tlp[17553]: power_source=ac
дек 30 23:10:37 bo-pc tlp[17553]: manual_mode=none
дек 30 23:10:37 bo-pc tlp[17553]: power_mode=ac
дек 30 23:10:37 bo-pc tlp[17553]: compare_and_save_power_state(0).equal
дек 30 23:10:38 bo-pc tlp[17584]: get_sys_power_supply(ADP1).ac_offline_remembered
дек 30 23:10:38 bo-pc tlp[17588]: get_sys_power_supply(ADP1).ac_offline_remembered
дек 30 23:10:38 bo-pc tlp[17584]: get_sys_power_supply(BAT0).bat_not_discharging_check1: bs=Full; syspwr=
дек 30 23:10:38 bo-pc tlp[17588]: get_sys_power_supply(BAT0).bat_not_discharging_check1: bs=Full; syspwr=
дек 30 23:10:38 bo-pc tlp[17584]: get_sys_power_supply(BAT0).bat_discharging_check2: bs=Discharging; syspwr=1
дек 30 23:10:38 bo-pc tlp[17584]: +++ auto (1.3.0-alpha.3) ++++++++++++++++++++++++++++++++++++++++
дек 30 23:10:38 bo-pc tlp[17584]: power_source=bat
дек 30 23:10:38 bo-pc tlp[17588]: get_sys_power_supply(BAT0).bat_discharging_check2: bs=Discharging; syspwr=1
дек 30 23:10:38 bo-pc tlp[17584]: manual_mode=none
дек 30 23:10:38 bo-pc tlp[17588]: +++ auto (1.3.0-alpha.3) ++++++++++++++++++++++++++++++++++++++++
дек 30 23:10:38 bo-pc tlp[17584]: power_mode=bat
дек 30 23:10:38 bo-pc tlp[17588]: power_source=bat
дек 30 23:10:38 bo-pc tlp[17588]: manual_mode=none
дек 30 23:10:38 bo-pc tlp[17588]: power_mode=bat
дек 30 23:10:38 bo-pc tlp[17584]: compare_and_save_power_state(1).different: old=0

linrunner, 2019-12-30:
OK. My additional logic was not needed at all ...

Your battery status needs a little more than 0.5 secs to change from "Full" to "Discharging". So i've changed the internal delay for a 2nd check from 0.5 to 0.8 secs to achieve a stable detection (hopefully).

Please test again with the newly uploaded revision and show the trace.

Thanks for your patience.

nnmlss, 2019-12-30:
No problem, I understand that in order to have a good software it is necessary to be tested and it is ok for me to participate. 

So, the last version do not work. The OS do not detect the unplugged charger until the battery is 98% again and there is no change into mode - again is AC, no matter that the os detected the change after 98% discharge. The trace command is just after the OS detected the event.

sudo tlp-stat -T
--- TLP 1.3.0-alpha.3 --------------------------------------------

-- Logs begin at Tue 2019-12-31 01:19:40 EET, end at Tue 2019-12-31 01:24:09 EET. --
дек 31 01:19:42 bo-pc tlp[1334]: Applying power save settings...done.
дек 31 01:19:42 bo-pc tlp[1334]: Setting battery charge thresholds...done.
дек 31 01:21:12 bo-pc tlp[3292]: get_sys_power_supply(ADP1).ac_online: syspwr=0
дек 31 01:21:12 bo-pc tlp[3292]: +++ auto (1.3.0-alpha.3) ++++++++++++++++++++++++++++++++++++++++
дек 31 01:21:12 bo-pc tlp[3292]: power_source=ac
дек 31 01:21:12 bo-pc tlp[3292]: manual_mode=none
дек 31 01:21:12 bo-pc tlp[3292]: power_mode=ac
дек 31 01:21:12 bo-pc tlp[3292]: compare_and_save_power_state(0).equal
дек 31 01:21:15 bo-pc tlp[3346]: get_sys_power_supply(ADP1).ac_offline_remembered
дек 31 01:21:15 bo-pc tlp[3348]: get_sys_power_supply(ADP1).ac_offline_remembered
дек 31 01:21:15 bo-pc tlp[3346]: get_sys_power_supply(BAT0).bat_not_discharging_check1: bs=Full; syspwr=
дек 31 01:21:15 bo-pc tlp[3348]: get_sys_power_supply(BAT0).bat_not_discharging_check1: bs=Full; syspwr=
дек 31 01:21:16 bo-pc tlp[3346]: get_sys_power_supply(BAT0).bat_not_discharging_check2: bs=Full; syspwr=0
дек 31 01:21:16 bo-pc tlp[3346]: +++ auto (1.3.0-alpha.3) ++++++++++++++++++++++++++++++++++++++++
дек 31 01:21:16 bo-pc tlp[3346]: power_source=ac
дек 31 01:21:16 bo-pc tlp[3348]: get_sys_power_supply(BAT0).bat_not_discharging_check2: bs=Full; syspwr=0
дек 31 01:21:16 bo-pc tlp[3346]: manual_mode=none
дек 31 01:21:16 bo-pc tlp[3348]: +++ auto (1.3.0-alpha.3) ++++++++++++++++++++++++++++++++++++++++
дек 31 01:21:16 bo-pc tlp[3346]: power_mode=ac
дек 31 01:21:16 bo-pc tlp[3348]: power_source=ac
дек 31 01:21:16 bo-pc tlp[3348]: manual_mode=none
дек 31 01:21:16 bo-pc tlp[3348]: power_mode=ac
дек 31 01:21:16 bo-pc tlp[3346]: compare_and_save_power_state(0).equal
дек 31 01:21:16 bo-pc tlp[3348]: compare_and_save_power_state(0).equal
дек 31 01:21:44 bo-pc tlp[7937]: get_sys_power_supply(ADP1).ac_online: syspwr=0
дек 31 01:21:44 bo-pc tlp[7937]: +++ auto (1.3.0-alpha.3) ++++++++++++++++++++++++++++++++++++++++
дек 31 01:21:44 bo-pc tlp[7940]: get_sys_power_supply(ADP1).ac_online: syspwr=0
дек 31 01:21:44 bo-pc tlp[7937]: power_source=ac
дек 31 01:21:44 bo-pc tlp[7937]: manual_mode=none
дек 31 01:21:44 bo-pc tlp[7940]: +++ auto (1.3.0-alpha.3) ++++++++++++++++++++++++++++++++++++++++
дек 31 01:21:44 bo-pc tlp[7937]: power_mode=ac
дек 31 01:21:44 bo-pc tlp[7940]: power_source=ac
дек 31 01:21:44 bo-pc tlp[7940]: manual_mode=none
дек 31 01:21:44 bo-pc tlp[7940]: power_mode=ac
дек 31 01:21:44 bo-pc tlp[7937]: compare_and_save_power_state(0).equal
дек 31 01:21:47 bo-pc tlp[7990]: get_sys_power_supply(ADP1).ac_online: syspwr=0
дек 31 01:21:47 bo-pc tlp[7990]: +++ auto (1.3.0-alpha.3) ++++++++++++++++++++++++++++++++++++++++
дек 31 01:21:47 bo-pc tlp[7990]: power_source=ac
дек 31 01:21:47 bo-pc tlp[7990]: manual_mode=none
дек 31 01:21:47 bo-pc tlp[7990]: power_mode=ac
дек 31 01:21:47 bo-pc tlp[7990]: compare_and_save_power_state(0).equal
дек 31 01:21:51 bo-pc tlp[8028]: get_sys_power_supply(ADP1).ac_online: syspwr=0
дек 31 01:21:51 bo-pc tlp[8028]: +++ start (1.3.0-alpha.3) ++++++++++++++++++++++++++++++++++++++++
дек 31 01:21:51 bo-pc tlp[8028]: power_source=ac
дек 31 01:21:51 bo-pc tlp[8028]: manual_mode=none
дек 31 01:21:51 bo-pc tlp[8028]: power_mode=ac
дек 31 01:21:51 bo-pc tlp[8028]: compare_and_save_power_state(0).equal
дек 31 01:22:00 bo-pc tlp[8468]: get_sys_power_supply(ADP1).ac_offline_remembered
дек 31 01:22:00 bo-pc tlp[8466]: get_sys_power_supply(ADP1).ac_offline_remembered
дек 31 01:22:00 bo-pc tlp[8468]: get_sys_power_supply(BAT0).bat_not_discharging_check1: bs=Full; syspwr=
дек 31 01:22:00 bo-pc tlp[8466]: get_sys_power_supply(BAT0).bat_not_discharging_check1: bs=Full; syspwr=
дек 31 01:22:01 bo-pc tlp[8468]: get_sys_power_supply(BAT0).bat_not_discharging_check2: bs=Full; syspwr=0
дек 31 01:22:01 bo-pc tlp[8466]: get_sys_power_supply(BAT0).bat_not_discharging_check2: bs=Full; syspwr=0
дек 31 01:22:01 bo-pc tlp[8468]: +++ auto (1.3.0-alpha.3) ++++++++++++++++++++++++++++++++++++++++
дек 31 01:22:01 bo-pc tlp[8466]: +++ auto (1.3.0-alpha.3) ++++++++++++++++++++++++++++++++++++++++
дек 31 01:22:01 bo-pc tlp[8468]: power_source=ac
дек 31 01:22:01 bo-pc tlp[8466]: power_source=ac
дек 31 01:22:01 bo-pc tlp[8468]: manual_mode=none
дек 31 01:22:01 bo-pc tlp[8466]: manual_mode=none
дек 31 01:22:01 bo-pc tlp[8468]: power_mode=ac
дек 31 01:22:01 bo-pc tlp[8466]: power_mode=ac
дек 31 01:22:01 bo-pc tlp[8468]: compare_and_save_power_state(0).equal

linrunner, 2019-12-31:
Your hardware / battery behaves differently every time you test.

    23:10:38 bo-pc tlp[17584]: get_sys_power_supply(ADP1).ac_offline_remembered
    23:10:38 bo-pc tlp[17584]: get_sys_power_supply(BAT0).bat_not_discharging_check1: bs=Full; syspwr=
    23:10:38 bo-pc tlp[17584]: get_sys_power_supply(BAT0).bat_discharging_check2: bs=Discharging; syspwr=1

After ~ 0.5 sec change from "Full" to "Discharging".

    01:22:00 bo-pc tlp[8466]: get_sys_power_supply(ADP1).ac_offline_remembered
    01:22:00 bo-pc tlp[8466]: get_sys_power_supply(BAT0).bat_not_discharging_check1: bs=Full; syspwr=
    01:22:01 bo-pc tlp[8466]: get_sys_power_supply(BAT0).bat_not_discharging_check2: bs=Full; syspwr=0

After 0.8 sec no change.

 There's no proper way to handle this automatically. I'll remove all the code added and introduce an official config variable. Stay tuned.

Btw: is your BIOS the newest version?

nnmlss, 2019-12-31:
Yes, the BIOS version I have is the latest for my model according to the HP website and the official HP windows tools. But my model is not into the Ubuntu Desktop HP certified hardware list, I cannot find an appropriate driver in order to see the fan speeds also. It looks like it is not the best hardware for a Linux OS, but actually I keep my windows partition only for BIOS updates if there are any.. Thanks for your effort to support it with your software. Keep up with the good work!

linrunner, 2020-01-02:
Done. You'll need to configure TLP_PS_IGNORE=BAT now to enable the workaround. 

Please re-test.

I hope this is the last round ...

nnmlss, 2020-01-03:
I want to keep my settings so all I did is to add the new line? Testing with `tlp-stat -s` works correctly, no matter that the os do not detect the source change.

6:41 unplugged
sudo tlp-stat -T
--- TLP 1.3.0-alpha.3 --------------------------------------------

-- Logs begin at Fri 2020-01-03 06:30:35 EET, end at Fri 2020-01-03 06:43:14 EET. --
яну 03 06:30:42 bo-pc tlp[2164]: Applying power save settings...done.
яну 03 06:30:43 bo-pc tlp[2164]: Setting battery charge thresholds...done.
яну 03 06:40:41 bo-pc tlp[18292]: get_sys_power_supply(ADP1).ac_online: syspwr=0
яну 03 06:40:41 bo-pc tlp[18292]: +++ start (1.3.0-alpha.3) ++++++++++++++++++++++++++++++++++++++++
яну 03 06:40:41 bo-pc tlp[18292]: power_source=ac
яну 03 06:40:41 bo-pc tlp[18292]: manual_mode=none
яну 03 06:40:41 bo-pc tlp[18292]: power_mode=ac
яну 03 06:40:41 bo-pc tlp[18292]: compare_and_save_power_state(0).equal
яну 03 06:41:03 bo-pc tlp[18625]: get_sys_power_supply(ADP1).ac_online: syspwr=0
яну 03 06:41:03 bo-pc tlp[18625]: +++ auto (1.3.0-alpha.3) ++++++++++++++++++++++++++++++++++++++++
яну 03 06:41:03 bo-pc tlp[18625]: power_source=ac
яну 03 06:41:03 bo-pc tlp[18625]: manual_mode=none
яну 03 06:41:03 bo-pc tlp[18625]: power_mode=ac
яну 03 06:41:03 bo-pc tlp[18625]: compare_and_save_power_state(0).equal
яну 03 06:41:03 bo-pc tlp[18655]: get_sys_power_supply(ADP1).ac_offline_remembered
яну 03 06:41:03 bo-pc tlp[18660]: get_sys_power_supply(ADP1).ac_offline_remembered
яну 03 06:41:03 bo-pc tlp[18655]: get_sys_power_supply(BAT0).bat_ignored: syspwr=
яну 03 06:41:03 bo-pc tlp[18660]: get_sys_power_supply(BAT0).bat_ignored: syspwr=
яну 03 06:41:03 bo-pc tlp[18655]: get_sys_power_supply(1).ac_offline: syspwr=1
яну 03 06:41:03 bo-pc tlp[18660]: get_sys_power_supply(1).ac_offline: syspwr=1
яну 03 06:41:03 bo-pc tlp[18655]: +++ auto (1.3.0-alpha.3) ++++++++++++++++++++++++++++++++++++++++
яну 03 06:41:03 bo-pc tlp[18660]: +++ auto (1.3.0-alpha.3) ++++++++++++++++++++++++++++++++++++++++
яну 03 06:41:03 bo-pc tlp[18655]: power_source=bat
яну 03 06:41:03 bo-pc tlp[18660]: power_source=bat
яну 03 06:41:04 bo-pc tlp[18655]: manual_mode=none
яну 03 06:41:04 bo-pc tlp[18660]: manual_mode=none
яну 03 06:41:04 bo-pc tlp[18655]: power_mode=bat
яну 03 06:41:04 bo-pc tlp[18660]: power_mode=bat
яну 03 06:41:04 bo-pc tlp[18655]: compare_and_save_power_state(1).different: old=0

@6.43 the os detected the power change, but there is nothing in the log about it, probably it should be like this.

linrunner, 2020-01-04:
Fine. 

What you call "OS" is the upower daemon, which apparently has its own heuristic to distinguish AC and battery mode.

nnmlss, 2020-01-04:
I see. But it looks like it depends on your code, because in one of your versions it was acting immediately after the power change in 100% charged, just like windows do. I have spent a lot of time testing in windows also, because I was wondering is it possible my computer to have a hardware power problem independent of the OS, but windows always detects the power change.

linrunner, 2020-01-04:
I can assure you there's no impact on upowerd. It's just the very erratic behaviour of your battery's charge state. I consider this a hardware "quirk". The manufacturer's driver for W****** may contain a workaround.

nnmlss, 2020-01-05:
I see, probalby it was a coincidence. But I test it with this version of yours like more tnen 20 times, with a few reboots of the machine and I was ready to blame your app for the delay of the upower daemon )

linrunner, 2020-01-17:
Please re-test with 1.3.0-beta.3 and report.
Instructions: https://github.com/linrunner/TLP/issues/449

closed 2020-01-17 by linrunner

nnmlss, 2020-01-21:
AC:
-- TLP 1.3.0-beta.3 --------------------------------------------

+++ Configured Settings:
/etc/tlp.conf L0015: TLP_ENABLE="1"
/etc/tlp.conf L0025: TLP_PERSISTENT_DEFAULT="0"
/etc/tlp.conf L0030: DISK_IDLE_SECS_ON_AC="0"
/etc/tlp.conf L0031: DISK_IDLE_SECS_ON_BAT="2"
/etc/tlp.conf L0035: MAX_LOST_WORK_SECS_ON_AC="15"
/etc/tlp.conf L0036: MAX_LOST_WORK_SECS_ON_BAT="60"
defaults.conf L0010: CPU_ENERGY_PERF_POLICY_ON_AC="balance_performance"
defaults.conf L0011: CPU_ENERGY_PERF_POLICY_ON_BAT="power"
/etc/tlp.conf L0097: SCHED_POWERSAVE_ON_AC="0"
/etc/tlp.conf L0098: SCHED_POWERSAVE_ON_BAT="1"
/etc/tlp.conf L0103: NMI_WATCHDOG="0"
/etc/tlp.conf L0125: DISK_DEVICES="nvme-KXG50ZNV256G_TOSHIBA_97BF2015F6FS ata-CT500BX100SSD1_1516F005DD27"
/etc/tlp.conf L0131: DISK_APM_LEVEL_ON_AC="254 254"
/etc/tlp.conf L0132: DISK_APM_LEVEL_ON_BAT="128 128"
defaults.conf L0018: DISK_IOSCHED="keep keep"
/etc/tlp.conf L0166: SATA_LINKPWR_ON_AC="med_power_with_dipm max_performance"
/etc/tlp.conf L0167: SATA_LINKPWR_ON_BAT="min-power med_power_with_dipm"
/etc/tlp.conf L0184: AHCI_RUNTIME_PM_TIMEOUT="15"
/etc/tlp.conf L0190: PCIE_ASPM_ON_AC="performance"
/etc/tlp.conf L0191: PCIE_ASPM_ON_BAT="powersave"
defaults.conf L0024: RADEON_POWER_PROFILE_ON_AC="default"
defaults.conf L0025: RADEON_POWER_PROFILE_ON_BAT="default"
defaults.conf L0026: RADEON_DPM_PERF_LEVEL_ON_AC="auto"
defaults.conf L0027: RADEON_DPM_PERF_LEVEL_ON_BAT="auto"
/etc/tlp.conf L0223: WIFI_PWR_ON_AC="off"
/etc/tlp.conf L0224: WIFI_PWR_ON_BAT="on"
/etc/tlp.conf L0228: WOL_DISABLE="Y"
/etc/tlp.conf L0233: SOUND_POWER_SAVE_ON_AC="0"
/etc/tlp.conf L0234: SOUND_POWER_SAVE_ON_BAT="2"
/etc/tlp.conf L0239: SOUND_POWER_SAVE_CONTROLLER="Y"
/etc/tlp.conf L0246: BAY_POWEROFF_ON_AC="0"
/etc/tlp.conf L0247: BAY_POWEROFF_ON_BAT="0"
defaults.conf L0036: BAY_DEVICE="sr0"
/etc/tlp.conf L0254: RUNTIME_PM_ON_AC="on"
/etc/tlp.conf L0255: RUNTIME_PM_ON_BAT="auto"
defaults.conf L0039: RUNTIME_PM_DRIVER_BLACKLIST="amdgpu mei_me nouveau nvidia pcieport radeon"
/etc/tlp.conf L0272: USB_AUTOSUSPEND="1"
/etc/tlp.conf L0284: USB_BLACKLIST_BTUSB="0"
/etc/tlp.conf L0289: USB_BLACKLIST_PHONE="1"
/etc/tlp.conf L0294: USB_BLACKLIST_PRINTER="1"
/etc/tlp.conf L0299: USB_BLACKLIST_WWAN="0"
/etc/tlp.conf L0309: USB_AUTOSUSPEND_DISABLE_ON_SHUTDOWN="1"
/etc/tlp.conf L0316: RESTORE_DEVICE_STATE_ON_STARTUP="0"
defaults.conf L0047: RESTORE_THRESHOLDS_ON_BAT="0"
/etc/tlp.conf L0369: NATACPI_ENABLE="1"
/etc/tlp.conf L0370: TPACPI_ENABLE="1"
/etc/tlp.conf L0371: TPSMAPI_ENABLE="1"
/etc/tlp.conf L0020: TLP_DEFAULT_MODE="BAT"
/etc/tlp.conf L0055: CPU_SCALING_GOVERNOR_ON_AC="powersave"
/etc/tlp.conf L0056: CPU_SCALING_GOVERNOR_ON_BAT="powersave"
/etc/tlp.conf L0062: CPU_SCALING_MIN_FREQ_ON_AC="800000"
/etc/tlp.conf L0063: CPU_SCALING_MAX_FREQ_ON_AC="2800000"
/etc/tlp.conf L0064: CPU_SCALING_MIN_FREQ_ON_BAT="800000"
/etc/tlp.conf L0065: CPU_SCALING_MAX_FREQ_ON_BAT="2800000"
/etc/tlp.conf L0072: CPU_HWP_ON_AC="balance_performance"
/etc/tlp.conf L0073: CPU_HWP_ON_BAT="balance_power"
/etc/tlp.conf L0080: CPU_MIN_PERF_ON_AC="0"
/etc/tlp.conf L0081: CPU_MAX_PERF_ON_AC="100"
/etc/tlp.conf L0082: CPU_MIN_PERF_ON_BAT="0"
/etc/tlp.conf L0083: CPU_MAX_PERF_ON_BAT="30"
/etc/tlp.conf L0091: CPU_BOOST_ON_AC="1"
/etc/tlp.conf L0092: CPU_BOOST_ON_BAT="0"
/etc/tlp.conf L0117: ENERGY_PERF_POLICY_ON_AC="balance_performance"
/etc/tlp.conf L0118: ENERGY_PERF_POLICY_ON_BAT="balance_power"
/etc/tlp.conf L0142: DISK_SPINDOWN_TIMEOUT_ON_AC="0 0"
/etc/tlp.conf L0143: DISK_SPINDOWN_TIMEOUT_ON_BAT="0 0"
[... 223 lines cut ...]
WWAN blacklist      = disabled

Bus 002 Device 001 ID 1d6b:0003 control = auto, autosuspend_delay_ms =    0 -- Linux Foundation 3.0 root hub (hub)
Bus 001 Device 004 ID 8087:0a2a control = auto, autosuspend_delay_ms = 2000 -- Intel Corp.  (btusb)
Bus 001 Device 003 ID 0bda:58eb control = auto, autosuspend_delay_ms = 2000 -- Realtek Semiconductor Corp.  (uvcvideo)
Bus 001 Device 002 ID 0fce:01e8 control = on,   autosuspend_delay_ms = 2000 -- Sony Ericsson Mobile Communications AB  (no driver)
Bus 001 Device 001 ID 1d6b:0002 control = auto, autosuspend_delay_ms =    0 -- Linux Foundation 2.0 root hub (hub)

+++ Battery Features: Charge Thresholds and Recalibrate
natacpi    = inactive (laptop not supported)
tpacpi-bat = inactive (laptop not supported)
tp-smapi   = inactive (laptop not supported)

+++ Battery Status: BAT0
/sys/class/power_supply/BAT0/manufacturer                   = HP
/sys/class/power_supply/BAT0/model_name                     = Primary
/sys/class/power_supply/BAT0/cycle_count                    = (not supported)
/sys/class/power_supply/BAT0/charge_full_design             =   4576 [mAh]
/sys/class/power_supply/BAT0/charge_full                    =   4576 [mAh]
/sys/class/power_supply/BAT0/charge_now                     =   4576 [mAh]
/sys/class/power_supply/BAT0/current_now                    =      0 [mA]
/sys/class/power_supply/BAT0/status                         = Full

Charge                                                      =  100.0 [%]
Capacity                                                    =  100.0 [%]

linrunner, 2020-01-22:
I'd say it still works.

nnmlss, 2020-01-23:
Yes, I can confirm - it works.