linrunner/TLP #448: Frequency scaling is not working on kernel 5.3

TiGR, 2020-01-08:
On Linux 5.3 (5.3.0-26 on Bionic [KDE Neon]) with i5-2450M, CPU frequency somehow locks at the lowest values on both AC and BAT. Intel p-state is used. Related settings are:

```
#CPU_SCALING_GOVERNOR_ON_AC=powersave
#CPU_SCALING_GOVERNOR_ON_BAT=powersave
#CPU_SCALING_MIN_FREQ_ON_AC=0
CPU_SCALING_MAX_FREQ_ON_AC=2500
#CPU_SCALING_MIN_FREQ_ON_BAT=0
CPU_SCALING_MAX_FREQ_ON_BAT=900
CPU_HWP_ON_AC=balance_performance
CPU_HWP_ON_BAT=balance_power
#CPU_MIN_PERF_ON_AC=0
CPU_MAX_PERF_ON_AC=80
#CPU_MIN_PERF_ON_BAT=0
CPU_MAX_PERF_ON_BAT=30
CPU_BOOST_ON_AC=0
#CPU_BOOST_ON_BAT=0
```

As you see, on AC max frequency is supposed to be 2500. But it never gets above 1100, and most of the time it is in 800s even under heavy load. Needless to say, everything is very, very slow, you can't even watch a video without it stuttering constantly.

Running `cpupower -c all frequency-set -u 2500Mhz` fixes things, and frequency starts to scale normally.

Before the upgrade to 5.0 it was okay.

**Expected behavior**

To run as it was any time before.

**To Reproduce**

Steps to reproduce the unexpected behavior:

1. Does the problem occur on battery or AC or both? Both.
2. Actions to reproduce the behaviour. Use kernel 5.3, I suppose?
3. Shell commands entered and their output
4. tlp-stat: https://gist.github.com/TiGR/b1c62add2058fd59e46cc13a60e54b26

linrunner, 2020-01-08:
Let's see:

- TLP hasn't changed
- Kernel API hasn't changed
- The only thing that changed was the kernel version
- So why do you think it's a TLP issue? 

I suggest to file a kernel bug instead.

TiGR, 2020-01-08:
I'm not sure if kernel API has not changed.

Why I think that it's a TLP issue?

1. Well, I use standard CLI tools to set p-state modes, and it works perfectly.
2. Then, I stop tlp service, and during its shutdown, it resets back to this slow condition.
3. Then, I manually change the frequency with above command. It works well.
4. Then I start TLP, and everything is slow again.
5. Go to repeat from the step 1.

Is it TLP issue? Well, maybe it's not, but from my point of view, TLP does something wrong. And even if it doesn't, I still won't be able to file kernel bug, since I have no idea of what happens in TLP under the hood.

linrunner, 2020-01-08:
First: don't try to disable or stop TLP by stopping the service. That's *plain WRONG* no matter how often you read it on the net. The documentation is clear about that: either configure `TLP_ENABLE=0` or uninstall and reboot.

Second: don't try to apply changed settings by (re-)starting the service. Also *plain WRONG*. The documentation tells to do it with

     tlp start

But the problem with applying new settings manually is, that it happens long after the change-of-power-source event. My suspicion is that if TLP does this right at the event, the kernel refuses to accept the new settings for unkown reasons. This is not the first issue about that.

Bottom line: I need traces. 

Please:
- Enable [Trace Mode](https://linrunner.de/en/tlp/docs/tlp-linux-advanced-power-management.html#trace)
- Unplug the charger
- Plug the charger
- run `tlp start`
- Show the output of `tlp-stat -T`

TiGR, 2020-01-09:
[Trace](https://gist.github.com/TiGR/b5b87dd57de48cc8801ac73a22e235ab)

I might have done some things in wrong order, but anyway, that's entire captured trace.

linrunner, 2020-01-10:
The trace confirms my suspicion. Look at these excerpts:

**tlp start invoked manually**

    09 14:56:47 lenovo-g570 tlp[13715]: get_sys_power_supply(ACAD).ac_online: syspwr=0
    09 14:56:47 lenovo-g570 tlp[13715]: +++ start (1.2.2) ++++++++++++++++++++++
    09 14:56:47 lenovo-g570 tlp[13715]: power_source=ac
    09 14:56:47 lenovo-g570 tlp[13715]: manual_mode=none
    09 14:56:47 lenovo-g570 tlp[13715]: power_mode=ac
    09 14:56:47 lenovo-g570 tlp[13715]: set_scaling_min_max_freq(0).max: 2500; ec=0 
    09 14:56:47 lenovo-g570 tlp[13715]: set_scaling_min_max_freq(0).max: 2500; ec=0
    09 14:56:48 lenovo-g570 tlp[13715]: set_scaling_min_max_freq(0).max: 2500; ec=0
    09 14:56:48 lenovo-g570 tlp[13715]: set_scaling_min_max_freq(0).max: 2500; ec=0

**System startup**

    09 15:00:46 lenovo-g570 tlp[960]: +++ init start (1.2.2) ++++++++++++++++++++++
    09 15:00:46 lenovo-g570 tlp[960]: power_source=ac
    09 15:00:46 lenovo-g570 tlp[960]: manual_mode=none
    09 15:00:46 lenovo-g570 tlp[960]: power_mode=ac
    09 15:00:46 lenovo-g570 tlp[960]: set_scaling_min_max_freq(0).max: 2500; ec=0
    09 15:00:46 lenovo-g570 tlp[960]: set_scaling_min_max_freq(0).max: 2500; ec=0
    09 15:00:46 lenovo-g570 tlp[960]: set_scaling_min_max_freq(0).max: 2500; ec=0
    09 15:00:46 lenovo-g570 tlp[960]: set_scaling_min_max_freq(0).max: 2500; ec=0

**AC unplugged**

    09 15:00:51 lenovo-g570 tlp[1415]: +++ auto (1.2.2) ++++++++++++++++++++++++
    09 15:00:51 lenovo-g570 tlp[1415]: power_source=bat
    09 15:00:51 lenovo-g570 tlp[1415]: manual_mode=none
    09 15:00:51 lenovo-g570 tlp[1415]: power_mode=bat
    09 15:00:52 lenovo-g570 tlp[1415]: set_scaling_min_max_freq(1).max: 900; ec=0
    09 15:00:52 lenovo-g570 tlp[1415]: set_scaling_min_max_freq(1).max: 900; ec=0
    09 15:00:52 lenovo-g570 tlp[1415]: set_scaling_min_max_freq(1).max: 900; ec=0
    09 15:00:52 lenovo-g570 tlp[1415]: set_scaling_min_max_freq(1).max: 900; ec=0

**AC plugged in**

    09 15:01:04 lenovo-g570 tlp[2335]: get_sys_power_supply(ACAD).ac_online: syspwr=0
    09 15:01:04 lenovo-g570 tlp[2335]: +++ auto (1.2.2)  ++++++++++++++++++++++++
    09 15:01:04 lenovo-g570 tlp[2335]: power_source=ac
    09 15:01:04 lenovo-g570 tlp[2335]: manual_mode=none
    09 15:01:04 lenovo-g570 tlp[2335]: power_mode=ac
    09 15:01:04 lenovo-g570 tlp[2335]: set_scaling_min_max_freq(0).max: 2500; ec=0
    09 15:01:04 lenovo-g570 tlp[2335]: set_scaling_min_max_freq(0).max: 2500; ec=0
    09 15:01:04 lenovo-g570 tlp[2335]: set_scaling_min_max_freq(0).max: 2500; ec=0
    09 15:01:04 lenovo-g570 tlp[2335]: set_scaling_min_max_freq(0).max: 2500; ec=0

`ec=0` says: all values written to the kernel sysfile without errors.

My conclusion: either the kernel silently ignores the values written (kernel problem) or some other tool changes them after tlp wrote them. 

Does cpupower contain a service? I suggest to uninstall it.

ps. does the "sluggishness" persist when you disable CPU_SCALING_MAX_FREQ_ON*?

btw. limiting frequencies is kind of deprecated for intel_pstate. Did you look at the CPU_MIN/MAX_PERF parameters?

linrunner, 2020-01-10:
OK, I've only just realized you use CPU_MAX_PERF_ON_AC/BAT too. You really shouldn't use it it together with CPU_SCALING_MAX_FREQ*.

TiGR, 2020-01-11:
Yes, removing CPU_SCALING_MAX_FREQ* options helped. I used it to prevent CPU from turbo boosting, since disabling it via CPU_BOOST_* in options didn't work (at least with older kernels).

It seems that starting with kernel 5.3 having both FREQ and PERF options is incompatible.

linrunner, 2020-01-11:
> It seems that starting with kernel 5.3 having both FREQ and PERF options is incompatible.
I think I should implement some locking and/or warning messages here. I put it on my list for 1.4. Don't close please.

LinuxOnTheDesktop, 2020-02-04:
Datapoint (or different problem): on tlp 1.3, when tlp is in battery mode, my CPU is limited to 800Mhz, whereas its base clock is 1800Mhz; that happens even if I enable boosting on battery and dial up all performance hints. On tlp 1.1-2, everything works: on battery, with boosting off and medium performance hints, I get 1800Mhz. I am kernel 5.3; I have not yet tested tlp 1.3 on other kernels (and indeed I have downgraded tlp to 1.1-2 and locked that version).

linrunner, 2020-02-04:
@LinuxOnTheDesktop : you have a different problem that is dealt with here: https://github.com/linrunner/TLP/issues/460

Fitz13009, 2020-02-06:
Hey,
Just write down something here because I also have a similar problem in Kernel 5.3. 

The cpu frequecny scaling doesn't work properly. The configuration "CPU_MAX(MIN)_PERF_ON_AC(BAT)" set failed and the cpu cannot do turbo boost after switch from battery to AC charger.

But after updating the kernel to 5.4.18, everything works as expected.

Fitz13009, 2020-02-06:
@LinuxOnTheDesktop I think Kernel 5.3 has some bugs. I also use TLP 1.3 and it failed in Turbo boost.  When I update my kernel, things work fine.

LinuxOnTheDesktop, 2020-02-07:
@Fitz13009

linrunner was right to say that my problem was different; it was the problem treated in #460

linrunner, 2020-11-10:
Since the problem seems to occur only with 5.3, I scrap my initial idea of 

> locking and/or warning message

This would just confuse users of other/newer kernels.  

The [documentation](https://linrunner.de/tlp/settings/processor.html#cpu-scaling-min-max-freq-on-ac-bat) already contains a hint and tlp.conf too now (see commit).

closed 2020-11-10 by linrunner