linrunner/TLP #449: [TLP 1.3 Beta] Call for testers

linrunner, 2020-01-08:
Hi @all,

I need help with the upcoming release.

**What has changed?**
From a user perspective, the most important change is the **new configuration scheme**
- **/etc/default/tlp** is replaced by **/etc/tlp.conf**
- Settings are read in the following order:
1. Intrinsic defaults
2. **/etc/tlp.d/*.conf** - Drop-in customizations
3. **/etc/tlp.conf**     - User configuration

- In case of identical parameters, the last occurence has precedence.
- Config files are now parsed instead of sourced by the shell – no more shell expansion.

Apart from that 1.3 is a comprehensive maintenance release. See the [changelog](https://github.com/linrunner/TLP/blob/master/changelog) for details.

**Where do I get packages?**
See the [download page](https://download.linrunner.de/packages/) for Arch Linux, Debian, Fedora and Ubuntu. If your distro is not contained, you may install the master branch (see ["Installing from Source"](https://linrunner.de/en/tlp/docs/tlp-developer-documentation.html#sourcecode)).

**Test Objective: Regressions**
The new configuration scheme is obviousy critical functionality. The maintenance release contains many other changes. With your help I want to make sure that no new problems have been introduced.

**What do I need from the testers?**
I'm expecting your outputs of

    sudo tlp-stat

on battery and AC (via https://gist.github.com/ please). 

**Thanks for testing!**

zauster, 2020-01-08:
[AC](https://gist.github.com/zauster/b63f363af42c7453ffc02a848bdd2042), [battery](https://gist.github.com/zauster/6763fc237cf34b12a2f5337c98cc017c)

sodaplayer, 2020-01-09:
>     2. **/etc/tlp.d/*.conf** - Drop-in customizations

Should this be /etc/tlp.**conf**.d/*.conf?

sodaplayer, 2020-01-09:
[AC](https://gist.github.com/sodaplayer/04a6c904b25d719c88ec90efb686176c), [battery](https://gist.github.com/sodaplayer/9a8bf0f80a833505079a7378ce4cc284)

dongsibo, 2020-01-09:
[AC](https://gist.github.com/dongsibo/f4be895a81e7e835bada1ab3ba9dd626), [battery](https://gist.github.com/dongsibo/a821af2cbd87d07c53ac55c1be8f6e0d)

linrunner, 2020-01-09:
@sodaplayer : it was with 1.3.0-beta.1, but since 1.3.0-beta.2 it's /etc/tlp.d/.

BarbUk, 2020-01-09:
[ac](https://gist.github.com/BarbUk/997f5af92312402e20dbe2a91f9b0ebe)
[bat](https://gist.github.com/BarbUk/cb6c6e66c7626f211551fe40c50d4797)

kaewhyes, 2020-01-09:
[AC](https://gist.github.com/kaewhyes/3cbe01d52bc7c57ce5d650ead24a29c9), and [battery](https://gist.github.com/kaewhyes/fa1e320f4860e3661b634f94cc7d606b)

linrunner, 2020-01-09:
@kaewhyes: 
> Notice: tlp.service is not enabled -- invoke "systemctl enable tlp.service" to correct this!

Enabling this service is mandatory (to apply settings on system start). Please enable, restart on AC and post tlp-stat output again. 

@ALL: thanks for your effort.

nariox, 2020-01-09:
[AC](https://gist.github.com/nariox/dc0f5d9515ee1ef79ed7909d41f697f8) and [Battery](https://gist.github.com/nariox/e863aa488203eaccaacfb9d02049c815)

KJoke70, 2020-01-09:
[AC](https://gist.github.com/KJoke70/b6629258994cbebac4b8ffb62866ce52) and [Battery](https://gist.github.com/KJoke70/3b3d52ca4fac0325a27dc9b78a9c2f7d)

beuerle, 2020-01-09:
[AC](https://gist.github.com/beuerle/e49dcea1b7923f93b71750e80650fe2d), [battery](https://gist.github.com/beuerle/d1c9e9aed7507caffdb7b30965f4ad17)

kaewhyes, 2020-01-09:
> Enabling this service is mandatory (to apply settings on system start). Please enable, restart on AC and post tlp-stat output again.

Shoot, I didn't realize.
Here's the updated [AC](https://gist.github.com/kaewhyes/c3feaf178f9056e3ad56c046b8b470db) and [battery](https://gist.github.com/kaewhyes/a09b98c50269890de434ef9ec6567386)

SebRut, 2020-01-10:
[battery and ac](https://gist.github.com/SebRut/5a3d85d78cbcb4f55aca62c7f2489f23)

arvl130, 2020-01-11:
TLP stat output: [AC and BAT](https://gist.github.com/arvl130/993f77e031bb645aeac67ef0afdb8ba8)

linrunner, 2020-01-11:
Packages for Debian and Fedora now available from (beta) repositories.

Alkotronikk, 2020-01-12:
Fedora 31

TLP stat output: [AC](https://gist.github.com/Alkotronikk/27903bee649ded7789a0b2c89cb400ba), [Bat](https://gist.github.com/Alkotronikk/50a4965f3ece917cef24aafc5906d4c2)

Also I'd like to point out that there is a mistake in your Fedora commands to download and switch to beta:

```dnf upgrade--enablerepo=tlp-updates-testing tlp tlp-rdw```

Should be:

```dnf upgrade --enablerepo=tlp-updates-testing tlp tlp-rdw```

There was also mistake in the first command, which I believe got fixed already.

nicolasdb, 2020-01-12:
TLP 1.3 Beta test 
Thinkpad T430 
SOLUS4
https://gist.github.com/nicolasdb/a069847baaaa4952c27e85165f7d121f

linrunner, 2020-01-12:
@nicolasdb : Thanks for participatin. Solus seems to have it's own package standard. If you really need the recalibration feature, you should ask at a Solus forum if an acpi_call package is available.

linrunner, 2020-01-12:
@Alkotronikk : fixed. Thanks.

linrunner, 2020-01-12:
@Alkotronikk : could you please generate the AC output once again? The new version was not started per 'tlp start'.

stefangweichinger, 2020-01-12:
https://gist.github.com/stefangweichinger/d9017053096c4da75f5e2c6aee2bdc78

on bat in first part, on AC in comment ...

montj2, 2020-01-12:
Dell Precision 5530 
[AC](https://gist.github.com/montj2/10e1a6552da1f291de5047d5924c6eb6)
[BATT](https://gist.github.com/montj2/357921bffea452b3aaac9418d2afbc6e)

linrunner, 2020-01-12:
@Alkotronikk : SELinux must be "permissive" or "disabled" for TLP to work :(.

Huluti, 2020-01-12:
Dell Inspiron 5370
[AC](https://gist.github.com/Huluti/b7537319cb6c5fe28bdffc007645d9e7)
[BATTERY](https://gist.github.com/Huluti/bc9bb7a78c1e67fff1e588d68213ef73)

Alkotronikk, 2020-01-12:
@linrunner Sorry, I didn't realize that. Updated my [AC](https://gist.github.com/Alkotronikk/27903bee649ded7789a0b2c89cb400ba) and [Bat](https://gist.github.com/Alkotronikk/50a4965f3ece917cef24aafc5906d4c2) gists with selinux set to permissive.

r4hul-in, 2020-01-12:
[image], and [image].

r4hul-in, 2020-01-12:
[AC](https://gist.github.com/r4hul-in/49a662008b67d2aa5833468bae29c30e), and [BAT](https://gist.github.com/r4hul-in/f85b73b1f1c27151e40fab389fdd1a27).

SteelChickens, 2020-01-13:
It finally suspends my touchscreen.

[tlp-ac.txt](https://github.com/linrunner/TLP/files/4051863/tlp-ac.txt)
[tlp-batt.txt](https://github.com/linrunner/TLP/files/4051865/tlp-batt.txt)

linrunner, 2020-01-13:
@Alkotronikk : no need to apologize. I only became aware through your outputs that there is a SELinux problem with TLP under Fedora.

validatedev, 2020-01-13:
AC and battery outputs: https://gist.github.com/validatedev/14f1edc3c6cc5087ddf21c4cb23d933c
It seems TLP doesn't respect my device blacklist setting but respect driver blacklist.

linrunner, 2020-01-13:
@validatedev : unfortunately you left out all the relevant information:

1. RUNTIME_PM_BLACKLIST is not configured (at least I guess you mean that parameter
2. You tell nothing about the device you want to blacklist

Please change your config to the one where the problem occurs, reboot, show the output again and mark the device question

validatedev, 2020-01-13:
> @validatedev : unfortunately you left out all the relevant information:
> 
>     1. RUNTIME_PM_BLACKLIST is not configured (at least I guess you mean that parameter)
> 
>     2. You tell nothing about the device you want to blacklist
> 
> 
> Please change your config to the one where the problem occurs, reboot, show the output again and mark the device in question

I did configure `RUNTIME_PM_BLACKLIST` in `tlp.conf`, I just assumed that the information is on the output of `tlp-stat`. There is the gist including `tlp.conf` and out of `tlp-stat`.

linrunner, 2020-01-13:
@validatedev : you're right and you found a regression. Congrats.

 Could you please change **/usr/share/tlp/tlp-readconf** line 99 hand?

    old: if ( $line =~ /^(?<name>[A-Z_]+[0-9]*)=(?:(?<val_bare>[0-9a-zA-Z_\-:]*)|"(?<val_dquoted>[0-9a-zA-Z _\-:]*)")\s*$/ ) {
    new: if ( $line =~ /^(?<name>[A-Z_]+[0-9]*)=(?:(?<val_bare>[0-9a-zA-Z_\-:\.]*)|"(?<val_dquoted>[0-9a-zA-Z _\-:\.]*)")\s*$/ ) {

Afterwards show 

    tlp-stat -c 

again.

validatedev, 2020-01-13:
@linrunner The change that you have provided works.
Gist: https://gist.github.com/validatedev/ff38f38704beed5d7d01cdc9180f04ac

linrunner, 2020-01-14:
@validatedev : merged. Thanks for your testing and feedback.

Anonymo, 2020-01-15:
`--- TLP 1.3.0-beta.3 --------------------------------------------

+++ Configured Settings:
defaults.conf L0004: TLP_ENABLE="1"
defaults.conf L0005: TLP_PERSISTENT_DEFAULT="0"
defaults.conf L0006: DISK_IDLE_SECS_ON_AC="0"
defaults.conf L0007: DISK_IDLE_SECS_ON_BAT="2"
defaults.conf L0008: MAX_LOST_WORK_SECS_ON_AC="15"
defaults.conf L0009: MAX_LOST_WORK_SECS_ON_BAT="60"
defaults.conf L0010: CPU_ENERGY_PERF_POLICY_ON_AC="balance_performance"
defaults.conf L0011: CPU_ENERGY_PERF_POLICY_ON_BAT="power"
defaults.conf L0012: SCHED_POWERSAVE_ON_AC="0"
defaults.conf L0013: SCHED_POWERSAVE_ON_BAT="1"
defaults.conf L0014: NMI_WATCHDOG="0"
defaults.conf L0015: DISK_DEVICES="nvme0n1 sda"
defaults.conf L0016: DISK_APM_LEVEL_ON_AC="254 254"
defaults.conf L0017: DISK_APM_LEVEL_ON_BAT="128 128"
defaults.conf L0018: DISK_IOSCHED="keep keep"
defaults.conf L0019: SATA_LINKPWR_ON_AC="med_power_with_dipm max_performance"
defaults.conf L0020: SATA_LINKPWR_ON_BAT="med_power_with_dipm min_power"
defaults.conf L0021: AHCI_RUNTIME_PM_TIMEOUT="15"
defaults.conf L0022: PCIE_ASPM_ON_AC="default"
defaults.conf L0023: PCIE_ASPM_ON_BAT="default"
defaults.conf L0024: RADEON_POWER_PROFILE_ON_AC="default"
defaults.conf L0025: RADEON_POWER_PROFILE_ON_BAT="default"
defaults.conf L0026: RADEON_DPM_PERF_LEVEL_ON_AC="auto"
defaults.conf L0027: RADEON_DPM_PERF_LEVEL_ON_BAT="auto"
defaults.conf L0028: WIFI_PWR_ON_AC="off"
defaults.conf L0029: WIFI_PWR_ON_BAT="on"
defaults.conf L0030: WOL_DISABLE="Y"
defaults.conf L0031: SOUND_POWER_SAVE_ON_AC="0"
defaults.conf L0032: SOUND_POWER_SAVE_ON_BAT="1"
defaults.conf L0033: SOUND_POWER_SAVE_CONTROLLER="Y"
defaults.conf L0034: BAY_POWEROFF_ON_AC="0"
defaults.conf L0035: BAY_POWEROFF_ON_BAT="0"
defaults.conf L0036: BAY_DEVICE="sr0"
defaults.conf L0037: RUNTIME_PM_ON_AC="on"
defaults.conf L0038: RUNTIME_PM_ON_BAT="auto"
defaults.conf L0039: RUNTIME_PM_DRIVER_BLACKLIST="amdgpu mei_me nouveau nvidia pcieport radeon"
defaults.conf L0040: USB_AUTOSUSPEND="1"
defaults.conf L0041: USB_BLACKLIST_BTUSB="0"
defaults.conf L0042: USB_BLACKLIST_PHONE="0"
defaults.conf L0043: USB_BLACKLIST_PRINTER="1"
defaults.conf L0044: USB_BLACKLIST_WWAN="0"
defaults.conf L0045: USB_AUTOSUSPEND_DISABLE_ON_SHUTDOWN="0"
defaults.conf L0046: RESTORE_DEVICE_STATE_ON_STARTUP="0"
defaults.conf L0047: RESTORE_THRESHOLDS_ON_BAT="0"
defaults.conf L0048: NATACPI_ENABLE="1"
defaults.conf L0049: TPACPI_ENABLE="1"
defaults.conf L0050: TPSMAPI_ENABLE="1"
/etc/tlp.conf L0424: START_CHARGE_THRESH_BAT0="75"
/etc/tlp.conf L0425: STOP_CHARGE_THRESH_BAT0="80"

+++ System Info
System         = LENOVO ThinkPad T495 20NJ0006US
BIOS           = R12ET47W(1.17 )
Kernel         = 5.4.12-AMD #1 SMP PREEMPT Tue Jan 14 21:41:22 CET 2020 x86_64
/proc/cmdline  = BOOT_IMAGE=/boot/vmlinuz-linux-amd root=UUID=8e9e6bd1-ba36-4b13-84b5-e0e9294412e4 rw loglevel=3 quiet
Init system    = systemd 
Boot mode      = UEFI

+++ TLP Status
State          = enabled
RDW state      = not installed
Last run       = 06:57:05 PM,     79 sec(s) ago
Mode           = AC
Power source   = AC

+++ Processor
CPU model      = AMD Ryzen 7 PRO 3700U w/ Radeon Vega Mobile Gfx
[... 178 lines cut ...]
+++ Battery Features: Charge Thresholds and Recalibrate
natacpi    = active (data, thresholds)
tpacpi-bat = active (recalibrate)
tp-smapi   = inactive (ThinkPad not supported)

+++ ThinkPad Battery Status: BAT0 (Main / Internal)
/sys/class/power_supply/BAT0/manufacturer                   = LGC
/sys/class/power_supply/BAT0/model_name                     = 02DL007
/sys/class/power_supply/BAT0/cycle_count                    =      3
/sys/class/power_supply/BAT0/energy_full_design             =  50500 [mWh]
/sys/class/power_supply/BAT0/energy_full                    =  51480 [mWh]
/sys/class/power_supply/BAT0/energy_now                     =  48910 [mWh]
/sys/class/power_supply/BAT0/power_now                      =      0 [mW]
/sys/class/power_supply/BAT0/status                         = Unknown (threshold may prevent charging)

/sys/class/power_supply/BAT0/charge_start_threshold         =     75 [%]
/sys/class/power_supply/BAT0/charge_stop_threshold          =     80 [%]
tpacpi-bat.BAT0.forceDischarge                              =      0

Charge                                                      =   95.0 [%]
Capacity                                                    =  101.9 [%]

+++ Recommendations
* Install ethtool to disable Wake On LAN (WOL)
* Install smartmontools for disk drive health info`

linrunner, 2020-01-15:
@Anonymo : thanks for your outputs. Next time via https://gist.github.com/ please. 

BAT, AC: https://gist.github.com/linrunner/c45bd3260a67b9d85e5c7078d241a187

validatedev, 2020-01-15:
@linrunner I'm still beta 2. Why it didn't appear on me?

linrunner, 2020-01-15:
@validatedev : what didn't appear?

validatedev, 2020-01-15:
@linrunner When I saw the gist of the @Anonymo , I realized that he/she is on TLP 1.3 Beta 3 but even if I try `sudo apt update`, the update didn't appear on me. It's the case because I'm on Ubuntu 20.04? I mean the `focal` repository.

linrunner, 2020-01-15:
@validatedev  : beta.3 is not released yet, but Arch Linux has a special package tlp-git (in AUR) which tracks the git master branch.

validatedev, 2020-01-15:
@linrunner Ah okay then, thanks!

Anonymo, 2020-01-15:
> @Anonymo : thanks for your outputs. Next time via https://gist.github.com/ please.

Sorry, wasn't familiar with Gist. I thought the CODE thing would condense it but it didn't.  I was gonna erase but but I figured it would be helpful anyways.  Thanks for fixing it for me.

linrunner, 2020-01-17:
Thank you very much for your contributions so far :).

1.3.0-beta.3 is out – Please re-test and show your tlp-stat outputs!

dongsibo, 2020-01-17:
[AC](https://gist.github.com/dongsibo/7c547e271733fe831b057aa70e18f4c0 'Github Gist - tlp-stat AC output'), [battery](https://gist.github.com/dongsibo/0b113965fa4604d218e4640f90256e29 'GitHub Gist - tlp-stat battery output')

BarbUk, 2020-01-17:
[AC and bat](https://gist.github.com/BarbUk/98c51369bb5f2989492ed881ef40c730)

dongsibo, 2020-01-18:
@BarbUk I took a look at your `tlp-stat` outputs from TLP 1.3.0-beta.3 and noticed that you don't have the same configured settings that you did for TLP 1.3.0–beta.2. For example, you are now missing the settings for setting Intel CPU P-state performance:
```
# The following are not present
...
/etc/tlp.conf L0109: CPU_MIN_PERF_ON_AC="0"
/etc/tlp.conf L0110: CPU_MAX_PERF_ON_AC="100"
/etc/tlp.conf L0111: CPU_MIN_PERF_ON_BAT="0"
/etc/tlp.conf L0112: CPU_MAX_PERF_ON_BAT="30"
...
```
I don't mean to pry but is this intentional? Note that the current AUR package for `tlp-git` has `backup` set to [`etc/default/tlp`](https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=tlp-git#n49 'AUR - tlp-git PKGBUILD') instead of `etc/tlp.conf` as mentioned [here](https://aur.archlinux.org/packages/tlp-git/#comment-723764 'AUR -tlp-git Comment') which may have overwitten `/etc/tlp.conf` without creating a `.pacnew` file when you updated to 1.3.0-beta.3.

This is a [patch](https://gist.github.com/dongsibo/c87b65703e6c100859128c0af300be8f 'GitHub Gist - tlp-git.patch') for the `PKGBUILD` (that also addresses the use of both en dashes and em dashes in the beta tags).

pflanzenandi, 2020-01-18:
[AC](https://github.com/pflanzenandi/bla/blob/master/AC)
[BAT](https://github.com/pflanzenandi/bla/blob/master/BAT)

BarbUk, 2020-01-19:
@dongsibo 
> I don't mean to pry but is this intentional?

Good catch, my config was replaced by the default following the upgrade to the last beta.

Updated [AC and bat](https://gist.github.com/7118aecdb57c343a229b15073063457d)

linrunner, 2020-01-19:
@dongsibo : thanks for your hint about the dashes in the beta tags, these should all be hyphens, Maybe my keyboard failed ... ;)

nnmlss, 2020-01-21:
[AC](https://github.com/linrunner/TLP/issues/446#issuecomment-576640803) [Battery](https://github.com/linrunner/TLP/issues/446#issuecomment-576641056)

KJoke70, 2020-01-21:
[AC](https://gist.github.com/KJoke70/f713db76756cc876667926019c83c2d3#file-tlp-1-3-beta-3-ac), [Battery](https://gist.github.com/KJoke70/f713db76756cc876667926019c83c2d3#file-tlp-1-3-beta-3-battery). The threshold was reset, but I'm assuming this was because I did a firmware-upgrade.

dongsibo, 2020-01-21:
@KJoke70 See [here](https://github.com/linrunner/TLP/issues/449#issuecomment-575938221).

KJoke70, 2020-01-21:
@dongsibo As far as I can tell I was already on the defaults in beta-2. Also, the thresholds don't have a default so they shouldn't change.

dongsibo, 2020-01-21:
@KJoke70 Oh yeah, that makes sense.

linrunner, 2020-01-25:
Thank you very much for your contributions again :).

**1.3.0-beta.4** is out – Please re-test and show your tlp-stat outputs!

dongsibo, 2020-01-25:
[AC](https://gist.github.com/dongsibo/234109239824547b4ba4e0d5e7af12a4#file-tlp-stat-1-3-0-beta-4-ac 'GitHub Gist - tlp-stat AC output') and [battery](https://gist.github.com/dongsibo/234109239824547b4ba4e0d5e7af12a4#file-tlp-stat-1-3-0-beta-4-bat 'GitHub Gist - tlp-stat battery output')

BarbUk, 2020-01-25:
[AC and battery](https://gist.github.com/712427621fd409e745d0cc4603f1b356
)

KJoke70, 2020-01-25:
[AC](https://gist.github.com/KJoke70/e25bc7ecfcf99dc43dd836594b6d6422#file-ac), [BAT](https://gist.github.com/KJoke70/e25bc7ecfcf99dc43dd836594b6d6422#file-bat)

stefangweichinger, 2020-01-25:
[both](https://gist.github.com/stefangweichinger/21b2e9e9ca957a62eb8fc84b100ca4b3)

[11 more replies]

closed 2020-01-31 by linrunner