linrunner/TLP #451: `tlp chargeonce` fails for 1.3beta

beuerle, 2020-01-09:
**Describe the bug**
- `tlp chargeonce` fails with the current 1.3beta version:

```bash
$ sudo tlp chargeonce
Error: incomplete or invalid threshold configuration for BAT0 (start: none, stop: none).
```

- installed `tlp` revision

```
aur/tlp-git 1.3.0b1.r1.e2fbc0c-1
```

- [tlp-stat](https://gist.github.com/beuerle/e49dcea1b7923f93b71750e80650fe2d)

**Expected behavior**

```bash
$ sudo tlp chargeonce
Setting temporary charge thresholds for BAT0:
  stop  = 80
  start = 76
```

dongsibo, 2020-01-10:
Your charge thresholds are currently 96 and 100. Have you set your battery thresholds in `/etc/tlp.conf`?

linrunner, 2020-01-10:
@dongsibo  is right and the error message says it too. tlp chargeonce works only when thresholds are configured.

beuerle, 2020-01-11:
This only popped up recently, after using tlp for years. I didn't do any changes.

> > /sys/class/power_supply/BAT0/charge_start_threshold         =     96 [%]
> > /sys/class/power_supply/BAT0/charge_stop_threshold          =    100 [%]
> 
> Your charge thresholds are currently 96 and 100. Have you specified your thresholds in `/etc/tlp.conf`?

In that case I was running `tlp fullcharge` to see if this was working, which it did.

In normal AC mode I get:

```bash
$ sudo tlp-stat | grep threshold
natacpi    = active (data, thresholds)
/sys/class/power_supply/BAT0/status                         = Unknown (threshold may prevent charging)
/sys/class/power_supply/BAT0/charge_start_threshold         =     40 [%]
/sys/class/power_supply/BAT0/charge_stop_threshold          =     80 [%]
```

Running `chargeonce` I am expecting `tlp` to set the start value to `76` and the stop value to `80`. At least that's how it was.

Do I have to specify this now somewhere in the configuration? Sorry for the confusion and your help.

- [`tlp.conf`](https://gist.github.com/beuerle/44153fec40e603cc3b2b49a0fa1fdc60)
- [tlp-stat](https://gist.github.com/beuerle/99884e0e526c9ced0d3d5cc2f93ebc7e)

linrunner, 2020-01-11:
You have active thresholds 40/80% from before, but you haven't configured them in **/etc/tlp.conf** yet. You need

    START_CHARGE_THRESH_BAT0=40
    STOP_CHARGE_THRESH_BAT0=80

Remember to remove the leading '#'s. 

Background: new configuration scheme comments out all parameters, because defaults are now defined elsewhere.

linrunner, 2020-01-11:
ps. the same applies to all customization you possibly had before.

beuerle, 2020-01-11:
Thanks for the clarification!

closed 2020-01-11 by beuerle