linrunner/TLP #453: Mode=AC Power source=Battery?

validatedev, 2020-01-13:
[x] I've read and accepted the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md)

**Describe the bug**

I plugged out my laptop but TLP doesn't immediately respect that. TLP knows I'm on the battery but doesn't switch. I had to run `sudo tlp start` to trigger the change or wait 1-2 min.

**Expected behavior**
TLP should change its mode immediately.

**To Reproduce**
Charge the battery 100%.
Plug out the cable from the laptop.
TLP won't take action immediately.

Output of `sudo tlp-stat`: https://gist.github.com/validatedev/1a2821d03619b15ba85dc7233a827edb

**Additional context**
I suspect that the issue https://github.com/AdnanHodzic/auto-cpufreq/issues/21 has the same root with that, but that is just the intuition.

linrunner, 2020-01-13:
I guess it's similar to #446 .

Please install 1.3 Beta: #449  Remember to read about the new configuration scheme in the 1st post and read the changelog too.

Then change your config file:

    TLP_PS_IGNORE=BAT

If it doesn't work then also try

    TLP_PS_IGNORE=AC

Please report back here, not in the beta issue.

validatedev, 2020-01-13:
Same issue still happens with 1.3 and `TLP_PS_IGNORE=BAT` or `TLP_PS_IGNORE=AC`.

amithiel, 2020-01-13:
Hi, i just noticed the same. when in bat mode, plugging the charger changes to AC mode. but the opposit does not happen. I unplugged my charger 10 min ago, and its still in ac mode

linrunner, 2020-01-13:
@amithiel : you don't have the same problem and you have different hardware, please keep out here and open your own issue. 

Remember to give the full outputs the bug report template tells you in a **readable form** i.e. via https://gist.github.com/

linrunner, 2020-01-13:
@validatedev : so this is not a reression from 1.2.2 to 1.3.

I need traces.

(1) Prerequisites

Enable trace mode by creating a file **/etc/tlp.d/00-debug.conf** containing only the line

    TLP_DEBUG="pm ps run"

 Connect AC.

(2) Disconnect the charger and _take note of the exact time hh:mm:ss_ you did it

(3) Show the trace

   tlp-stat -T

via https://gist.github.com/

validatedev, 2020-01-13:
I did plug out on exactly `23:41:30`.  Here is the output of `tlp-stat -T`:
https://gist.github.com/validatedev/2520c99b8b83016dafd82ef0c94e263f

validatedev, 2020-01-13:
Well, right now the `TLP_PS_IGNORE=BAT` works but the battery is not fully charged, waiting the charging. There is a improvement though.

validatedev, 2020-01-13:
It works flawlessly. I don't know why there is the problem at the first attempt. Maybe I misconfigured the configuration. Thanks for your interest!

closed 2020-01-13 by validatedev