linrunner/TLP #454: Thinkpad e495 - Fedora 31 - kernel 5.4.8 - TLP battery thresholds stopped working after kernel update from kernel 5.3.x

choeffer, 2020-01-13:
[x] I've read and accepted the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md)

After updating the kernel via dnf, thresholds seem to be ignored and the laptop charges till 100%.

**Expected behavior**

The thresholds should be applied and the device should stop charging at the `charge_stop_threshold` value.

**To Reproduce**

Steps to reproduce the unexpected behavior:

Update kernel to 5.4.x, set charging thresholds.

1. Does the problem occur on battery or AC or both? AC
2. Actions to reproduce the behaviour, see above.
3. Shell commands entered and their output
4. **Full output of `tlp-stat` via https://gist.github.com/ for *all*
   matching cases of 1.**

https://gist.github.com/choeffer/484cb7140559f996df1e4600a8d1e9c5

**Additional context**

If I can provide any logs which might help to narrow down the issue, please let me know.

linrunner, 2020-01-13:
Your output shows, that TLP works exactly as expected:

   /sys/class/power_supply/BAT0/charge_start_threshold         =     45 [%]
   /sys/class/power_supply/BAT0/charge_stop_threshold          =     60 [%]

TLP has written the thresholds to the Embedded Controller (EC). tlp-stat reads them back correctly.

Conclusion: the EC – or it's firmware – is the culprit.

Did you discharge once  below 45% yet? If not, try it. If that doesn't work, look for BIOS updates.

Apart from that I can't help you. TLP performs correctly.

choeffer, 2020-01-14:
Thanks for your fast reply.

And what could cause firmware/uefi stop working after a kernel/TLP update?

As I said, I have not applied any update regarding firmware/uefi in the meantime, the machine has no Windows dual-boot and the e495 is not part of LVFS. So there is no possibility a firmware/uefi update could have been applied accidentially before it stopped working. And the latest uefi is installed since beginning, so that part has not changed. I even have not touched the uefi settings in the meantime.

I also have tried discharging below 45% and then plugging in the notebook again, but it was directly charged to 100% again.

I am just wondering what else I can try to narrow down what is going on.

choeffer, 2020-01-15:
**Followed steps**

Started with battery level between `START_CHARGE_THRESH_BAT0`  and `STOP_CHARGE_THRESH_BAT0` (~65% at this time).

```
sudo tlp-stat
...
+++ Configured Settings: /etc/default/tlp
TLP_ENABLE=1
START_CHARGE_THRESH_BAT0=50
STOP_CHARGE_THRESH_BAT0=70
NATACPI_ENABLE=1
TPACPI_ENABLE=1
TPSMAPI_ENABLE=1
...
```

Executed `sudo tlp fullcharge` to set `START_CHARGE_THRESH_BAT0=96` and `STOP_CHARGE_THRESH_BAT0=100`  to get back to factory defaults (values were applied correctly).

Then also followed https://linrunner.de/en/tlp/docs/tlp-faq.html#stop-threshold-sticks precisely (e.g. disconnect AC AFTER shutdown completed) to reset the battery/controller.

After first startup, TLP set the thresholds back to `START_CHARGE_THRESH_BAT0=50` and `STOP_CHARGE_THRESH_BAT0=70` .

**Result**

Battery thresholds are applied correctly again and working, so it it/was a EC/firmware issue.

```
--- TLP 1.2.2 --------------------------------------------

+++ Battery Features: Charge Thresholds and Recalibrate
natacpi    = active (data, thresholds)
tpacpi-bat = inactive (kernel module 'acpi_call' not installed)
tp-smapi   = inactive (ThinkPad not supported)

+++ ThinkPad Battery Status: BAT0 (Main / Internal)
/sys/class/power_supply/BAT0/manufacturer                   = SMP
/sys/class/power_supply/BAT0/model_name                     = 01AV446
/sys/class/power_supply/BAT0/cycle_count                    =      7
/sys/class/power_supply/BAT0/energy_full_design             =  45280 [mWh]
/sys/class/power_supply/BAT0/energy_full                    =  48130 [mWh]
/sys/class/power_supply/BAT0/energy_now                     =  33480 [mWh]
/sys/class/power_supply/BAT0/power_now                      =      0 [mW]
/sys/class/power_supply/BAT0/status                         = Unknown (threshold effective)

/sys/class/power_supply/BAT0/charge_start_threshold         =     50 [%]
/sys/class/power_supply/BAT0/charge_stop_threshold          =     70 [%]

Charge                                                      =   69.6 [%]
Capacity                                                    =  106.3 [%]

+++ Recommendations
* Install acpi_call kernel module for ThinkPad battery recalibration
```

Thanks again for pointing me into the right direction.

linrunner, 2020-01-15:
Thanks for your feedback. I'll mention this the FAQ -> Reopen

damnsl, 2020-01-19:
I just came here to thank you! I got my **first** _10 year old thinkpad t410_ and reading about it came across this beauty. Now I can preserve My battery for a longer time!! (hopefully). 

once again thank you! Hope I'm not spamming. Please  don't be mad.

choeffer, 2020-01-21:
Thanks for adding the additional information in the FAQ. But the part with 

> Executed `sudo tlp fullcharge` to set `START_CHARGE_THRESH_BAT0=96` and `STOP_CHARGE_THRESH_BAT0=100` to get back to factory defaults (values were applied correctly).

is missing. So first `sudo tlp fullcharge` to get back to factory default values and then apply https://linrunner.de/en/tlp/docs/tlp-faq.html#stop-threshold-sticks.

linrunner, 2020-02-05:
Added to the FAQ. Thank you.

choeffer, 2020-05-01:
As with `BIOS = R11ET36W (1.16 )`, `sudo tlp fullcharge`+ reboot is sufficient to get thresholds working properly again.
I have had the issue, that after updating to the mentioned BIOS version, the thresholds have not been working anymore and the battery kept charging. After `sudo tlp fullcharge`+ reboot everything is working fine again and TLP restores the thresholds to the before defined. I will report if I encounter issues. If not, 

> then continue with the workaround for Disabling the charge thresholds does not work.

can be removed as addtional step, because it is not needed anymore.

linrunner, 2020-05-01:
Aye aye. FAQ change ok like this?

choeffer, 2020-05-01:
Perfectly described and summarized, thanks!