linrunner/TLP #459: tlp doesn't start

pltrz, 2020-02-02:
reproduced this installing via package manager (alpine) or installing manually from github (1.3.0 tag)

trying to run tlp manually:

```
# tlp init start
/usr/sbin/tlp: line 403: /usr/share/tlp/tlp-readconfs: not found
```

this doesn't make sense: `/usr/share/tlp/tlp-readconfs` does indeed exist.

dywisor, 2020-02-02:
Please post the output of

1. ```uname -r```

2. ```find /usr/share/tlp -ls | sort```

3. ```/usr/share/tlp/tpacpi-bat --help```  (= trying to run any other script in that directory)

nero, 2020-02-02:
[Downstream issue](https://gitlab.alpinelinux.org/alpine/aports/issues/11181)

linrunner, 2020-02-02:
Something seems wrong with the package. tlp-readconfs is new, but Perl scripts are not new to TLP. Please sort this with the Alpine maintainer.

pltrz, 2020-02-02:
Yes, it failed due to perl not being installed. The packager for tlp in Alpine also mentioned they are no longer interested in maintaining this package, so I will update the package in Alpine and add the perl dependency.

linrunner, 2020-02-02:
Thanks for your feedback. Please close here when it's done.

linrunner, 2020-02-05:
Solved downstream.

closed 2020-02-05 by linrunner