linrunner/TLP #460: Default CPU_ENERGY_PERF_POLICY_ON_BAT=power is too aggressive

fthx, 2020-02-02:
Hi,

In TLP 1.3beta4, running Ubuntu 20.04 kernel 5.4, Thinkpad T480s.

The default option **CPU_ENERGY_PERF_POLICY_ON_BAT=power** makes my i7 8550u cpu stay at 800 MHz. I suggest reverting to (1.2 default, right ? I don't remember that) **CPU_ENERGY_PERF_POLICY_ON_BAT=balance_power**.

Thanks for your work !

linrunner, 2020-02-02:
Fixed. Thanks.

LinuxOnTheDesktop, 2020-02-04:
I do not think that my TLP 1.2 config even had a CPU_ENERGY_PERF_POLICY_ON_BAT. It did have a CPU_HWP_ON_BAT. I've made the necessary adjustments.

linrunner, 2020-02-04:
It's explained in the [changelog](https://github.com/linrunner/TLP/blob/master/changelog):

> Processor:
>       - CPU_ENERGY_PERF_POLICY_ON_AC/BAT: backward compatible merge of
>         settings for Intel energy vs. performance policies
>         EPB (ENERGY_PERF_POLICY_ON_AC/BAT) and HWP.EPP (CPU_HWP_ON_AC/BAT);

[1.2.2](https://github.com/linrunner/TLP/blob/1.2.2/default) had CPU_HWP_ON_BAT=balance_power

linrunner, 2020-02-05:
note @self: FAQ?

The-First-King, 2020-02-13:
There is a mistake in documentation at page: https://linrunner.de/en/tlp/docs/tlp-configuration.html#scaling

Hints:
- Requires Intel Core i Gen. 5 (Skylake) or newer CPU and the intel_pstate scaling driver

On the i7-4510U processor i.e. Gen. 4 min/max P-states are working great the same as "turbo boost".

I didn't see e-mail for a feedback and post it here. Thanks!

linrunner, 2020-02-13:
@The-First-King : the restriction to Gen. 5 refers to  CPU_ENERGY_PERF_POLICY and CPU_HWP (HWP.EPP or EPB ) not to CPU_MIN/MAX_PERF and CPU_BOOST. May I see the output of 

    tlp-stat -p

please.

The-First-King, 2020-02-13:
```
--- TLP 1.1 --------------------------------------------

+++ Processor
CPU model      = Intel(R) Core(TM) i7-4510U CPU @ 2.00GHz

/sys/devices/system/cpu/cpu0/cpufreq/scaling_driver    = intel_pstate
/sys/devices/system/cpu/cpu0/cpufreq/scaling_governor  = powersave
/sys/devices/system/cpu/cpu0/cpufreq/scaling_available_governors = performance powersave
/sys/devices/system/cpu/cpu0/cpufreq/scaling_min_freq  =   800000 [kHz]
/sys/devices/system/cpu/cpu0/cpufreq/scaling_max_freq  =  1900000 [kHz]

/sys/devices/system/cpu/cpu1/cpufreq/scaling_driver    = intel_pstate
/sys/devices/system/cpu/cpu1/cpufreq/scaling_governor  = powersave
/sys/devices/system/cpu/cpu1/cpufreq/scaling_available_governors = performance powersave
/sys/devices/system/cpu/cpu1/cpufreq/scaling_min_freq  =   800000 [kHz]
/sys/devices/system/cpu/cpu1/cpufreq/scaling_max_freq  =  1900000 [kHz]

/sys/devices/system/cpu/cpu2/cpufreq/scaling_driver    = intel_pstate
/sys/devices/system/cpu/cpu2/cpufreq/scaling_governor  = powersave
/sys/devices/system/cpu/cpu2/cpufreq/scaling_available_governors = performance powersave
/sys/devices/system/cpu/cpu2/cpufreq/scaling_min_freq  =   800000 [kHz]
/sys/devices/system/cpu/cpu2/cpufreq/scaling_max_freq  =  1900000 [kHz]

/sys/devices/system/cpu/cpu3/cpufreq/scaling_driver    = intel_pstate
/sys/devices/system/cpu/cpu3/cpufreq/scaling_governor  = powersave
/sys/devices/system/cpu/cpu3/cpufreq/scaling_available_governors = performance powersave
/sys/devices/system/cpu/cpu3/cpufreq/scaling_min_freq  =   800000 [kHz]
/sys/devices/system/cpu/cpu3/cpufreq/scaling_max_freq  =  1900000 [kHz]

/sys/devices/system/cpu/intel_pstate/min_perf_pct      =  25 [%]
/sys/devices/system/cpu/intel_pstate/max_perf_pct      =  30 [%]
/sys/devices/system/cpu/intel_pstate/no_turbo          =   1
/sys/devices/system/cpu/intel_pstate/turbo_pct         =  50 [%]
/sys/devices/system/cpu/intel_pstate/num_pstates       =  24

x86_energy_perf_policy: program not installed.

/sys/module/workqueue/parameters/power_efficient       = Y
/proc/sys/kernel/nmi_watchdog                          = 0

+++ Undervolting
PHC kernel not available.
```

On battery - text reading

```
$ lscpu | grep MHz
CPU MHz:             988.219
CPU max MHz:         3100,0000
CPU min MHz:         800,0000
```

On battery - 720p video at youtube

```
$ lscpu | grep MHz
CPU MHz:             997.623
CPU max MHz:         3100,0000
CPU min MHz:         800,0000
```

linrunner, 2020-04-29:
Better late than never: I see no evidence for your claim in the outputs. HWP.EPP would look like (TLP 1.3):

> /sys/devices/system/cpu/cpu0/cpufreq/energy_performance_preference = balance_performance [HWP.EPP]
> /sys/devices/system/cpu/cpu0/cpufreq/energy_performance_available_preferences = default performance balance_performance balance_power power

closed 2020-04-29 by linrunner