linrunner/TLP #461: TLP caused system dead when laptop in battery mode (Ubuntu 18.04 + 10710U, MX250)

Fitz13009, 2020-02-03:
Hello,

I am setting up my new laptop with Ubuntu 18.04 and TLP as power management.
I found when Tlp is in battery mode, the laptop will be dead shortly, completely no response with mouse and keyboard. I can only use power button to restart the laptop. 

Here is some info of my laptop:
Manufacturer: Lenovo
Model: Xiaoxin Pro 13
Cpu: Intel Core I7-10710U
GPU: Nvidia MX250
Linux Kernel: 5.3.0-28-generic
TLP's configuration: default

I did some coarse experiment to make sure that the problem is coming from TLP:
0. I use TLP when I use laptop's battery, laptop dies. 
1. I shutdown TLP when I use laptop's battery and the laptop works fine.
2. I use TLP's battery mode even when laptop connects to AC charger, the laptop dies.

I would appreciate any suggestion if this is not a bug of TLP, otherwise I am glad to offer more info you desired to help debug.

Thanks a lot.
All the best,
Fitz

linrunner, 2020-02-03:
Hi Fitz,

I do not do support by mail but I think this could be helpful:

https://linrunner.de/en/tlp/docs/tlp-troubleshooting.html

Thomas

Fitz13009, 2020-02-03:
> Hi Fitz, I do not do support by mail but I think this could be helpful: https://linrunner.de/en/tlp/docs/tlp-troubleshooting.html Thomas
> […](#)

Thank you very much. The problem comes from the Nvidia card. After put it into tlp‘s blacklist and let Nvidia prime to manage it, the problem has been solved :D

closed 2020-02-03 by Fitz13009