linrunner/TLP #462: No command to show version

LinuxOnTheDesktop, 2020-02-03:
[x] I've read and accepted the [Bug Reporting Howto](https://github.com/linrunner/TLP/blob/master/.github/Bug_Reporting_Howto.md)

**Describe the bug**

`tlp --version` and variants throw an error.

**Expected behavior**

The version number should be displayed.

**To Reproduce**

`tlp --version`

**Additional context**

tlp version . . er . . 1.2.2.1. :)

linrunner, 2020-02-04:
Every output of tlp-stat shows the TLP version at the top, which is perfectly adequate.

LinuxOnTheDesktop, 2020-02-04:
Well, that is not standard and hence it will not occur to everybody (not to mention the fact that there was a recent change of that very command). Why flout a useful, very widely adhered to, easy to adhere to, convention?

closed 2020-02-05 by linrunner

LinuxOnTheDesktop, 2023-08-08:
Just now, in trying to determine which of my computers I was testing the tlp beta on, I ran into this problem again.

I note/add:

- not everyone will know to - or think to - use `tlp-stat`;
- `tlp-stat` needs root;
- if one is after only the version number then, without a grep, one has to let the output finish and then scroll to the top of it.

Why not just implement this 'quality of life' enhancement, viz., `tlp --version`? I add, finally: there's something on the 'net somewhere about someone who was fairly new to Linux, wanted to know the version of some software, thought 'well, `--version` works on other programs . . ', ran it on the program in question, and was chuffed with Linux when the command worked.

linrunner, 2023-08-08:
I made a note for the backlog of 1.7.

linrunner, 2024-04-01:
Done.

linrunner, 2024-09-05:
Hi @LinuxOnTheDesktop : TLP 1.7 Beta 1 is out with the corresponding change. Please verify -> https://github.com/linrunner/TLP/issues/760

linrunner, 2024-09-27:
The beta test is complete, and [TLP 1.7.0](https://github.com/linrunner/TLP/releases/tag/1.7.0) is now available. Have fun!

A huge thank you to all of our wonderful contributors, testers, and bug reporters! :+1: