linrunner/TLP #468: Add configuration setting for intel's p-state HWP_DYNAMIC_BOOST flag

tomgar, 2020-02-10:
From the manual https://01.org/linuxgraphics/gfx-docs/drm/admin-guide/pm/intel_pstate.html#user-space-interface-in-sysfs : 
"hwp_dynamic_boost

    This attribute is only present if intel_pstate works in the active mode with the HWP feature enabled in the processor. If set (equal to 1), it causes the minimum P-state limit to be increased dynamically for a short time whenever a task previously waiting on I/O is selected to run on a given logical CPU (the purpose of this mechanism is to improve performance).

    This setting has no effect on logical CPUs whose minimum P-state limit is directly set to the highest non-turbo P-state or above it."

, something like HWP_DYNAMIC_BOOST_ON_AC = and  HWP_DYNAMIC_BOOST_ON_BAT = 0|1 to enable|disable would suffice.

Thanks.

linrunner, 2020-02-13:
Hi,

did you check what really happens (power comsunption, frequencies / p-states) when you enable the feature? 

> causes the minimum P-state limit to be increased dynamically for a short time

Sounds like increased power consumption.

tomgar, 2020-02-13:
Hi,

 I'm not considering its energy expenditure since I'd like to have that extra performance. My use case is that when I have my laptop connected to AC I'm not using the Performance governor, but I'd like some extra-performance boost while I'm using it if something's taking a lot of CPU for a moment. I'd rather have tasks finished earlier than later due to going back to lower p-states. Besides going back & forth with p-states increases latency in I/O.

  Currently, I have a systemd tmpfile.d file to activate the flag, but that means than when my laptop is not connected to AC, I'll have to disable that feature manually which is a chore. Please, add it but the defaults should be 0 (or not touching it if not defined) for both, AC and BAT. It's no different than the turbo flag currently in TLP, actually.

Thanks.

linrunner, 2020-02-13:
Thank you for the explanation and specification of the requirement.

linrunner, 2021-04-09:
Done. Please test with the *main* branch and show the output of

    tlp-stat -s -c -p

for AC and battery mode.

closed 2021-09-24 by linrunner